polhenarejos/pico-fido #77: Pico Fido flashing crashed when using ESP32 Flasher

piotr277, 2024-12-05:
I'm reporting serious issue which just happened to me.
My device is **Waveshare ESP32-S3-Zero**. The host machine is Linux, web browser - Chrome.
I started flashing process using online ESP32 Flasher (https://www.picokeys.com/esp32-flasher). The flashing process was going well but stuck at info "wrapping up" for several minutes. I've checked the browser console. Here are the logs:
```
Serial port WebSerial VendorID 0x303a ProductID 0x1001
install-dialog-BWZCBYvU.js?module:477 Connecting...
install-dialog-BWZCBYvU.js?module:477 .
install-dialog-BWZCBYvU.js?module:477 

install-dialog-BWZCBYvU.js?module:477 Detecting chip type... 
install-dialog-BWZCBYvU.js?module:477 ESP32-S3
install-dialog-BWZCBYvU.js?module:477 Chip is ESP32-S3
install-dialog-BWZCBYvU.js?module:477 Features: Wi-Fi,BLE
install-dialog-BWZCBYvU.js?module:477 Crystal is 40MHz
install-dialog-BWZCBYvU.js?module:477 MAC: 24:ec:4a:26:88:78
install-dialog-BWZCBYvU.js?module:477 Uploading stub...
install-dialog-BWZCBYvU.js?module:477 Running stub...
install-dialog-BWZCBYvU.js?module:477 Stub running...
install-dialog-BWZCBYvU.js?module:477 Manufacturer: 20
install-dialog-BWZCBYvU.js?module:477 Device: 4016
install-dialog-BWZCBYvU.js?module:477 Detected flash size: 4MB
install-dialog-BWZCBYvU.js?module:477 Erasing flash (this may take a while)...
install-dialog-BWZCBYvU.js?module:477 Chip erase completed successfully in 14.59s
install-dialog-BWZCBYvU.js?module:477 Not changing the image
install-dialog-BWZCBYvU.js?module:477 Compressed 591632 bytes to 306232...
install-dialog-BWZCBYvU.js?module:477 Writing at 0x0... (5%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x119cd... (10%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x202cc... (15%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x27b64... (21%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x2cbfe... (26%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x34aef... (31%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x3af4a... (36%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x4109c... (42%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x46ff0... (47%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x4d876... (52%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x54672... (57%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x5a9fc... (63%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x60964... (68%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x67224... (73%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x6d041... (78%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x76f37... (84%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x7e871... (89%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x85736... (94%)
install-dialog-BWZCBYvU.js?module:477 Writing at 0x8bde9... (100%)
install-dialog-BWZCBYvU.js?module:477 Wrote 591632 bytes (306232 compressed) at 0x0 in 5.247 seconds.
install-dialog-BWZCBYvU.js?module:477 Leaving...
install-dialog-BWZCBYvU.js?module:807 HARD RESET
install-dialog-BWZCBYvU.js?module:807 DISCONNECT
esp32-flasher/:1 Uncaught (in promise) NetworkError: Failed to execute 'open' on 'SerialPort': Failed to open serial port.
```
Now the device _seems to be dead_  - no led lights, nothing when connecting to USB. Here are dmesg logs when I connect it to USB:
```
[   51.982370] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[   57.251445] usb 1-2: device descriptor read/all, error -110
```
What can I do to make it alive again..? help, please.

polhenarejos, 2024-12-05:
You can put it in DFU mode and try flashing through espressif tools.

piotr277, 2024-12-08:
After I kept pressed BOOT + RESET buttons, then released RESET, the board became again visible to OS as serial / uart. Then I flashed it **with success** with online ESP32 Flasher. No more error during that process. 
However, now when I plug it to my Linux machine I still get the same dmesg messages as previously:
```
[  960.116722] usb 1-2: new full-speed USB device number 8 using xhci_hcd
[  965.450015] usb 1-2: device descriptor read/all, error -110
```
Is this a normal behaviour?
Next I tried to use online Pico Commisioner (https://www.picokeys.com/pico-commissioner/) but without success.. :( the device is not accessible.

polhenarejos, 2024-12-08:
After flashing, is it listed by `lsusb`?

piotr277, 2024-12-09:
No it isn't. 
After pressed BOOT + RESET and releasing RESET it appears as serial / uart.

polhenarejos, 2024-12-09:
This is normal. Then you have to flash it. After flashing the image, do you see it at `lsusb`?

piotr277, 2024-12-09:
No, after successful flashing I don't see it at `lsusb`.

polhenarejos, 2024-12-09:
By default should appear with VID/PID FFFE:FDFC. If it doesn't appear, it might have some problem with USB stack. Which board do you use?

polhenarejos, 2024-12-09:
I have the same. I'd recommend to use Espressif tools to avoid any problem from the browser.

piotr277, 2024-12-09:
Ok, so you recommend to clone repo and build from the source, right?

polhenarejos, 2024-12-09:
It would be nice, but you can try first downloading the firmware from here https://github.com/polhenarejos/polhenarejos.github.io/blob/main/esp/fw/pico_fido_esp32-s3_6.0.bin
and upload it to your board. If it does not work (very strange), then you can try building it. I use VSCode but you can install the tools standalone:
```bash
sudo apt install -y git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh esp32s3
. ./export.sh
cd ..
idf.py set-target esp32s3
idf.py all
```

piotr277, 2024-12-09:
Thanks. I'll try in the afternoon and let you know.

piotr277, 2024-12-09:
Ok, so I downloaded the bin firmware from the link you provided.
Here is the flash command I used:
```
esptool.py -p /dev/ttyACM0 write_flash -e 0x00000 pico_fido_esp32-s3_6.0.bin
```
The command result was success:
```
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 24:ec:4a:26:88:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.4s
Compressed 591632 bytes to 306232...
Wrote 591632 bytes (306232 compressed) at 0x00000000 in 5.1 seconds (effective 930.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
But the final behaviour is the same - I got the same messages in `dmesg`:
```
[ 3324.767822] usb 1-2: new full-speed USB device number 18 using xhci_hcd
[ 3329.971481] usb 1-2: device descriptor read/all, error -110
```
and the device is not visible in `lsusb` :(.

polhenarejos, 2024-12-09:
You have to use the `development` branch.

piotr277, 2024-12-09:
Ok, build succeded. Thanks for the tip.
But result is still the same..
Here are the logs:
```
idf.py flash monitor
```
```
Executing action: flash
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory /home/piotr/src/esp32s/pico-fido/build
Executing "ninja flash"...
[1/5] cd /home/piotr/src/esp32s/pico-fido/build/esp-idf/esptool_py && /home/piotr/.espressif...ild/partition_table/partition-table.bin /home/piotr/src/esp32s/pico-fido/build/pico_fido.bin
pico_fido.bin binary size 0x79530 bytes. Smallest app partition is 0x100000 bytes. 0x86ad0 bytes (53%) free.
[1/1] cd /home/piotr/src/esp32s/pico-fido/build/bootloader/esp-idf/esptool_py && /home/piotr...ffset 0x8000 bootloader 0x0 /home/piotr/src/esp32s/pico-fido/build/bootloader/bootloader.bin
Bootloader binary size 0x5490 bytes. 0x2b70 bytes (34%) free.
[4/5] cd /opt/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/esp-idf -D "S...piotr/src/esp32s/pico-fido/build -P /opt/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 pico_fido.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 24:ec:4a:26:88:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00089fff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21648 bytes to 13318...
Writing at 0x00000000... (100 %)
Wrote 21648 bytes (13318 compressed) at 0x00000000 in 0.3 seconds (effective 507.8 kbit/s)...
Hash of data verified.
Compressed 496944 bytes to 268151...
Writing at 0x00086cef... (100 %)
Wrote 496944 bytes (268151 compressed) at 0x00010000 in 4.3 seconds (effective 919.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 117...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (117 compressed) at 0x00008000 in 0.0 seconds (effective 605.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /home/piotr/src/esp32s/pico-fido
Executing "/home/piotr/.espressif/python_env/idf5.3_py3.12_env/bin/python /opt/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --decode-coredumps info --target esp32s3 --revision 0 /home/piotr/src/esp32s/pico-fido/build/pico_fido.elf -m '/home/piotr/.espressif/python_env/idf5.3_py3.12_env/bin/python' '/opt/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
--- 0x40041a76: ets_delay_us in ROM

waiting for download

```

polhenarejos, 2024-12-09:
Did you reset after upload? It says “waiting for download” which means it is still in DFU.

piotr277, 2024-12-10:
Yes, I did. After reset it is not discoverable in `lsusb` unless I put it in DFU mode.

polhenarejos, 2024-12-10:
A bad board could be an option. The next I'd do is trying one official example from here:

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/

`tusb_hid` is a good choice.

piotr277, 2024-12-10:
Ok, one more observation..
I've managed to build from the source **pico-hsm** and flashed the board. And it is discoverable after flashing!
Here is `dmesg`:
```
[23715.798368] usb 1-2: new full-speed USB device number 81 using xhci_hcd
[23721.005924] usb 1-2: device descriptor read/all, error -110
[23721.334904] usb 1-2: new full-speed USB device number 82 using xhci_hcd
[23721.476169] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[23721.476918] usb 1-2: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[23721.476928] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[23721.476932] usb 1-2: Product: Pico Key
[23721.476935] usb 1-2: Manufacturer: Pol Henarejos
[23721.476938] usb 1-2: SerialNumber: 7888264AEC240000
```
and `lsusb`:
```
Bus 001 Device 082: ID feff:fcfd Pol Henarejos Pico Key
```

So what's wrong with **pico-fido**..?

polhenarejos, 2024-12-10:
There’s something in the HID descriptors you OS doesn’t like. 
Even with Pico HSM your OS is complaining about bInterval parameter but in this case, for whatever reason, it decides continue registering the device. For Pico-fido it decides to abort it but withou any additional information. 

Which OS do you use?

piotr277, 2024-12-11:
> Which OS do you use? Btw make sure your board blinks after reset.

Manjaro Linux kernel 6.6.63-1
I don't see any blink of LED on my board.. 

> Also it seems you are using a USB3 port. Try using a USB2 port.

It's a pity but I don't have any USB2 port.. :(

piotr277, 2024-12-11:
@polhenarejos keeping fingers crossed it will help :)

polhenarejos, 2024-12-11:
I pushed a fix in `development`. You can build yourself and try if needed.

piotr277, 2024-12-11:
@polhenarejos I've pulled the code from `development` branch and now the board is visible in dmesg, hurray! :smile: 
```
[14451.424028] usb 1-2: new full-speed USB device number 14 using xhci_hcd
[14456.570713] usb 1-2: device descriptor read/all, error -110
[14456.900512] usb 1-2: new full-speed USB device number 15 using xhci_hcd
[14457.042602] usb 1-2: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 7.00
[14457.042609] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14457.042611] usb 1-2: Product: Pico Key
[14457.042613] usb 1-2: Manufacturer: Pol Henarejos
[14457.042614] usb 1-2: SerialNumber: 7888264AEC240000
[14457.047902] hid-generic 0003:1050:0407.0007: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input0
[14457.051011] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1050:0407.0008/input/input17
[14457.187251] hid-generic 0003:1050:0407.0008: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input1
```
and in lsusb (I've commisioned it earlier via Pico Commissioner when I had pico-hsm installed):
```
Bus 001 Device 015: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
``` 

Huge thanks @polhenarejos ! :clap:  Now it's time for testing..

piotr277, 2024-12-11:
Ok.. so it's better but not perfect.. ;)
If I try to use Yubikey Personalization Tool or WebAuthn site (like webauthn.io) it enters an infinite loop of disconnect -> connect:
```
[15547.466402] usb 1-2: USB disconnect, device number 57
[15548.030586] usb 1-2: new full-speed USB device number 58 using xhci_hcd
[15548.172368] usb 1-2: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 7.00
[15548.172374] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15548.172376] usb 1-2: Product: Pico Key
[15548.172377] usb 1-2: Manufacturer: Pol Henarejos
[15548.172379] usb 1-2: SerialNumber: 7888264AEC240000
[15548.176525] hid-generic 0003:1050:0407.0062: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input0
[15548.178158] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1050:0407.0063/input/input68
[15548.313914] hid-generic 0003:1050:0407.0063: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input1
[15550.031539] usb 1-2: USB disconnect, device number 58
[15550.550362] usb 1-2: new full-speed USB device number 59 using xhci_hcd
[15551.171381] usb 1-2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[15551.171387] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15551.171389] usb 1-2: Product: USB JTAG/serial debug unit
[15551.171391] usb 1-2: Manufacturer: Espressif
[15551.171392] usb 1-2: SerialNumber: 24:EC:4A:26:88:78
[15551.173029] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[15551.209328] usb 1-2: USB disconnect, device number 59
[15551.623709] usb 1-2: new full-speed USB device number 60 using xhci_hcd
[15551.765486] usb 1-2: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 7.00
[15551.765492] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15551.765494] usb 1-2: Product: Pico Key
[15551.765495] usb 1-2: Manufacturer: Pol Henarejos
[15551.765497] usb 1-2: SerialNumber: 7888264AEC240000
[15551.769273] hid-generic 0003:1050:0407.0064: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input0
[15551.770612] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1050:0407.0065/input/input69
[15551.907260] hid-generic 0003:1050:0407.0065: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input1
[15554.456223] usb 1-2: USB disconnect, device number 60
[15555.046856] usb 1-2: new full-speed USB device number 61 using xhci_hcd
[15555.188826] usb 1-2: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 7.00
[15555.188831] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15555.188834] usb 1-2: Product: Pico Key
[15555.188835] usb 1-2: Manufacturer: Pol Henarejos
[15555.188836] usb 1-2: SerialNumber: 7888264AEC240000
[15555.192928] hid-generic 0003:1050:0407.0066: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input0
[15555.194205] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1050:0407.0067/input/input70
[15555.327049] hid-generic 0003:1050:0407.0067: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-2/input1
[15557.799033] usb 1-2: USB disconnect, device number 61

```

polhenarejos, 2024-12-11:
Can you pull latest fixes and build it? It might be related with #72

piotr277, 2024-12-12:
Yes, I've pulled the latest commit (https://github.com/polhenarejos/pico-fido/commit/022503fdc0632b8438d946dceefc9ff3ab0c593e) from `development` with all submodules. Then I build it. For being 100% sure I also erased flash with `esptool.py -p /dev/ttyACM0 erase_flash`. Then I flashed board with image I had built.
But unfortunately, the result is still the same - disconnect / connect loop, when I try to use webauthn.io:
```
[ 8765.988674] usb 1-1: USB disconnect, device number 36
[ 8766.578548] usb 1-1: new full-speed USB device number 37 using xhci_hcd
[ 8766.729731] usb 1-1: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[ 8766.729737] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8766.729739] usb 1-1: Product: Pico Key
[ 8766.729741] usb 1-1: Manufacturer: Pol Henarejos
[ 8766.729742] usb 1-1: SerialNumber: 7888264AEC240000
[ 8766.733847] hid-generic 0003:FEFF:FCFD.002B: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input0
[ 8766.735376] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:FEFF:FCFD.002C/input/input35
[ 8766.859061] hid-generic 0003:FEFF:FCFD.002C: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input1
[ 8769.453853] usb 1-1: USB disconnect, device number 37
[ 8770.041943] usb 1-1: new full-speed USB device number 38 using xhci_hcd
[ 8770.187175] usb 1-1: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[ 8770.187181] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8770.187183] usb 1-1: Product: Pico Key
[ 8770.187185] usb 1-1: Manufacturer: Pol Henarejos
[ 8770.187186] usb 1-1: SerialNumber: 7888264AEC240000
[ 8770.193751] hid-generic 0003:FEFF:FCFD.002D: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input0
[ 8770.196042] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:FEFF:FCFD.002E/input/input36
[ 8770.338556] hid-generic 0003:FEFF:FCFD.002E: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input1
[ 8772.930910] usb 1-1: USB disconnect, device number 38
[ 8773.534982] usb 1-1: new full-speed USB device number 39 using xhci_hcd
[ 8773.680731] usb 1-1: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[ 8773.680748] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8773.680753] usb 1-1: Product: Pico Key
[ 8773.680757] usb 1-1: Manufacturer: Pol Henarejos
[ 8773.680762] usb 1-1: SerialNumber: 7888264AEC240000
[ 8773.687537] hid-generic 0003:FEFF:FCFD.002F: hiddev96,hidraw2: USB HID v1.11 Device [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input0
[ 8773.689112] input: Pol Henarejos Pico Key as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:FEFF:FCFD.0030/input/input37
[ 8773.831388] hid-generic 0003:FEFF:FCFD.0030: input,hidraw3: USB HID v1.11 Keyboard [Pol Henarejos Pico Key] on usb-0000:00:14.0-1/input1
```

I've tried also with https://www.token2.com/tools/fido2-demo but no success. The behavior is the same - the device is not discovered and "crash loop" begins.

polhenarejos, 2024-12-12:
Do you see a light change behavior between these loops? Like switching between red green blue or blinking frequency.

HeavyDutySoul, 2024-12-12:
Same, yes, led blinking different colours.

piotr277, 2024-12-12:
@polhenarejos for me no.. I see no LED light at all :(

polhenarejos, 2024-12-13:
Please follow these instructions to generate a coredump https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-esp-coredump-enable-to-flash

1. Append this line to `pico-fido/pico-keys-sdk/config/esp32/partitions.csv`:
```
coredump, data, coredump,,        64K
```
2. Add these lines to `pico-fido/sdkconfig.defaults`:
```
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
# CONFIG_ESP_COREDUMP_ENABLE_TO_UART is not set
# CONFIG_ESP_COREDUMP_ENABLE_TO_NONE is not set
# CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN is not set
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
# CONFIG_ESP_COREDUMP_CHECKSUM_SHA256 is not set
# CONFIG_ESP_COREDUMP_CAPTURE_DRAM is not set
CONFIG_ESP_COREDUMP_CHECK_BOOT=y
CONFIG_ESP_COREDUMP_ENABLE=y
CONFIG_ESP_COREDUMP_LOGS=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=64
# CONFIG_ESP_COREDUMP_FLASH_NO_OVERWRITE is not set
CONFIG_ESP_COREDUMP_STACK_SIZE=0
```
3. Remove `pico-fido/sdkconfig`.
4. Build again and flash.
5. A new `sdkconfig` should be regenerated.
6. Reset your board and hopefully it will generate a coredump in the flash.
7. Put the board in DFU again.
8. Type `idf.py coredump-info` to retrieve the coredump.
9. Paste here the whole coredump displayed.

piotr277, 2024-12-16:
Ok, I tried but this is what I got:
```
Executing action: coredump-info
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
WARNING:root:The core dump image offset is not specified. Use partition offset: 0x300000.
ERROR:root:Incorrect size of core dump image: 4294967295, use partition size instead: 65536
Failed to load core dump: Core dump version "0xffff" is not supported!
```
Something wrong with `partitions.cvs`?
Here is the content:
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap

nvs,      data, nvs,     0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory,   0x10000, 1M,
part0,    0x40, 0x1,     0x200000,  1M,
coredump, data, coredump,        ,  64K,
```
and `sdkconfig.defaults`:
```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) Project Minimal Configuration
#
IGNORE_UNKNOWN_FILES_FOR_MANAGED_COMPONENTS=1

CONFIG_TINYUSB=y

CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="pico-keys-sdk/config/esp32/partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="pico-keys-sdk/config/esp32/partitions.csv"
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_MODE_PERF=y
COMPILER_OPTIMIZATION="Performance"

CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_CHACHA20_C=y
CONFIG_MBEDTLS_POLY1305_C=y
CONFIG_MBEDTLS_CHACHAPOLY_C=y
CONFIG_MBEDTLS_HKDF_C=y
CONFIG_MBEDTLS_HARDWARE_ECC=y
CONFIG_MBEDTLS_HARDWARE_GCM=y
# CONFIG_MBEDTLS_HARDWARE_MPI is not set
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_HARDWARE_AES=y
# CONFIG_MBEDTLS_ROM_MD5 is not set
CONFIG_MBEDTLS_SHA512_C=y
CONFIG_MBEDTLS_TLS_DISABLED=y
# CONFIG_MBEDTLS_TLS_ENABLED is not set
# CONFIG_ESP_TLS_USE_DS_PERIPHERAL is not set
# CONFIG_ESP_WIFI_ENABLED is not set
# CONFIG_ESP_WIFI_MBEDTLS_CRYPTO is not set
# CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT is not set
# CONFIG_WPA_MBEDTLS_CRYPTO is not set
# CONFIG_MBEDTLS_PSK_MODES is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_RSA is not set
# CONFIG_MBEDTLS_SSL_RENEGOTIATION is not set
# CONFIG_MBEDTLS_SSL_PROTO_TLS1_2 is not set
# CONFIG_MBEDTLS_SSL_PROTO_GMTSSL1_1 is not set
# CONFIG_MBEDTLS_SSL_PROTO_DTLS is not set
# CONFIG_MBEDTLS_SSL_ALPN is not set
# CONFIG_MBEDTLS_CLIENT_SSL_SESSION_TICKETS is not set
# CONFIG_MBEDTLS_SERVER_SSL_SESSION_TICKETS is not set
# CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE is not set
# CONFIG_ESP32_WIFI_ENABLE_WPA3_OWE_STA is not set
# CONFIG_ESP_WIFI_ENABLE_WPA3_SAE is not set
# CONFIG_ESP_WIFI_ENABLE_WPA3_OWE_STA is not set

# CONFIG_ESP_COREDUMP_ENABLE_TO_UART=y

CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
# CONFIG_ESP_COREDUMP_ENABLE_TO_UART is not set
# CONFIG_ESP_COREDUMP_ENABLE_TO_NONE is not set
# CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN is not set
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
# CONFIG_ESP_COREDUMP_CHECKSUM_SHA256 is not set
# CONFIG_ESP_COREDUMP_CAPTURE_DRAM is not set
CONFIG_ESP_COREDUMP_CHECK_BOOT=y
CONFIG_ESP_COREDUMP_ENABLE=y
CONFIG_ESP_COREDUMP_LOGS=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=64
# CONFIG_ESP_COREDUMP_FLASH_NO_OVERWRITE is not set
CONFIG_ESP_COREDUMP_STACK_SIZE=0
```

polhenarejos, 2024-12-16:
Ok, then try with this line in `partitions.csv`:
```
coredump, data, coredump, 0x300000,        64K
```

piotr277, 2024-12-16:
I've updated `partitions.csv` but the result seems the same:
```
Executing action: coredump-info
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
WARNING:root:The core dump image offset is not specified. Use partition offset: 0x300000.
ERROR:root:Incorrect size of core dump image: 4294967295, use partition size instead: 65536
Failed to load core dump: Core dump version "0xffff" is not supported!
```
But build seems to respect updated `partitions.csv`:
```
Executing action: all (aliases: build)
Running ninja in directory /home/piotr/src/esp32s/pico-fido/build
Executing "ninja all"...
[5/1149] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
part0,64,1,0x200000,1M,
coredump,data,coredump,0x300000,64K,
*******************************************************************************
[1100/1149] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.47.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/piotr/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/piotr/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/piotr/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /home/piotr/src/esp32s/pico-fido/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /opt/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /opt/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /opt/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /opt/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /opt/esp-idf/components/bootloader /opt/esp-idf/components/bootloader_support /opt/esp-idf/components/efuse /opt/esp-idf/components/esp_app_format /opt/esp-idf/components/esp_bootloader_format /opt/esp-idf/components/esp_common /opt/esp-idf/components/esp_hw_support /opt/esp-idf/components/esp_rom /opt/esp-idf/components/esp_system /opt/esp-idf/components/esptool_py /opt/esp-idf/components/freertos /opt/esp-idf/components/hal /opt/esp-idf/components/log /opt/esp-idf/components/bootloader/subproject/main /opt/esp-idf/components/bootloader/subproject/components/micro-ecc /opt/esp-idf/components/newlib /opt/esp-idf/components/partition_table /opt/esp-idf/components/soc /opt/esp-idf/components/spi_flash /opt/esp-idf/components/xtensa
-- Configuring done (19.8s)
-- Generating done (0.2s)
-- Build files have been written to: /home/piotr/src/esp32s/pico-fido/build/bootloader
[113/114] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/piotr/src/esp32s/pico-fido/build/bootloader/bootloader.bin
[114/114] cd /home/piotr/src/esp32s/pico-fido/build/bootloader/esp-idf/esptool_py && /home/p...ffset 0x8000 bootloader 0x0 /home/piotr/src/esp32s/pico-fido/build/bootloader/bootloader.bin
Bootloader binary size 0x5490 bytes. 0x2b70 bytes (34%) free.
[1148/1149] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/piotr/src/esp32s/pico-fido/build/pico_fido.bin
[1149/1149] cd /home/piotr/src/esp32s/pico-fido/build/esp-idf/esptool_py && /home/piotr/.esp...ild/partition_table/partition-table.bin /home/piotr/src/esp32s/pico-fido/build/pico_fido.bin
pico_fido.bin binary size 0x7a0f0 bytes. Smallest app partition is 0x100000 bytes. 0x85f10 bytes (52%) free.

Project build complete. To flash, run:
 idf.py flash
```

polhenarejos, 2024-12-16:
Did you flash it correctly? It seems it is taking the old image.

piotr277, 2024-12-16:
Yes, I suppose.. here is the log of flashing:
```
Executing action: flash
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory /home/piotr/src/esp32s/pico-fido/build
Executing "ninja flash"...
[1/1] cd /home/piotr/src/esp32s/pico-fido/build/bootloader/esp-idf/esptool_py && /home/piotr...ffset 0x8000 bootloader 0x0 /home/piotr/src/esp32s/pico-fido/build/bootloader/bootloader.bin
Bootloader binary size 0x5490 bytes. 0x2b70 bytes (34%) free.
[2/5] cd /home/piotr/src/esp32s/pico-fido/build/esp-idf/esptool_py && /home/piotr/.espressif...ild/partition_table/partition-table.bin /home/piotr/src/esp32s/pico-fido/build/pico_fido.bin
pico_fido.bin binary size 0x7a0f0 bytes. Smallest app partition is 0x100000 bytes. 0x85f10 bytes (52%) free.
[4/5] cd /opt/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/esp-idf -D "S...piotr/src/esp32s/pico-fido/build -P /opt/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 pico_fido.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 24:ec:4a:26:88:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0008afff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21648 bytes to 13306...
Writing at 0x00000000... (100 %)
Wrote 21648 bytes (13306 compressed) at 0x00000000 in 0.3 seconds (effective 515.1 kbit/s)...
Hash of data verified.
Compressed 499952 bytes to 269429...
Writing at 0x00087258... (100 %)
Wrote 499952 bytes (269429 compressed) at 0x00010000 in 4.4 seconds (effective 916.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 132...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (132 compressed) at 0x00008000 in 0.0 seconds (effective 607.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

HeavyDutySoul, 2024-12-16:
I tried on linux, webauthn.io work like a charm on chromium with pico-fido. In windows, crash and bootloop when i try set pin in FIDO2.1 Manager. Try to get coredump, but idf.py gives an error:
`Failed to load core dump: Invalid core dump CRC 31b59a50, should be ffffffff`

polhenarejos, 2024-12-16:
This is new. Can you open a separate issue?

HeavyDutySoul, 2024-12-16:
Ok, i try.

polhenarejos, 2024-12-16:
@piotr277 I pushed a small change that might block the device. Can you pull, build & try?

piotr277, 2024-12-17:
@polhenarejos I pulled, build and tried. But it's still the same:
```
Executing action: coredump-info
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
WARNING:root:The core dump image offset is not specified. Use partition offset: 0x300000.
ERROR:root:Incorrect size of core dump image: 4294967295, use partition size instead: 65536
Failed to load core dump: Core dump version "0xffff" is not supported!
```
and LED is not working. The behavior at https://www.token2.com/tools/fido2-demo is still the same (infinite loop). 
What does it mean: `Core dump version "0xffff" is not supported!` ?

HeavyDutySoul, 2024-12-18:
Rebuilt from the development branch and tried to install the pin in ykman under Linux. In this time I get a coredump.
`===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fcecff8, name: 'TinyUSB', GDB name: 'process 1070518264'
Crashed task is not in the interrupt context
Panic reason: ***ERROR*** A stack overflow in task TinyUSB has been detected.

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x40378e0b
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40375b24          0x40375b24 <panic_abort+16>
lbeg           0x400570e8          1074098408
lend           0x400570f3          1074098419
lcount         0x0                 0
sar            0x1c                28
ps             0x60023             393251
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8037a9e0          -2143835680
a1             0x3fcebf70          1070514032
a2             0x3fcebfbc          1070514108
a3             0x3fcebffb          1070514171
a4             0x3fcec1a0          1070514592
a5             0x3fcec180          1070514560
a6             0x3fce9710          1070503696
a7             0x44                68
a8             0x3fc970e0          1070166240
a9             0x1                 1
a10            0x44                68
a11            0x804               2052
a12            0x3fcec180          1070514560
a13            0x3fcec160          1070514528
a14            0x3fced9e0          1070520800
a15            0x44                68

==================== CURRENT THREAD STACK =====================
#0  panic_abort (details=0x3fcebfbc "***ERROR*** A stack overflow in task TinyUSB has been detected.") at /opt/esp-idf/components/esp_system/panic.c:466
#1  0x4037a9e0 in esp_system_abort (details=0x3fcebfbc "***ERROR*** A stack overflow in task TinyUSB has been detected.") at /opt/esp-idf/components/esp_system/port/esp_system_chip.c:92
#2  0x4037b88d in vApplicationStackOverflowHook (xTask=0x3fcecff8, pcTaskName=0x3fced02c "TinyUSB") at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
#3  0x4037ccd2 in vTaskSwitchContext () at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701
#4  0x4037b953 in _frxt_dispatch () at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
Backtrace stopped: Cannot access memory at address 0xa5a5a599

======================== THREADS INFO =========================
  Id   Target Id          Frame 
[... 69 lines cut ...]
.dram0.data 0x3fc93b00 0x342c RW A
.flash.text 0x42000020 0x4dfda R XA
.flash.appdesc 0x3c050020 0x100 R  A
.flash.rodata 0x3c050120 0x176d8 RW A
.iram0.data 0x40383b00 0x0 RW  
.iram0.bss 0x40383b00 0x0 RW  
.dram0.heap_start 0x3fca9700 0x0 RW  
.coredump.tasks.data 0x3fcaaccc 0x154 RW 
.coredump.tasks.data 0x3fcaaa30 0x290 RW 
.coredump.tasks.data 0x3fcecff8 0x154 RW 
.coredump.tasks.data 0x3fcebeb0 0x1140 RW 
.coredump.tasks.data 0x3fced15c 0x154 RW 
.coredump.tasks.data 0x3fcb1de0 0x350 RW 
.coredump.tasks.data 0x3fcacecc 0x154 RW 
.coredump.tasks.data 0x3fcacc40 0x280 RW 
.coredump.tasks.data 0x3fcac764 0x154 RW 
.coredump.tasks.data 0x3fcac4d0 0x280 RW 
.coredump.tasks.data 0x3fcaa664 0x154 RW 
.coredump.tasks.data 0x3fcaa3c0 0x290 RW 
.coredump.tasks.data 0x3fceb99c 0x154 RW 
.coredump.tasks.data 0x20000000 0x70 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!`

polhenarejos, 2024-12-18:
Unfortunately it doesn’t show anything meaningful. Can you cmake with `-DCMAKE_BUILD_TYPE=Debug` and build it again?

HeavyDutySoul, 2024-12-18:
All the same...
`===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fcecff8, name: 'TinyUSB', GDB name: 'process 1070518264'
Crashed task is not in the interrupt context
Panic reason: ***ERROR*** A stack overflow in task TinyUSB has been detected.

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x40378e0b
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40375b24          0x40375b24 <panic_abort+16>
lbeg           0x400570e8          1074098408
lend           0x400570f3          1074098419
lcount         0x0                 0
sar            0x1f                31
ps             0x60023             393251
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8037a9e0          -2143835680
a1             0x3fcebf70          1070514032
a2             0x3fcebfbc          1070514108
a3             0x3fcebffb          1070514171
a4             0x3fcec1b0          1070514608
a5             0x3fcec190          1070514576
a6             0x3fcec4f0          1070515440
a7             0x3fcec210          1070514704
a8             0x3fc970e0          1070166240
a9             0x1                 1
a10            0x3fcec4f4          1070515444
a11            0x3fcec4f0          1070515440
a12            0x3fcec190          1070514576
a13            0x3fcec170          1070514544
a14            0x3fcec210          1070514704
a15            0x3fcec558          1070515544

==================== CURRENT THREAD STACK =====================
#0  panic_abort (details=0x3fcebfbc "***ERROR*** A stack overflow in task TinyUSB has been detected.") at /opt/esp-idf/components/esp_system/panic.c:466
#1  0x4037a9e0 in esp_system_abort (details=0x3fcebfbc "***ERROR*** A stack overflow in task TinyUSB has been detected.") at /opt/esp-idf/components/esp_system/port/esp_system_chip.c:92
#2  0x4037b88d in vApplicationStackOverflowHook (xTask=0x3fcecff8, pcTaskName=0x3fced02c "TinyUSB") at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
#3  0x4037ccd2 in vTaskSwitchContext () at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701
#4  0x4037b953 in _frxt_dispatch () at /opt/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

======================== THREADS INFO =========================
  Id   Target Id          Frame 
[... 69 lines cut ...]
.dram0.data 0x3fc93b00 0x342c RW A
.flash.text 0x42000020 0x4dfda R XA
.flash.appdesc 0x3c050020 0x100 R  A
.flash.rodata 0x3c050120 0x176d8 RW A
.iram0.data 0x40383b00 0x0 RW  
.iram0.bss 0x40383b00 0x0 RW  
.dram0.heap_start 0x3fca9700 0x0 RW  
.coredump.tasks.data 0x3fcaaccc 0x154 RW 
.coredump.tasks.data 0x3fcaaa30 0x290 RW 
.coredump.tasks.data 0x3fcecff8 0x154 RW 
.coredump.tasks.data 0x3fcebeb0 0x1140 RW 
.coredump.tasks.data 0x3fced15c 0x154 RW 
.coredump.tasks.data 0x3fcb1de0 0x350 RW 
.coredump.tasks.data 0x3fcacecc 0x154 RW 
.coredump.tasks.data 0x3fcacc40 0x280 RW 
.coredump.tasks.data 0x3fcac764 0x154 RW 
.coredump.tasks.data 0x3fcac4d0 0x280 RW 
.coredump.tasks.data 0x3fcaa664 0x154 RW 
.coredump.tasks.data 0x3fcaa3c0 0x290 RW 
.coredump.tasks.data 0x3fceb99c 0x154 RW 
.coredump.tasks.data 0x20000000 0x70 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!`

polhenarejos, 2024-12-18:
`core0` is where pico-fido code resides and strangely it is not displayed. 

What do you do to reproduce it?

HeavyDutySoul, 2024-12-18:
I just tried to set the pin right after reflashing: ykman fido access change-pin -n 111111 Previously, webauthn.io offered to set the pin in the browser, now it doesn't. I've reflashed the device several times already.

polhenarejos, 2024-12-18:
Reflashing doesn’t clear user space. If set a pin, it is still present. To clean you have to call `esptool.py erase_flash` and then flash.

HeavyDutySoul, 2024-12-18:
I'm always erase flash before reflashing.

HeavyDutySoul, 2024-12-18:
`Crashed task handle: 0x3fcecff8, name: 'TinyUSB', GDB name: 'process 1070518264'`
Why task handle and process name in coredump's the same? It is normal?

polhenarejos, 2024-12-18:
Can you increase the stack size?

Change `4096*5` by `4096*10` at 

https://github.com/polhenarejos/pico-keys-sdk/blob/e627b3fc865df1ffc573dfe33d0ebe864f8beea4/src/main.c#L345C9-L345C110

a1bert01, 2024-12-19:
development branch coredump when:

`ssh-keygen -t ecdsa-sk -C  "descr"`

[coredump.txt.gz](https://github.com/user-attachments/files/18203118/coredump.txt.gz)

polhenarejos, 2024-12-19:
Did you increase the stack size ?

a1bert01, 2024-12-19:
previous coredump was with 4096*5, but looks the same with:
```
diff -u3 pico-keys-sdk/src/main.c~ pico-keys-sdk/src/main.c
--- pico-keys-sdk/src/main.c~	2024-12-18 19:13:51.000000000 +0100
+++ pico-keys-sdk/src/main.c	2024-12-20 00:18:52.518624493 +0100
@@ -332,7 +332,7 @@
 #endif
 
 #ifdef ESP_PLATFORM
-    xTaskCreatePinnedToCore(core0_loop, "core0", 4096*5, NULL, CONFIG_TINYUSB_TASK_PRIORITY - 1, &hcore0, 0);
+    xTaskCreatePinnedToCore(core0_loop, "core0", 4096*10, NULL, CONFIG_TINYUSB_TASK_PRIORITY - 1, &hcore0, 0);
 #else
     core0_loop();
 #endif

[coredump.txt.gz](https://github.com/user-attachments/files/18204116/coredump.txt.gz)

```

polhenarejos, 2024-12-20:
coredumps to UART have to be decoded by yourself, as it depends on the hash of the build and mine and yours differ. Please paste the result of 
`idf.py coredump-info -c coredump.txt`

a1bert01, 2024-12-20:
[c.txt.gz](https://github.com/user-attachments/files/18208711/c.txt.gz)

HeavyDutySoul, 2024-12-20:
> Did you increase the stack size ?
I tried with a larger stack size, the coredump output did not change. Now the device does not work at all after reflashing in Linux, when trying to set the pin code in Chromium, the LED blinks orange and green in turn and nothing happens. I managed several times to set pin in Chrome at Windows, after that the device works fine in both Linux and Windows.

a1bert01, 2024-12-20:
Are we increasing the right stack?? IMO stack overflow happens in "TinyUSB"  Thread, but we are increasing "core0" stack 
```
       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3fcb09a8          TinyUSB      5/5         3888/204
0x3fcaafe8             ipc1    24/24          656/620
0x3fcc4b04            core0      4/4        832/81072   <---------------------
0x3fcad01c            IDLE1      0/0          640/888
0x3fcac8c0            IDLE0      0/0          640/884
0x3fcaa98c             ipc0    24/24          656/616
0x3fcaf238         neopixel    24/24          704/316

       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3fcb09a8          TinyUSB      5/5          4112/20
0x3fcaafe8             ipc1    24/24          656/620
0x3fcbab04            core0      4/4        832/40112  <------------------------------
0x3fcad01c            IDLE1      0/0          640/888
0x3fcac8c0            IDLE0      0/0          640/884
0x3fcaa98c             ipc0    24/24          656/616
0x3fcaf238         neopixel    24/24          704/316
```

polhenarejos, 2024-12-20:
Yes, I think it is more probable having an stack overflow in my code rather than in TinyUSB. 

You can try increase the stack of TinyUSB by adding
```
CONFIG_TINYUSB_TASK_STACK_SIZE=16384
```
to `pico-fido/sdkconfig.defaults`. Be sure you remove `pico-fido/sdkconfig` first before building.

HeavyDutySoul, 2024-12-20:
> CONFIG_TINYUSB_TASK_STACK_SIZE=16384

It's work :)

a1bert01, 2024-12-20:
looks  better ;)
```

set_report 0 0 2
command 6
len 8
SCAN
KEY DEVICE is empty. Generating SECP256R1 curve... done!
SCAN
[fbfc0] scan fid cc00, len 32
[fbdeb] scan fid ce00, len 457
[fbddb] scan fid c000, len 4
[fbdaf] scan fid 1090, len 32
[fbd92] scan fid 1101, len 17
set_report 0 0 2
command 10
len 1
set_report 0 0 2
command 10
len 143
set_report 0 0 2
set_report 0 0 2
set_report 0 0 2
command 11
len 0
```

HeavyDutySoul, 2024-12-20:
Is there any hope that someday will be an eddsa firmware for esp32?

[8 more replies]

commit 2eca0816 (2024-12-11): ESP32-S3 only supports 4 IN endpoints. Fixes #77.
closed 2025-01-15 by polhenarejos