polhenarejos/pico-fido #83: Esp32-s3 n16r8 does not show the pin to be entered FIDO 6

Lasa76, 2024-12-26:
A very good project, but I can't make it work. 
I recently purchased an esp32-s3 n16r8 wroom 1, and followed the procedure outlined on the pico flash page, made via the web. After finishing, the red led flashes and I go to do the test and it would be as if it was not being recognized. I have already checked the card's collection in Windows 10 and it is being recognized, I did the test with uart and usb, they are recognized by Windows, but it is recognized by Google. 
In addition to following the procedure where everything is being done online, would there be another procedure after the flash? 
I just want to use it as a two-factor password, 2fa.
I have little idea of programming, but I know how to follow it step by step.
If anyone can help with each step, I would appreciate it.

polhenarejos, 2024-12-26:
There has been some changes and bug fixes for ESP32 recently. Try cloning the development branch and build a firmware. Take a look on https://github.com/polhenarejos/pico-fido/issues/77#issuecomment-2527307551 and successive replies.

Lasa76, 2024-12-26:
> Houve algumas mudanças e correções de bugs para o ESP32 recentemente. Tente clonar o branch de desenvolvimento e construir um firmware. Dê uma olhada em [#77 (comentário)](https://github.com/polhenarejos/pico-fido/issues/77#issuecomment-2527307551) e respostas sucessivas.

I'll try and report back, thanks!

Lasa76, 2024-12-26:
> Houve algumas mudanças e correções de bugs para o ESP32 recentemente. Tente clonar o ramo de desenvolvimento e construir um firmware. Dê uma olhada em [#77 (comentário)](https://github.com/polhenarejos/pico-fido/issues/77#issuecomment-2527307551) e respostas sucessivas.

I will try and then get back to you, thanks

closed 2025-01-15 by polhenarejos