polhenarejos/pico-fido #85: Wrong ID is set when building Nitro3

m23project, 2025-01-03:
Hi,

when running

```
PICO_SDK_PATH=$(realpath ../../pico-sdk)
cmake .. -DVIDPID=Nitro3
```

it gives out the correct ID

```
-- VIDPID:                       'Nitro3'
...
-- USB VID/PID:                  0x20A0:0x42B2
```

but when I plug in the newly build and flashed uf2 `lsusb` reports

```
Bus 003 Device 020: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
```

PS Previously I had flashed an uf2 Yubikey5 file, but flashed the Nitro3 version 3 times just to be sure.

PPS Thank you for your hint about the Pico-SDK in version 2.0.0. Maybe you could the readme?

polhenarejos, 2025-01-04:
How did you change VIDPID to Yubikey 5?

m23project, 2025-01-04:
I tried to set the VIDPID by Pico Commissioner under Brave, Chromium and Firefox. There is no feedback in the browser that would indicate any change. There is no actual change when I run ``lsusb``.

Compiling an uf2 file for another VIDPID doesn't change the actual VIDPID on the RP2040? Is the VIDPID "burnt in" in some place in the RP2040 that isn't changed by writing a new uf2 file on it?

By "nuke the board" you mean to destroy it? Is there another way to change the VIDPID?

polhenarejos, 2025-01-05:
VIDPID is taken from hardcode (cmake && build) only if there is no commission. If you previously changed the VIDPID via software (pico-***-tool.py or web commission) it has priority in front of hardcode. 

To nuke it you can use [pico-nuke](https://github.com/polhenarejos/pico-nuke/releases), it will erase your flash memory entirely. Then drag & drop your firmware with Nitro3 VIDPID built in it.

closed 2025-01-15 by polhenarejos