polhenarejos/pico-fido #88: Pico Commissioner does nothing?

rayray794, 2025-01-18:
Hey there,

I'm having issues setting up a working "Fido2 Key for Secure Logins" .

At my disposal there is a plugged in **ESP32-S3-Zero** 
As for my OS I'm running a **Linux** distro with a **Chrome Browser** installed,
greeting me with the following pop-up when using the ESP32 Flasher:

[image]

After selecting the "seriual debug unit" from the list and pressing 'Connect',
I am then greeted with the following screens in sequence:

[image]
[image]
[image]
[image]
[image]

The last popup says "Installation complete" , but I when I'm re-checking for any Logs, there seems to be NOTHING.

Also when trying to use the Pico Commissioner with mostly default settings,
after I've selected my device from the list, nothing happens at all.

After a while , the following appears: "Error commissioning: The operation either timed out or was not allowed. See: https://www.w3.org/TR/webauthn-2/#sctn-privacy-considerations-client."

I can't get "Commission via WebAuthn" to work either.
When selecting the 'USB security key' option, 
I can press any combination of the ESP32 buttons but they do nothing, leaving me stuck with this screen:

[image]

Before I used the ESP32 Flasher for the very first time, the LED lights on the ESP32 were also flashing.
Now they have gone off and haven't displayed any colour since.

What can I do to make it work? 

Thanks for your effort looking into this.

polhenarejos, 2025-01-18:
It doesn’t seem detected by the browser. It should appear “Pico Key”. From the first picture I see only ttys appear.

rayray794, 2025-01-26:
I've been trying out various things  , different browsers, connecting, reconnecting, ereasing the device multiple times,
even asking chat-gpt for assistance,  as I know very little about coding, except for "copying and pasting" (haha!)

Today, I came up with the following steps and somehow made it work:
1) Flashed the **Pico HSM Development version** on my recognised "serial debug unit (ttyAMC0)" 
2) Flashed the **Pico Fido Development version** right afterwards
3) I used the "dmesg -w" command, showing me the device recognised as a Pico Key (seen in the screenshot below)

[image]

4) I then went to https://www.picokeys.com/pico-commissioner/ ,
 selected a vendor (didn't touch any other settings), 
-> clicked on "Comission via WebUSB" .

Nothing seemed to be happening for like 3 seconds, so I clicked on "Comission via WebAuthn" straight away,
and voila, for some reason, I got a confirmation message and didn't even have to press any buttons on my device.

I tried out the key immediately on a website I had intended to register on, and it worked out flawlessly.
I still do not understand where I went wrong in the first place,  but all seems to be working now!

Thanks! 

Ps: It's not really a concern for me personally, but despite the device working as a security key now, 
I just wanted to let you know that the LED lights never came back to life.

polhenarejos, 2025-01-31:
Perhaps uses a different GPIO. Which board you use? I'd need the vendor model.

Oakwen, 2025-03-01:
I had the same problem with waveshare_rp2350_zero.
Click "Commission via WebUSB", select PIco Keys device and nothing happens.
"Commission via WebAuthn" works fine, and after about 3 seconds the green "Commissioned successfully!" will appear.

Thanks!

Oakwen, 2025-03-04:
BTW
I just bought a waveshare_rp2040_zero.
I have the same problem.

Oakwen, 2025-03-06:
Pico Commissioner works fine on Chrome browser on openSUSE.
Waveshare_rp2040_zero and waveshare_rp2350_zero work fine. 
It just doesn't work on Chrome browser on Windows 11.

490shigure, 2025-03-14:
> I had the same problem with waveshare_rp2350_zero. Click "Commission via WebUSB", select PIco Keys device and nothing happens. "Commission via WebAuthn" works fine, and after about 3 seconds the green "Commissioned successfully!" will appear.
> 
> Thanks!

same issue. Tested on my and friend's laptops with `waveshare_rp2350_one`.
Environment: 
```
OS: Windows 10 x64 22H2
Browser: Chrome 134.0.6998.89
```
and 
```
OS: Windows 11 24H2
Browser: Edge
```

Steps to reproduce:
- open Pico Commissioner web page.
- custom some configurations.
- plug the device into laptop.
- click `Commission via WebUSB`, select `Pico Keys`
- wait for 1-2 minutes but nothing happens.
- unplug the device

The output from console as the picture below:
1. page loaded
2. plug the device
3. select the device
4. unplug the device

[image]

BTW, I tried to Commission via WebUSB on my android phone. It worked properly and showed "Commissioned successfully!".

polhenarejos, 2025-03-14:
When clicking on WebUSB a window like this should appear:

[image]

If it does not appear it might be a problem of permissions or your browser.

Oakwen, 2025-03-14:
This window will appear. Select the pico device, click "Connect" and nothing will happen.

Raito00, 2025-03-14:
> This window will appear. Select the pico device, click "Connect" and nothing will happen.

I get flashed custom data like this:

1. Edit all Fields
2. Click Commission via WebUSB and select Pico Key

[image]

3. Then trigger flash with - Commission via WebAuthn

[image]

Accept enter PIN and wait ...

Raito00, 2025-03-14:
> Actually if you opt for WebAuthn, there's no need to click on WebUSB. Both are different methods for commissioning.

So then when try use WebUSB nothing happening for me also after select Pico key ...

Oakwen, 2025-03-14:
What is very strange is that WebUSB does not work under Windows, but works fine under openSUSE and Arch Linux.

Raito00, 2025-03-14:
> What is very strange is that WebUSB does not work under Windows, but works fine under openSUSE and Arch Linux.

Can You please also check your Fido key here: https://webauthn.bin.coffee/

There is all ok with: AAGUID:  xxxxxxxxxxxxxxxx  ???

Oakwen, 2025-03-14:
@Raito00 

[image]

You are right.

polhenarejos, 2025-03-17:
To get the AAGUID you need to use Direct attestation. It is not shown in coffee.bin but it works in webauthn.io.

intensecity, 2025-06-13:
Just reaffirming that the WebUSB method does not work in Windows. Currently working around this by using Android instead.
Cheers