polhenarejos/pico-fido #89: "Yubikey core error: Timeout" with HMAC challenge-response

samsell, 2025-01-18:
Hi! This is kind of two issues, but both cause the same problem. 

When I try to add a HMAC challenge-response slot using ykpersonalize, although it says "successful", there is still nothing programmed to the slot. If I take `-oserial-api-visible` out of the command, it fails with `Yubikey core error: write error`. 

`ykpersonalize -v -2 -ochal-resp -ochal-hmac -ohmac-lt64 -oserial-api-visible -ochal-btn-trig` fails to actually change the otp slots: 
[image]

Using `yubikey-personalization-gui` results in the same thing, although it reads successful write. My udev rules are all updated correctly. It may be worth noting that trying to write with this used to fail:
[image]
[image]

However this wasn't a massive issue, as I could still write it with
`ykman otp chalresp -t -g 2`, which would work fine, and can be interacted with through `ykman otp calculate`: 
[image]

But using it as a HMAC key gives `Yubikey core error: Timeout`. I think this may be to do with not having the correct configuration through ykman, because I think that the `-oserial-api-visible` is needed for it to work with this:
[image]

pmnlla, 2025-01-19:
All I've noticed is that the HMAC SHA1 key is hardcoded into the firmware source - and is an int with a value of "1" and a bytelength of KEYLENGTH+2 - no other references to the key in code that I've seen, although I haven't fully looked into it- so I wonder why it worked one way, but not the other. @polhenarejos is invited to correct me, as I think I might be a little wrong here.
Regardless, I'd love to follow along, as I also need HMAC-SHA1.

pmnlla, 2025-01-19:
@qcoral 👀

samsell, 2025-01-19:
Maybe related that `ykman otp delete` seems to always act as if it fails, although it still deleted the slot:
[image]

polhenarejos, 2025-01-19:
Please provide:
- Board you are using
- Version of OS
- Version of client software

(Offtopic: may I ask you the software console/terminal? It looks nice)

samsell, 2025-01-19:
- The board is a raspberry pi pico (just the regular 2040)
- OS is arch linux with 6.6.72 kernel
- Both `yubikey-manager` and `yubikey-personalization` are latest versions
  -   ykman version `1:5.5.1-3`
  -   yubikey-personalization version `1.20.0-4`
  -   yubikey-personalization-gui version `3.1.25-3`
- The version on the pico is `6.2`, from last wednesday

I also now have the error `Yubikey core error: write error` after reinstalling on the pico to double check it was fully updated, although I was also getting this error yesterday after updating then. I no longer remember what I did to fix it, might have just restarted:
[image]

The terminal is foot, with fish running on Hyprland. I didn't do much, it's from [these dotfiles.](https://github.com/end-4/dots-hyprland/tree/main) Definitely very pretty :)

polhenarejos, 2025-01-19:
I fixed the deletion problem but I do not see the problem with chalresp.

First, personalization tools are no longer maintained, so I'll not give support for them since they are discontinued.

If you do:
```
ykman otp chalresp -t -g 2
ykman otp calculate -T -d 6 2 12345678
```
doesn't work?

samsell, 2025-01-20:
`ykman otp calculate` works fine, it's only when used with `ykchalresp`, although I don't know if that is part of `ykman` or `yuibkey-personalize`:
[image]

samsell, 2025-01-22:
I've adjusted the code I was using it for to use ykman, but it isn't a drop in replacment; it has a much larger file size when packaged as an ELF binary (due to being in python), meaning my initramfs is larger than prefered. It does work properly with ykman however, so I'll close this issue and maybe support for ykchalresp could be added in the future at some point? 

This is an awesome project btw, really interesting to go through!

commit ed9c46de (2025-01-19): Fix slot deletion.

Fixes #89.
closed 2025-01-22 by samsell