polhenarejos/pico-fido #96: Unknown Access Key for Yubikey Slot Emulation

rolandweibel, 2025-01-24:
TL;DR: I need to know the default access key for Yubikey slots

I've commissioned an RP2350 with the latest 6.2 firmware version as a Yubikey Neo with secure boot and secure lock. When opening the Yubikey Authenticator App, there are two slots that can be configured. However, changing their configuration requires an access key. I have tried: all zeros, 0-12 (hex), and left and right padded serial no. with zeros. I have also searched Google and noted information about a log file being generated when commissioned, but I don't see anything.

Two questions: is this functionality supported? I'm trying to use the OTP functionality on short press of one slot. If so, what is the access code I should use?

polhenarejos, 2025-01-24:
Can you put an screenshot? What are you trying to configure the slot for?

rolandweibel, 2025-01-25:
I'm attaching a few images, as I don't know what will be most helpful

[image]
_serial no. and hardware_

I then click Slots >> Slot 1 >> Yubico OTP and fill out the required info:

[image]
_yubico otp configuration_

Upon clicking save, it asks for an access code and all combinations above show the same result:

[image]
_"Computer says no"_

HTH

metabo7000, 2025-01-29:
> Csatolok pár képet, mert nem tudom mi lenne a leghasznosabb
> 
> [image] _sorozatszám és hardver_
> 
> Ezután rákattintok a Slots >> Slot 1 >> Yubico OTP lehetőségre, és kitöltöm a szükséges adatokat:
> 
> [image] _yubico otp konfiguráció_
> 
> A Mentés gombra kattintva hozzáférési kódot kér, és az összes fenti kombináció ugyanazt az eredményt mutatja:
> 
> [image] _"A számítógép nemet mond"_
> 
> HTH

we get an access code window, we start deleting the challenge-response, it asks for 12 characters, what is this?! If we create a new challenge-response, the access code window also repeats itself!
default access code?

https://drive.google.com/file/d/1FvJ-1ahbVRSyBkh-pRgdTIajSPjsQsHX/view?usp=drive_link

polhenarejos, 2025-01-29:
I cannot reproduce it. I tested it in macOS and Windows. 

Can you try with ykman? `ykman otp delete 1`

metabo7000, 2025-01-29:
> I cannot reproduce it. I tested it in macOS and Windows.
> 
> Can you try with ykman? `ykman -l TRAFFIC otp delete 1`

cannot create otp,challenge-response because it throws up an access code window in which any password of 12 length is invalid WIN/ UBUNTU no works!

polhenarejos, 2025-01-29:
Then provide the log of `ykman -l TRAFFIC otp chalresp 1 -T -g -f`

metabo7000, 2025-01-29:
Yubikey core error: write error! v6.2

`
host@host-virtual-machine:~$ ykpersonalize -2 -ochal-resp -ochal-hmac -ohmac-lt64 -a 303132333435363738393a3b3c3d3e3f40414243
Firmware version 6.2.0 Touch level 14 Program sequence 1

Configuration data to be written to key configuration 2:

fixed: m:
uid: n/a
key: h:303132333435363738393a3b3c3d3e3f40414243
acc_code: h:000000000000
OATH IMF: h:0
ticket_flags: CHAL_RESP
config_flags: CHAL_HMAC|HMAC_LT64
extended_flags: 

Commit? (y/n) [n]: y
Yubikey core error: write error

`

metabo7000, 2025-01-29:
v5.8 no problem!

ost@host-virtual-machine:~$ ykpersonalize -2 -ochal-resp -ochal-hmac -ohmac-lt64 -a 303132333435363738393a3b3c3d3e3f40414243
Firmware version 5.8.0 Touch level 12 Program sequence 1

Configuration data to be written to key configuration 2:

fixed: m:
uid: n/a
key: h:303132333435363738393a3b3c3d3e3f40414243
acc_code: h:000000000000
OATH IMF: h:0
ticket_flags: CHAL_RESP
config_flags: CHAL_HMAC|HMAC_LT64
extended_flags: 

Commit? (y/n) [n]: y
host@host-virtual-machine:~$ ykchalresp -2 'Sample #2'
be2a3a6e2ef0b6b1bb4e21a4f6ec21d49c50c7c2

polhenarejos, 2025-01-29:
This is not the command I adviced you.

metabo7000, 2025-01-29:
> This is not the command I adviced you.

I understand what you are doing ykman but none of the programs see the key I create KeePassXC,yubikey login v6.2 firmware and it is there!
ykman otp static --generate 2 --length 38

polhenarejos, 2025-01-31:
There was a problem with OTP and Linux, as it uses the PCSC interface instead of the OTP one.

PD: use Yubikey 5 VIDPID, otherwise it will appear as two devices (NEO and 5A) but fully functional.

closed 2025-01-31 by polhenarejos

metabo7000, 2025-01-31:
You can see in the video Yubico Authenticator y5 5A I use 1050:0407 usb setting firmware all versions v6.2 I tried what you uploaded the developer pcbn 16m
-pico_fido_archi-6.2.uf2
-pico_fido_archi-6.2 Nightly.uf2
-pico_fido_pico-6.2.uf2
-pico_fido_pico-6.2_ Nightly.uf2
I also tried win11 but the otp, hmac functions are missing the access code window problem can't get past this (USB device inaccessible error)!

https://drive.google.com/file/d/1FvJ-1ahbVRSyBkh-pRgdTIajSPjsQsHX/view?usp=drive_link

polhenarejos, 2025-01-31:
Nightly build will be available tomorrow morning

metabo7000, 2025-01-31:
> Az Nightly build holnap reggel lesz elérhető

I just tried what you posted in the dev version and it works fine now, hmac otp 2 storage bin codes and all functions are visible in the YubiKey Personalization Tool! Thank you for the full usability in v6.2! :)

commit 353d7829 (2025-01-31): Fix OTP command issues in Linux.

Fixes #96.