polhenarejos/pico-fido #100: Do you support ESP32S2？

afterow, 2025-01-27:
Hello, does this project support ESP32S2

polhenarejos, 2025-01-27:
Not at this moment.

afterow, 2025-02-22:
I flashed the Pico Fido with version 6.4, but it doesn't work properly, keeps plugging and ejecting

[image]

polhenarejos, 2025-02-22:
It crashes on boot.

Which model of S2 do you use?

afterow, 2025-02-23:
My equipment is esp32-s2 mini
```
C:\Users\afterow\Desktop\esptool-s3>esptool --chip esp32s2 --port COM3 read_mac
esptool.py v4.8.1
Serial port COM3
Connecting.....
Chip is ESP32-S2FNR2 (revision v1.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: d8:3b:da:c9:7c:e8
Uploading stub...
Running stub...
Stub running...
MAC: d8:3b:da:c9:7c:e8
Error: ESP32-S2FNR2 (revision v1.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_res
```

polhenarejos, 2025-02-23:
Do you have the chance to debug on UART0?

afterow, 2025-02-23:
I don't know what UART0 is, but I can help you, for example through remote desktop

polhenarejos, 2025-02-24:
It is the serial port for debugging. It requires an additional hardware like a USB-to-TTL.

afterow, 2025-02-24:
I have a CH340B, I wonder if it can be used.

polhenarejos, 2025-02-24:
Not sure how it works but if you know how to open a serial port it might work. Look for uart0 tx and rx pins in the datasheet of your board and probe them with your ch340b.

emf37, 2025-03-07:
> Not sure how it works but if you know how to open a serial port it might work. Look for uart0 tx and rx pins in the datasheet of your board and probe them with your ch340b.

hello, I have same issue. the Pico-FIDO with ESP32-S2FH4 just keeps plugging and ejecting on windows 10. 
Here is a log file exported from the console on "www.picokeys.com/esp32-flasher",also I flashed by this.
Hope this helps. 
Thank you.
 
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/19123057/esp-web-tools-logs.txt)

polhenarejos, 2025-03-07:
Thanks, very useful log. I see the problem: ESP32-S2 is single core and Pico Fido is designed to work with two cores at least. I was not aware of it. I'll try to leverage FreeRTOS for core management but I do not have an ESP32-S2 to test it.

Do you have skills for building ESP32-S2 images?

afterow, 2025-03-07:
I will compile it myself

---- Replied Message ----
| From | Pol ***@***.***> |
| Date | 03/07/2025 16:41 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [polhenarejos/pico-fido] Do you support ESP32S2？ (Issue #100) |

Thanks, very useful log. I see the problem: ESP32-S2 is single core and Pico Fido is designed to work with two cores at least. I was not aware of it. I'll try to leverage FreeRTOS for core management but I do not have an ESP32-S2 to test it.

Do you have skills for building ESP32-S2 images?

—

Thanks, very useful log. I see the problem: ESP32-S2 is single core and Pico Fido is designed to work with two cores at least. I was not aware of it. I'll try to leverage FreeRTOS for core management but I do not have an ESP32-S2 to test it.

Do you have skills for building ESP32-S2 images?

afterow, 2025-03-07:
I will compile it myself

polhenarejos, 2025-03-07:
I pushed a fix for esp32-s2 in the development branch. You can build yourself or wait until tomorrow for the nightly development build.

79812b, 2025-03-17:
> I pushed a fix for esp32-s2 in the development branch. You can build yourself or wait until tomorrow for the nightly development build.

btw will this work for esp32 c3 speed xiao

polhenarejos, 2025-03-17:
No, C3 does not have USB OTG stack.

afterow, 2025-03-17:
> No, C3 does not have USB OTG stack.

I tested the latest development version on the ESP32-S2 today, and the issue remains unresolved. I have decided to give up on using this device since the price of the ESP32-S3 is currently the same as that of the ESP32-S2 here in China. 

Fortunately, the project [esp32_u2f](https://github.com/jocover/esp32_u2f) can be used on the ESP32-S2.

commit 529a12e7 (2025-03-07): Only pin to core in ESP32-S3 since it is multicore.

Fixes #100
closed 2025-04-10 by polhenarejos