polhenarejos/pico-fido #103: Code base for esp32s3

kingsmen732, 2025-01-30:
# Codebase for esp32S3 
- I wanted to have the esp codebase for esp32 architecture so i can try some experiments on top of it .

## Plans to integrate Fingerprint stack 
- essentially for user presence in biometric
- makes more safer 2 factor authentication

polhenarejos, 2025-01-31:
Don't understand this issue.

kingsmen732, 2025-02-01:
this is not a issue , we are not able to label this as question. So coming back to the point, where do we have the esp32s3 codebase which is being flashed currently through the web. 
I was planning to increase some functionality available in the esp32 like adding up userpresence using  fingerprint module which is fido compliant.

ihavetenfingers, 2025-02-04:
> this is not a issue , we are not able to label this as question. So coming back to the point, where do we have the esp32s3 codebase which is being flashed currently through the web. I was planning [...]

It is in the same source files as everything else if you read it.. not separated from the 2040 code.

Here: https://github.com/polhenarejos/pico-fido

kingsmen732, 2025-02-10:
> It is in the same source files as everything else if you read it.. not separated from the 2040 code.
>
> Here: https://github.com/polhenarejos/pico-fido

So for rpi2040 the flashing method is different but for the esp instance there is only web flashing methodology. I wanted to do offline flashing of esp by my serial. Did you do that methodology ?, cause i want to seek for debugging dev tools to listen for board signals. I need the code separate for it so i will be able to help contribute BIO-FIDO on it.
@polhenarejos @ihavetenfingers