polhenarejos/pico-fido #105: ESP32-S3-Zero LED pin on GPIO21

a1bert01, 2025-02-02:
looks like esp32 build has hardcoded LED PIN (GPIO48), but  ESP32-S3-Zero LED is on GPIO21, LED PIN written by pico commissioner is ignored (for esp32?)

```
--- pico-keys-sdk/src/led/led_neopixel.c~	2024-12-18 19:13:51.000000000 +0100
+++ pico-keys-sdk/src/led/led_neopixel.c	2025-02-02 18:06:13.657589657 +0100
@@ -36,7 +36,7 @@
 };
 
 void led_driver_init() {
-    uint8_t gpio = GPIO_NUM_48;
+    uint8_t gpio = GPIO_NUM_21;
     if (phy_data.led_gpio_present) {
         gpio = phy_data.led_gpio;
     }

```

polhenarejos, 2025-02-02:
This is not harcoded as it can be oberwritten by the commissioned value.

closed 2025-02-02 by polhenarejos

a1bert01, 2025-02-02:
as I wrote: "LED PIN written by pico commissioner is ignored" at least for me, I can change the brightness but not the GPIO PIN

dolence, 2025-03-08:
I can confirm this issue. Just got an ESP32-S3-ZERO and set the pin to 21 and as result the led doesn't work at all. WS2812 are used on this tiny devices.

polhenarejos, 2025-03-08:
It has been fixed in the development branch.