polhenarejos/pico-fido #106: How to enable secure boot with custom key & signing

lyc8503, 2025-02-09:
Hi, Thanks for your great project! It's working well as a open-source alternative for yubikey.

There's a small problem though: I'm making some small customization on the source code (e.g. customize LED behaviour and add support for Waveshare RP2350 which hasn't been added), but I can't find a way to enable secure boot & secure lock with custom boot keys. 

I also didn't find source code for pico commissioner which I can modify.

I found some tools/script from rpi which can enable secure boot: https://github.com/raspberrypi/rp2350_hacking_challenge, but I'm worried about bricking my RP2350 so I haven't tried. Are there any specific docs on this topic?

polhenarejos, 2025-02-09:
It will not brick it but tie to a specific key. Your board will run only firmwares signed with a specific key. 
Challenge scripts are a good starting point. The rest is at rp2350 datasheet.

lyc8503, 2025-02-09:
As I understand it, in order to get the self-signed picokey firmware working, I have to do these steps:

1. Generate my own key pair
2. Sign the firmware with CMakeList.txt commands
```
pico_sign_binary(project_name ${CMAKE_CURRENT_SOURCE_DIR}/ec_private_key.pem)
pico_hash_binary(project_name)
pico_set_otp_key_output_file(project_name ${CMAKE_CURRENT_LIST_DIR}/otp.json)
```
3. Enable secure boot and lock the chip with these commands
```bash
picotool otp load otp.json
# Disable debugging features
picotool otp set OTP_DATA_CRIT1.DEBUG_DISABLE 1 
# Disable other boot keys
picotool otp set OTP_DATA_BOOT_FLAGS1.KEY_INVALID 0xe
# Enable glitch detector
picotool otp set OTP_DATA_CRIT1.GLITCH_DETECTOR_ENABLE 1
# Highest sensitivity
picotool otp set OTP_DATA_CRIT1.GLITCH_DETECTOR_SENS 3

# Lock writes to PAGE1
picotool otp set --raw OTP_DATA_PAGE1_LOCK1 0x101010
picotool otp set --raw OTP_DATA_PAGE2_LOCK1 0x101010
```

Is there anything wrong or missing? Should I manually set `OTP_DATA_PAGE1_LOCK1`/`OTP_DATA_PAGE2_LOCK1` here?
I guess the firmware will initialize and take care of the master key in OTP so I don't have to do extra setups here.

polhenarejos, 2025-02-09:
Yep

ViZiD, 2025-03-28:
@lyc8503
Hi, how to obtain otp.json file?

lyc8503, 2025-03-28:
> [@lyc8503](https://github.com/lyc8503) Hi, how to obtain otp.json file?

cmake with `pico_set_otp_key_output_file` should generate that file