polhenarejos/pico-fido #107: Error while editing user/name of TOTP in Yubico Authenticator

staticle, 2025-02-12:
[image]

polhenarejos, 2025-02-12:
I need more info: OS, key version, board, etc. With just a snapshot I can do nothing.

staticle, 2025-02-13:
> I need more info: OS, key version, board, etc. With just a snapshot I can do nothing.

Sorry for incomplete description
Board - Raspberry Pi PICO 2 (RP2350)
OS - Screenshot Windows 11, (Tested on Mac OS and android but same issue)
Key Version - 6.2 EDSA  (with pico patcher set to Yubikey 4/5)

please let me know if you need any additional info, thanks

polhenarejos, 2025-02-17:
Interesting. They are using an undocumented command and this is why it returns "unknown command".

commit 250de29c (2025-02-17): Added support for OATH rename.

Fixes #107.
closed 2025-02-19 by polhenarejos