polhenarejos/pico-fido #108: SSH resident key creation doesn't work on Windows: requested feature not supported

lyc8503, 2025-02-12:
I am trying to create a resident SSH key, but it failed with error message `Key enrollment failed: requested feature not supported`:
```
$ ssh -V
OpenSSH_9.9p1, OpenSSL 3.2.3 3 Sep 2024

$ ssh-keygen -vvv -t ed25519-sk -O resident
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=1273
debug3: ssh_msg_send: type 5 len 50
debug3: ssh_msg_send: done
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x21, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device windows://hello
debug1: check_sk_options: option uv is on
debug1: key_lookup: fido_dev_get_assert: FIDO_ERR_UNSUPPORTED_OPTION
debug1: ssh_sk_enroll: key_lookup failed
debug1: sshsk_enroll: provider "internal" failure -2
debug1: ssh-sk-helper: Enrollment failed: requested feature not supported
debug1: main: reply len 8
debug3: ssh_msg_send: type 5 len 8
debug3: ssh_msg_send: done
debug1: client_converse: helper returned error -59
debug3: reap_helper: pid=1273
Key enrollment failed: requested feature not supported
```

I am using the official `pico_fido_pico2-6.2-eddsa1.uf2` firmware from GitHub Release with `VID:PID` set to Yubikey 4/5, My OS is `Windows 10 LTSC 21H2 19044.5371`

It works well with non-resident SSH ED25519 keys:
```bash
$ ssh-keygen -vvv -t ed25519-sk
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=1251
debug3: ssh_msg_send: type 5 len 50
debug3: ssh_msg_send: done
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x01, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device windows://hello
debug1: ssh_sk_enroll: self-attested credential
debug1: ssh_sk_enroll: authdata len=255
debug1: main: reply len 611
debug3: ssh_msg_send: type 5 len 611
debug3: ssh_msg_send: done
debug3: reap_helper: pid=1251
Enter file in which to save the key (/c/Users/lyc8503/.ssh/id_ed25519_sk): /dev/null
/dev/null already exists.
Overwrite (y/n)? y
Enter passphrase for "/dev/null" (empty for no passphrase):
Enter same passphrase again:
Your identification has been saved in /dev/null
Your public key has been saved in /dev/null.pub
The key fingerprint is:
SHA256:tm4/jCkrPMASY7/Fbc0+i5VimeQrhtw4Thi4yoXVGpU lyc8503@Laptop
The key's randomart image is:
+[ED25519-SK 256]-+
|                 |
|      .          |
|     E           |
|+.  o            |
|o=.o....S        |
|..B.ooo+o+.      |
|.+.** .*o*       |
|o o*++o.Ooo      |
|....ooo*.o+.     |
+----[SHA256]-----+
```

polhenarejos, 2025-02-12:
Are you using the eddsa branch?

lyc8503, 2025-02-12:
> Are you using the eddsa branch?

I am using the firmware with `eddsa` in their name, so I guess they're from the eddsa branch.

Also I can generate a non-resident `ed25519-sk` key without any problems, only have problems with `-O resident` parameter.

polhenarejos, 2025-02-12:
Do Secp256r1 resident key work though?

lyc8503, 2025-02-12:
> Do Secp256r1 resident key work though?

Seems not, logs attached:

```bash
$ ssh-keygen -t ecdsa-sk -O resident -vvv
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=1375
debug3: ssh_msg_send: type 5 len 50
debug3: ssh_msg_send: done
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x21, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device windows://hello
debug1: check_sk_options: option uv is on
debug1: key_lookup: fido_dev_get_assert: FIDO_ERR_UNSUPPORTED_OPTION
debug1: ssh_sk_enroll: key_lookup failed
debug1: sshsk_enroll: provider "internal" failure -2
debug1: ssh-sk-helper: Enrollment failed: requested feature not supported
debug1: main: reply len 8
debug3: ssh_msg_send: type 5 len 8
debug3: ssh_msg_send: done
debug1: client_converse: helper returned error -59
debug3: reap_helper: pid=1375
Key enrollment failed: requested feature not supported
```

and non-resident `ecdsa-sk` works just fine like `ed25519-sk`

lyc8503, 2025-02-12:
There's some progress: when I tried the same command again in powershell (which uses the default OpenSSH installation `OpenSSH_for_Windows_9.5p1, LibreSSL 3.8.2` comes from Windows instead of the one comes from git-for-windows `OpenSSH_9.9p1, OpenSSL 3.2.3 3 Sep 2024`), it successfully created the key:

```
PS C:\Users\lyc8503> ssh-keygen -t ed25519-sk -O resident -vvv
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug1: find_helper: using "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as helper
debug3: Creating process with CREATE_NO_WINDOW
debug3: spawning "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as subprocess
debug3: start_helper: started pid=6660
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: client_converse: helper returned error -44
debug3: reap_helper: pid=6660
A resident key scoped to 'ssh:' with user id 'null' already exists.
Overwrite key in token (y/n)? y
You may need to touch your authenticator again to authorize key generation.
debug1: find_helper: using "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as helper
debug3: Creating process with CREATE_NO_WINDOW
debug3: spawning "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as subprocess
debug3: start_helper: started pid=19196
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug3: reap_helper: pid=19196
Enter file in which to save the key (C:\Users\lyc8503/.ssh/id_ed25519_sk):
C:\Users\lyc8503/.ssh/id_ed25519_sk already exists.
Overwrite (y/n)? y
Enter passphrase (empty for no passphrase):
Enter same passphrase again:
Your identification has been saved in C:\Users\lyc8503/.ssh/id_ed25519_sk
Your public key has been saved in C:\Users\lyc8503/.ssh/id_ed25519_sk.pub
The key fingerprint is:
SHA256:Ar93HpN1aAu+GANujDxp/qvCrB9LIvDYRUlCW25XFmo lyc8503@Laptop
The key's randomart image is:
+[ED25519-SK 256]-+
| .o o   +.       |
|   * . +         |
|  . * E          |
|   o =       .   |
|.   . + S . + .  |
|.+ o = + . = o   |
|+o= * = + * .    |
|.o+= o . * +     |
|.oo.ooo.. o      |
+----[SHA256]-----+
PS C:\Users\lyc8503> ssh -V
OpenSSH_for_Windows_9.5p1, LibreSSL 3.8.2
```

However, when I tries to read the key, there's another error `Unable to load resident keys: invalid format` happening:
```
PS C:\Users\lyc8503> ssh-keygen -K -vvv
Enter PIN for authenticator:
You may need to touch your authenticator to authorize key download.
debug1: find_helper: using "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as helper
debug3: Creating process with CREATE_NO_WINDOW
debug3: spawning "C:\\Windows\\System32\\OpenSSH\\ssh-sk-helper.exe" as subprocess
debug3: start_helper: started pid=18604
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=18604
Unable to load resident keys: invalid format
```

lyc8503, 2025-02-18:
> Are you in an Admin Console? It requires admin privileges.

Oh, that's exactly the problem. The error message is not so intuitive. Thanks for your help, it now works great in an admin powershell.

closed 2025-02-18 by lyc8503