polhenarejos/pico-fido #109: Pico Fido does not work on android webauthn.io

staticle, 2025-02-13:
OS - Android 14
Key Version - 6.2
Hardware - Raspberry Pi Pico 2 RP2350

Error - An Unknown error occured while talking to credentials manager

ImMALWARE, 2025-03-10:
Same for me, Android doesn't even ask for a password

polhenarejos, 2025-03-10:
Did you update to nightly development build?

ImMALWARE, 2025-03-10:
@polhenarejos, right now I have updated to nightly development build. Raspberry Pi Pico 2. Using Arch Linux, web auth in Firefox works without any problems, Yubico Authenticator also works just fine, it detects Passkeys after I enter password:
[image]

When I connect Pico 2 to Android device and open Yubico Authenticator, it also detects Yubikey 5A. But at the Passkeys tab it says "Insert or tap a YubiKey". When I press BOOTSEL, nothing happens.

When I try to Authenticate on webauthn.io via chrome, it tells me to connect the key and press the button. When I press the button, it says "An Unknown error occured while talking to credentials manager". It doesn't even ask me for a password.
Tried on POCO X5 Pro 5G and on Galaxy Tab S6 Lite, same result.

ImMALWARE, 2025-03-10:
@polhenarejos, yes, it is connected via usb. I guess this is the error with key:
```
03-10 20:36:09.318  5019  5180 I WindowManager: wms.showSurfaceRobustly mWin:Window{9e3abb7 u0 com.google.android.gms/com.google.android.gms.fido.fido2.ui.Fido2FullScreenActivity}
03-10 20:36:09.318  1722  4871 D MI-SF   : [FpsStat] construct...
03-10 20:36:09.320  5019  5180 V ActivityTaskManager: wms.updateAllDrawn allDrawn: ActivityRecord{9af8156 u0 com.google.android.gms/.fido.fido2.ui.Fido2FullScreenActivity t461} interesting=2 drawn=2
03-10 20:36:09.325  1722  1722 W Layer   : gatherBufferInfo: Received unknown dataspace from gralloc
03-10 20:36:09.330  5793  6135 D SoScUtils: monitorTransitionFinished transition:android.os.BinderProxy@4574293 transaction: android.view.SurfaceControl$Transaction@c6128a6 mRequestCloseTransition:null mRequestOpenTransition:null mPairsOpenTransition:null mPairsOpenTransitionNull:null mRequestOpenTriggerTask:-1 mRequestCloseTriggerTask:-1 mOrientationTransition: null mMainStartingTask:-1 mSideStartingTask:-1
03-10 20:36:09.330  5793  6135 V WindowManagerShell: Transition animation finished (aborted=false), notifying core (#47)android.os.BinderProxy@4574293@0
03-10 20:36:09.332  5019  5158 V WindowManager: Finish Transition #47: created at 03-10 20:36:09.086 collect-started=0.095ms request-sent=9.767ms started=11.14ms ready=67.445ms sent=134.37ms finished=245.648ms
03-10 20:36:09.334  1722  1722 W Layer   : gatherBufferInfo: Received unknown dataspace from gralloc
03-10 20:36:09.334  5793  6135 V WindowManagerShell: Track 0 became idle
03-10 20:36:09.334  5793  6135 V WindowManagerShell: All active transition animations finished
03-10 20:36:09.337 17099 17099 I HandWritingStubImpl: refreshLastKeyboardType: 1
03-10 20:36:09.337 17099 17099 I HandWritingStubImpl: getCurrentKeyboardType: 1
03-10 20:36:09.338  5019 13735 I ImeTracker: com.google.android.gms:9d057d0b: onRequestHide at ORIGIN_SERVER_HIDE_INPUT reason HIDE_UNSPECIFIED_WINDOW
03-10 20:36:09.338  5019 13735 I ImeTracker: com.google.android.gms:9d057d0b: onCancelled at PHASE_SERVER_SHOULD_HIDE
03-10 20:36:09.339 10723 12172 D MirrorManager:  sendAcceptInput 0 0
03-10 20:36:09.340  1722  4871 D MI-SF   : updateScene: module = 270 (Input), value = 0, pkg = com.google.android.gms, Vrr = 1
03-10 20:36:09.340  1722  4871 D RefreshRateSelector: Touch Boost
03-10 20:36:09.340  1722  4871 D MI-SF   : mIsSupportAutomodeForMaxFpsSetting is 0(0), getCurrentBrightness() = 199(328), mIsKeepAutoMode 0
03-10 20:36:09.340  1722  4871 D RefreshRateSelector: fixedMode_group: 0, isFindModeGroup 0
03-10 20:36:09.340  1722  4871 D RefreshRateSelector: find max refresh rate:120, mode Group:0
03-10 20:36:09.340  1722  4871 D MI-SF   : updateScene mSetInputFlag status 0
03-10 20:36:09.341  1722  1722 W Layer   : gatherBufferInfo: Received unknown dataspace from gralloc
03-10 20:36:09.342  7152  7152 I GoogleInputMethodService: GoogleInputMethodService.onFinishInput():2589 
03-10 20:36:09.343  7152  7152 I Module  : DeviceLockedStatusModuleProvider$Module.updateDeviceLockedStatus():100 repeatCheckTimes = 0, locked = false
03-10 20:36:09.343  7152  7152 I GoogleInputMethodService: GoogleInputMethodService.onStartInput():1568 onStartInput(EditorInfo{EditorInfo{packageName=com.google.android.gms, inputType=0, inputTypeString=NULL, enableLearning=false, autoCorrection=false, autoComplete=false, imeOptions=0, privateImeOptions=null, actionName=UNSPECIFIED, actionLabel=null, initialSelStart=-1, initialSelEnd=-1, initialCapsMode=0, label=null, fieldId=0, fieldName=null, extras=Bundle[mParcelledData.dataSize=72], hintText=null, hintLocales=[]}}, false)
03-10 20:36:09.344  5019 13735 I WindowManager: cancelAnimation removed leash : Surface(name=Surface(name=4a9ca20 InputMethod#162)/@0x51f907f - animation-leash of insets_animation#582)/@0x8092894
03-10 20:36:09.344  7152  7152 I Module  : DeviceLockedStatusModuleProvider$Module.updateDeviceLockedStatus():100 repeatCheckTimes = 1, locked = false
03-10 20:36:09.344  1722  4871 D MI-SF   : [FpsStat] construct...
03-10 20:36:09.344  5019 14307 W PackageConfigPersister: App-specific configuration not found for packageName: com.google.android.gms and userId: 0
03-10 20:36:09.345  7152  7152 I AndroidIME: InputBundleManager.loadActiveInputBundleId():407 loadActiveInputBundleId: und-Latn-x-password, password
03-10 20:36:09.349  1722  1722 D MI-SF   : [FpsStat] distroy...
03-10 20:36:09.350  1722  1722 D RefreshRateSelector: Touch Boost
03-10 20:36:09.350  1722  1722 D MI-SF   : mIsSupportAutomodeForMaxFpsSetting is 0(0), getCurrentBrightness() = 199(328), mIsKeepAutoMode 0
03-10 20:36:09.350  1722  1722 D RefreshRateSelector: fixedMode_group: 0, isFindModeGroup 0
03-10 20:36:09.350  1722  1722 D RefreshRateSelector: find max refresh rate:120, mode Group:0
03-10 20:36:09.353 17099 17137 I Fido    : [Ctap2AuthenticatorGetInfoDataOperation] Executing Ctap2AuthenticatorGetInfoDataOperation [CONTEXT service_id=269 ]
03-10 20:36:09.353 17099 17137 I Fido    : [UsbChannel] Sending CTAP2 command over USB channel: Ctap2Command{type=AUTHENTICATOR_GET_INFO, cborMap=null}
03-10 20:36:09.357  1722  1722 D MI-SF   : [FpsStat] distroy...
03-10 20:36:09.425 17099 17159 I Fido    : [AuthenticatorGetInfoResponseData] Unrecognized key present in response map: 21
03-10 20:36:09.425 17099 17159 I Fido    : [AuthenticatorGetInfoResponseData] Response map for AuthenticatorGetInfo contains an invalid extensions string "thirdPartyPayment"
[... cut ...]
03-10 20:36:09.435  1573  3813 I android.hardware.usb@1.2-service-qti: usb interface does not support autosuspend /sys/devices/platform/soc/a600000.ssusb/a600000.dwc3/xhci-hcd.1.auto/usb2/2-1
03-10 20:36:09.435 17099 17134 D UsbDeviceConnectionJNI: close
03-10 20:36:09.497  5019 13735 D FileUtils: Rounded bytes from 7539814400 to 8000000000
03-10 20:36:09.498  5019 13735 D FileUtils: Rounded bytes from 7539814400 to 8000000000
03-10 20:36:09.498  5019 13735 D FileUtils: Rounded bytes from 7539814400 to 8000000000
03-10 20:36:09.505 17099 17134 I Fido    : [Ctap1RequestController] Submitting U2f request. [CONTEXT service_id=274 ]
```

polhenarejos, 2025-03-10:
Android does not support ES256K algorithm (same like bitcoin) and throws an error, despite it is described in the specs. I do not know if it can be solved through a sw update.

ImMALWARE, 2025-03-11:
@polhenarejos, so that means that pico-fido is uncompatible with android?

polhenarejos, 2025-03-11:
Not really, should work disabling ES256K. I’ll push a fix for this.

ImMALWARE, 2025-03-11:
> Not really, should work disabling ES256K. I’ll push a fix for this.

Okay, I will wait for new nightly development build and try it

ImMALWARE, 2025-03-12:
Now authentication in Firefox on Android works fine. Yubico Authenticator on Android still doesn't work

> When I connect Pico 2 to Android device and open Yubico Authenticator, it also detects Yubikey 5A. But at the Passkeys tab it says "Insert or tap a YubiKey". When I press BOOTSEL, nothing happens.

But i kinda dont care about this. Thank you so much for your work! @polhenarejos

staticle, 2025-03-19:
@polhenarejos also in yubico authenticator android i see different serial number than on yubico authenticator win 11

polhenarejos, 2025-03-19:
Please use tomorrow’s nigthly development build and put a webauthn.io snapshot.

staticle, 2025-03-19:
@polhenarejos my bad i forgot to nuke the board its working on authn.io on android,  but on yubico authenticator android i see different serial number than on yubico authenticator win 11.

But on yubico authenticator android app  > Passkeys it says
Insert your Yubikey

commit 6069cf94 (2025-04-10): ES256K1 is disabled by default for compatibility. It can be enabled via Pico Commissioner.

Fixes #109.
closed 2025-04-10 by polhenarejos