polhenarejos/pico-fido #112: can't back up full content with picotool since v6.2

lifetyper, 2025-02-20:
Hi,
I used to back up my rp2040 keys with picotool by pico save -a dump.bin and pico load dump.bin command.

But this won't work after upgrading to v6.2~v6.4

Steps to reproduce the issue:

1. 1.Prepare two rp2040 key(Key A and Key B) and clear the spi flash with flash_nuke.uf2 from raspberrypi.
2. 2.program Key A with firmware version between v6.2~v6.4
3. 3.setup the device PIN under windows sign-in option.
4. 4.boot Key A in bootsel mode and dump the content with `picotool save -a dump.bin` command
5. 5.boot Key B in bootset mode and load the dumped binary with `picotool load dump.bin` command
6. 6.insert Key B and open windows sign-in option,the firmware works and there is a device PIN set 
7. 7.but when you try to change the device PIN using PIN set in Key A,it will tell you the PIN is incorrect.
8. 8.reproduce step 1~7 using firmware v6.0,everything works fine and PIN is duplicated to Key B.

Seems like the picotool does dumped and loaded the spi flash content as the firmware is working,but why the credentials not duplicated.
Is this a issue or just designed to prevent backup the spi flash conent?

Thanks!

lifetyper, 2025-02-20:
After roll back and test some commits, I found this issue(or design?) as introduced in commit https://github.com/polhenarejos/pico-fido/commit/6a678000570ccf95dbc67446a0fe4fdddfc5f28e

polhenarejos, 2025-02-20:
I don’t understand the problem. If Key A is th donor, it is irrelevant if you dump the flash. Dumping does not modify the flash, so the PIN is untouched. v6.2 introduced a feature to protect the content via the PIN. If you change the PIN in A, perhaps is something broken with this step. It is strange btw since it passess all tests, including change PIN.

lifetyper, 2025-02-20:
> I don’t understand the problem. If Key A is th donor, it is irrelevant if you dump the flash. Dumping does not modify the flash, so the PIN is untouched. v6.2 introduced a feature to protect the [...]

I just did one more test here:

1.dump flash content from Key A and save it as dump.bin
2.use flash_nuke.uf2 to erase both Key A and Key B
3.picoload dump.bin to Key A and Key B

The PIN keeps the same for Key A,but changed to something I don't know for Key B.

The only different I can see is the RP2040  chip may have different CHIP ID or something like that.

Maybe the encrypt & decrypt process use some info from that which lead to different decrypt result  on different chip?

lifetyper, 2025-02-22:
To simplify my question:is the PIN itself encrypted?
If  yes,then is the encrypt process  use any information from the RP2040 chip?

polhenarejos, 2025-02-22:
PIN is used for encryption; it’s never encrypted but hashed. 
For silent authentication the board’s serial is used, but this is a feature introduced in 6.4 and it does not affect the PIN. 
In the previous tests, how do you know that PIN has changed? What is the command you are testing that fails? Which is the error code?

lifetyper, 2025-02-22:
> PIN is used for encryption; it’s never encrypted but hashed. For silent authentication the board’s serial is used, but this is a feature introduced in 6.4 and it does not affect the PIN. In the previous tests, how do you know that PIN has changed? What is the command you are testing that fails? Which is the error code?
> 
> Also: are both boards equal? You cannot load a dumped content to another board that has less flash size.
> 
> Also 2: paste the command you use to dump the contents.

I use windows hello (windows sign-in option) setup feature to mange the key,there is a change PIN function.After entering the old PIN and new PIN to set,it returned a PIN error message.

[image]

In the mean time,I tried to dumped ssh key with "ssh-keygen -K" command,it will also ask for PIN,it returned a msg "Unable to load resident keys: invalid format". The command used to work properly if the PIN is correct.

polhenarejos, 2025-02-22:
Let me add a recommendation for backing up. The proper way to backup is not cloning the key, but registering a secondary key as a backup. That is, you have two independent working keys. You register both keys in all the sites you want to backup. Usually you'll use the primary one, but you have a secondary key to login as a backup in case your primary fails. Then, once you logged in with the secondary, you can register a new key as a replacement of the failing.
This is the recommendation of FIDO Alliance.

lifetyper, 2025-02-22:
Yes, that's exactly how I think it works, and how it should work.
 The problem lies in the old firmware, not in the new version. 
Thank you for your explanation.

closed 2025-02-22 by lifetyper