polhenarejos/pico-fido #114: Security Key Not Recognized on Websites Without PIN Prompt Until Used on a PIN-Prompting Website

lifetyper, 2025-03-07:
I have registered several security keys successfully on both Google and Porkbun. However, I encountered an issue where a newly inserted security key is not recognized on websites that do not prompt for a PIN (such as Porkbun), but works fine on websites that do prompt for a PIN (such as Google).

### Steps to Reproduce

1.     Insert the security key into the computer.
2.     Try to log in to Porkbun, which does not prompt for a PIN.
3.     Windows Security displays the following error message:   "This security key doesn't look familiar, please try a different one."

[image]

5.     Use the same security key to log in to Google, which prompts for a PIN.
6.     Enter the PIN, and the login to Google succeeds.

[image]

7.     Return to Porkbun and try logging in again using the same security key.
8.     This time, the security key is recognized, and the login proceeds normally.
[image]

### Expected Behavior

    The security key should work consistently on all registered websites without requiring a prior authentication on a PIN-prompting website.

### Actual Behavior

    When first inserted, the security key is not recognized on websites that do not prompt for a PIN.
    After being used once on a website that prompts for a PIN (e.g., Google), it starts working correctly on all websites, including those without PIN prompts.

### Additional Information

    The security key was successfully registered on both Google and Porkbun.
    This issue occurs on a Windows system where Windows Security manages authentication.
    It seems that the security key remains in an uninitialized state until a PIN is entered at least once after insertion.

lifetyper, 2025-03-07:
I also read [this issue](https://github.com/polhenarejos/pico-fido/issues/110) and [your reply](https://github.com/polhenarejos/pico-fido/issues/110#issuecomment-2663707348).
Is this correct behavior if the website doesn't prompts for a PIN?Or I must used it in a website prompts PIN then it can be used other place?
From that issue and my experience,seems both github and porkbun don't prompt PIN and lead to the key not recognized.

lifetyper, 2025-03-07:
ok,I just build and install the latest develop branch,now porkbun prompts for PIN also and it works well.

closed 2025-03-07 by lifetyper

cmichel5, 2025-08-22:
I have the same exact issue with 1password.com
Using firmware 6.6 (not EdDSA).

Error message when inserting the key : "This security key doesn't look familiar, please try a different one."
I have to use either Yubico Authenticator, or any websites that ask for my PIN first, and then the key will work on 1password