polhenarejos/pico-fido #117: How to set AAGUID

Raito00, 2025-03-13:
Hello!
I flashed my ESP32 S3 with web ESP32 Flasher - www.picokeys.com/esp32-flasher/

but right now there is: AAGUID: 00000000-0000-0000-0000-000000000000

[image]

It is possible set some random AAGUID if just using WEB esp32-flasher and Pico Commissioner on Windows PC ?

polhenarejos, 2025-03-13:
AAGUID is set at https://github.com/polhenarejos/pico-fido/blob/eb857df3e14732976b1b0607c52d107491d7ae7a/src/fido/cbor.c#L47

If you see all zeros might because of your browser.

Raito00, 2025-03-14:
Tested browsers: Chrome, Edge, Mozilla - the same problem
also other webauthn test website
webauthn.bin.coffee/

[image]

Authentication working all OK just AAGUID problem

polhenarejos, 2025-03-14:
This is how it works.
When Attestation is set to `None`, AAGUID is replaced by all zeros (more info https://www.w3.org/TR/webauthn-2/#sctn-createCredential).
If you want to see the AAGUID, go to webauthn.io and select Attestation -> Direct.

closed 2025-03-14 by polhenarejos