polhenarejos/pico-fido #120: Version 6.4 Fails to Take Effect in pico-commissioner

comrade6660, 2025-03-16:
First of all, thank you! Your project is truly excellent!

I've tried using Version 6.4 of pico-commissioner, but the configuration fails to take effect. Commission via WebAuthn cannot be debugged successfully, while Nightly Development functions without issue.

On March 12th, 2025, Version 6.4 worked fine, but I suspect this problem is similar to issue [.](https://github.com/polhenarejos/pico-fido/issues/118)

Also, I have a question: if I need to simulate a Yubikey 5 on the rp2025-one, how can I confirm that I've enabled Secure Boot and Secure Lock? Or are they automatically enabled upon successful flashing? Since pico-commissioner isn't working recently, I'm quite concerned about this.

Raito00, 2025-03-16:
Check Previous issue: https://github.com/polhenarejos/pico-fido/issues/118

polhenarejos, 2025-03-17:
Yes, you are right. Latest enhancements in the development branch broke Pico Commissioner for stable version. I'd recommend you to use the development nightly build. It seems pretty stable and works with the Commissioner.

I hope to release a new version on Wed.

comrade6660, 2025-03-17:
Thank you for your reply!

Both Nightly Stable and Version 6.4 exhibit the same issue. Nightly Development appears to be functioning normally. I look forward to subsequent releases.

However, I've recently encountered a problem using pico-commissioner. After flashing the firmware and enabling Secure Boot and Secure Lock, should debugging be selected as shown in the image, or should debugging be initiated after rp2350 enters boot, also as shown in the image?

[image]

I have tried both methods, but there is no response after entering boot. Furthermore, "Commission via WebUSB" fails to debug successfully using Edge browser on win11_24H2. Debugging is successful in the Edge browser on Android, but there is no response in my Ubuntu 22.4 virtual machine. Please look into this situation!

polhenarejos, 2025-03-17:
What do you mean by "debugging"? Pico Commissioner is for commissioning Pico Keys, not for debugging.

comrade6660, 2025-03-17:
Apologies, my description was incorrect, and I have corrected it above.

What I meant to convey is that upon clicking "Commission via WebUSB", Android indicates success, while both the Edge browser on win11 and the Firefox browser on Ubuntu 22.04 show absolutely no response.

Furthermore, this bug prevents me from confirming whether Secure Boot and Secure Lock have been successfully enabled. I'd like to know how to detect or check if Secure Boot and Secure Lock are active.

polhenarejos, 2025-03-17:
Firefox does not support WebUSB. Edge should work since 80.

My recommendation is using a Chromium-based browser.

comrade6660, 2025-03-19:
Currently, I've tried using Edge or Chrome browsers on an Android phone, and it works.

Docteh, 2025-05-19:
> It seems USB drivers must be replaced with Zadig, start edge with webUSB

Ya, just tried setting this up, with Chrome 136 and Edge 136, and the commissioning page didn't give any errors at all, but only got anywhere after I found the comment here to try zadig to assign a driver. I think it should be mentioned somewhere. 

Can I assign a different vendor ID/product name after enrolling a passkey with a website like github?

polhenarejos, 2025-05-19:
Which driver you assigned? 
WebAuthn commissioning should work without zadig though. 

I didn’t try it, but it should work as the credential is not related with vidpid.

closed 2026-02-24 by GN998