polhenarejos/pico-fido #123: Challenge-Response settings are not working properly

IsayIsee, 2025-03-18:
Waveshare 2350 one, version 6.4, the challenge response setting requires touch, but in actual use it passes without touching. Flashing pico2 version 6.4 also has the same problem

metabo7000, 2025-03-18:
It never worked under rp2040 either! I'm deaf to this error!
[https://github.com/polhenarejos/pico-fido/issues/104](url)

IsayIsee, 2025-03-19:
Tested on KeePassXC, thanks for the quick fix

a15355447898a, 2025-03-19:
I also encountered this problem.
I want to ask, if I re-flash a new firmware, will the previously saved keys be cleared?

polhenarejos, 2025-03-19:
If you reflash with the same firmware, nothing will change. Keys are preserved until the board is nuked.

metabo7000, 2025-03-19:
> If you reflash with the same firmware, nothing will change. Keys are preserved until the board is nuked.

Slot1 and slot2 have content and you turn it on so that it fits in. When you touch it, nothing happens, only the challenge response, so this has not been fixed yet!

polhenarejos, 2025-03-19:
> I tried the new firmware, but KeePassXC on Arch Linux still can't find the hardware key. However, KeePassXC on Windows can find it. Where did I go wrong?

Did you change the Product Name?

a15355447898a, 2025-03-20:
> > I tried the new firmware, but KeePassXC on Arch Linux still can't find the hardware key. However, KeePassXC on Windows can find it. Where did I go wrong?
> 
> Did you change the Product Name?

I set the Product Name to yubico yubikey.

Keepassxc on Windows recognizes it normally.

[image]

Yubico YubiKey can also be seen in Device Manager.

[image]

polhenarejos, 2025-03-20:
Use latest nightly development build. Yesterday I pushed a fix.

a15355447898a, 2025-03-20:
> Use latest nightly development build. Yesterday I pushed a fix.

Problem solved, thanks for the development work!

metabo7000, 2025-03-20:
> > Use latest nightly development build. Yesterday I pushed a fix.
> 
> Problem solved, thanks for the development work!

If you turn on Challenge-Response, touch does not work in slot1 and slot2 swap mode, there are still bugs in the development version!

IsayIsee, 2025-03-31:
There is another problem. When the static password is set, setting and deleting the password for the slot does not affect the setting. However, whether you set the challenge first and then the password, or set the password first, set the challenge and then delete the password, it will cause verification to fail.

polhenarejos, 2025-03-31:
I'm not getting the point. Can you describe step by step the problem to reproduce it? The better with ykman.

IsayIsee, 2025-04-01:
To put it simply, as long as the otp is set to challenge the password attribute of the modified slot, the challenge data will be damaged and the challenge verification will fail.
For example, after setting the challenge, setting the password through the ykman otp settings -A 576bc54882f3 2 command will cause the challenge verification to fail.

commit 94f8d5f6 (2025-03-19): Add support for Require Touch in ChalResp OTP slots.

Fixes #123 #104
commit 37d7d7fa (2025-03-19): OTP can flow through FIDO interface as a report type 3.

Fixes #123.