polhenarejos/pico-fido #124: ykman application compatibility issues

IsayIsee, 2025-03-19:
Here's a professionally worded issue report in English:

## Waveshare 2350 First Generation Hardware, Firmware Version: Development [cb99b8f]

### Anomalous Behavior:

1. **OTP Slot Swapping Issue**:  
   When executing `ykman otp swap` command, the following error is returned:
   ```
   ERROR: Failed to write to the YubiKey. Make sure the device does not have restricted access (see "ykman otp --help" for more info).
   ```
   However, the slot swap operation has actually been successfully executed, despite the system failing to correctly acknowledge the operation status.

2. **OTP Settings Configuration Issue**:  
   When executing `ykman otp settings -A 576bc54882f3 2` command, the following error is returned:
   ```
   ERROR: Failed to write to the YubiKey. Make sure the device does not have restricted access (see "ykman otp --help" for more info).
   ```
   Despite the error message, the password was successfully set, but the original data in slot 2 has been corrupted.

Command execution status inconsistent with actual results, potentially leading to device management confusion and data integrity issues.

IsayIsee, 2025-03-19:
During testing, I confirmed that HOTP and Static Password functions work correctly after setting passwords.

Prior to setting a password for Challenge-Response, I could successfully create and decrypt a new database using KeePassXC. However, after setting a password for slot 2, I am no longer able to decrypt the database

Additionally, I specifically configured only slot 2 for Challenge-Response mode. Yet when setting up KeePassXC, both slots appear as available options. In reality, slot 1 is using Static Password mode.

IsayIsee, 2025-03-19:
If I set a password for slot 2 and then execute the swap command, slot 2 adopts the previous settings of slot 1. However, slot 1 does not inherit the previous settings of slot 2 - instead, it becomes empty. The password does not provide the expected protection.

polhenarejos, 2025-03-19:
There's a problem with swap from 1 -> 2. After making the swap, it appears as empty. Therefore, if you program slot 2 again, it puts slot 2 twice and, from the filesystem perspective, is like you are having two slots 2. First delete every slot and reset after deletion. If multiple files are stored in that slot, the slot will appear as configured again after the reset. Do it until appears as empty after reset.
In the meantime I am looking for the fix, reset your board after slot each config/update/swap/deletion to force the filesystem to reload.

IsayIsee, 2025-03-31:
i flash [0a2ee65](https://github.com/polhenarejos/pico-fido/commit/0a2ee6523fbd97877e57b09c07c5ec3ee82ac114), swap slot is ok. but #123 has reappeared

polhenarejos, 2025-04-07:
What is exactly broken? I cannot reproduce it.

IsayIsee, 2025-04-07:
[image]

[image]

but now use [c8dbc21](https://github.com/polhenarejos/pico-fido/commit/c8dbc213a0ca5fe78a5f9fc225afa12a7916d200) it ok

commit 49c0179c (2025-03-19): Fix swap files.

When a dynamic file is deleted, all scoped references to other dynamic files are invalidated.

Fixes #124
commit eacb8a04 (2025-03-19): Increase config_seq on swap and update.

Fixes #124.