polhenarejos/pico-fido #126: There is a problem adding a pass key to mac.

tianjiaoshanzai, 2025-03-20:
[image]

The Chrome browser can be used to log in to GitHub normally, but adding two-factor authentication on Mac fails.

I'm using pico and pico2 hardware, and have tried 6.4, nightly stable and development. Also, Android doesn't work.

manobendro, 2025-05-01:
### This is console log message
SecKeyVerifySignature failed: Error Domain=NSOSStatusErrorDomain Code=-67808 "RSA signature verification failed, no match" UserInfo={numberOfErrorsDeep=0, NSDescription=RSA signature verification failed, no match}

[image]

polhenarejos, 2025-05-01:
It seems a problem with device attestation, probably due to only accepted by few vendors.