polhenarejos/pico-fido #127: Challange response mismatch compared with genuine yubikey

d-tischler, 2025-03-23:
Hi,
I configured a pico-fido (rp2040-w, pico_fido_pico_w-6.4.uf2 nightly developer 23.3. ) for challenge response using the gui yubico authenticator. I did the same with a genuine Yubikey 5 NFC using the exact same secret. Unfortunately, the result does not seem to be the same. To investigate further, I tried the following:

ykman otp calculate 1 1122334455667788

The results for the pico-fido and the yubikey do not match.

I also tried to change the last character of the secret but the response of the above command did not change at all. Only when I changed the secret more rigorously, the response to the ever same challenge changed.

Am I missing something? 
Any help is appreciated.

polhenarejos, 2025-03-23:
Please put the key you are using and the outputs of both boards for comparison. Also all the commands you use.

d-tischler, 2025-03-23:
So I redid the tests and this is what I got. Each time I programmed slot 1 using the yubico authenticator on ubuntu hitting "challenge-response, Program a challenge-response credential" (Actually, I am not sure, how to get the same effect on the command line):

1. 

Secret:
`b3cd9bc5afa573fc2ea2b7baa995af99fef9b2b4`

yubikey:
```
>ykman otp calculate 1 1122334455667788
884701c5995a388f2832246fdee22196943825bf
```
pico-fido:
```
>ykman otp calculate 1 1122334455667788
8f58a52e81d9065bd80e6b6cfbf5d9d30d990737
```
So the both responses differ!?

2.
Then, I changed the last letter of the secret from 4 to 5:
`b3cd9bc5afa573fc2ea2b7baa995af99fef9b2b5`

yubikey:
```
>ykman otp calculate 1 1122334455667788
f1b9a46e4d778e49c07a5bcbb2947c99a1a74903
```
pico-fido:
```
>ykman otp calculate 1 1122334455667788
8f58a52e81d9065bd80e6b6cfbf5d9d30d990737
```
For the yubikey, this also changed the response completely. With the pico-fido, I get the same result as with the original secret.

3.
Last experiment: secret shifted:
`00b3cd9bc5afa573fc2ea2b7baa995af99fef9b2`

yubikey:
```
>ykman otp calculate 1 1122334455667788
690a9cbe0043e02b316cfa48e6ef1bafcc912e89
```
pico-fido:
```
>ykman otp calculate 1 1122334455667788
7600183467d7491f26515f765c9e5bff712cdec8
```
Now the new secret has an effect on both responses, yet, they do not match.

Hope this helps. I'm still not sure, whether I am using it right. Especially if the secret is sensible.

polhenarejos, 2025-03-23:
Yes, it’s really helpful, thanks. How do you change the secret? Do you overwrite the slot directly or first you reset the board?

d-tischler, 2025-03-23:
I just had it overwrite the slot. So I repeated step 1 and 2 from above like this:
1. flashed "flash_nuke.u2f"
2. flashed pico fido developer nightly from today
3. used the pico-commissioner with "initialize" and "yubikey4/5"
4. programmed the secret for slot1 using the yubico authenticator

same result as above.

polhenarejos, 2025-03-23:
I pushed a fix in 751fcf0.

Could test TOTP and HOTP with your Yubikey and Pico Fido to see whether both return same results? Use the new development firmware published tonight.

d-tischler, 2025-03-24:
Ha, I managed to build your very cool project! 
And challenge response now works as expected. I tried all three above secrets and always got the same result as from the original yubikey. Also tested unlocking my keepass with pico-fido: works like a charm.
Thank you very much for your fast response. This was fun, working together so quickly.

commit 751fcf05 (2025-03-23): Fix HMAC-SHA1 calculation.

Fixes #127.
closed 2025-04-10 by polhenarejos