polhenarejos/pico-fido #129: Firefox Pico keys error

TheMaxMur, 2025-03-23:
Devices for test:
1. Raspberry pi pico2 (RP2350). Firmware: pico_fido_Yubikey5-pico2-6.4.uf2
2. Flipper zero (STM32WB55). Firmware: flipper-z-f7-full-1.2.0.dfu

Sites for testing:
1. https://webauthn.bin.coffee/
2. https://webauthn.me/
3. https://webauthn.io/

Testing browsers:
1. Librewolf 136.0.1-1 (NixOS 25.05.20250318.b6eaf97)
2. Chromium 134.0.6998.88 (NixOS 25.05.20250318.b6eaf97)
3. Firefox 131.0.2 (Fedora 41)
4. Google Chrome 136.0.7086 (Fedora 41)

Device 1:
| Testing site | Browser 1 | Browser 2 | Browser 3 | Browser 4 | 
| --- | --- | --- | --- | --- |
| 1 | Error 1| Success | Error 1 | Success |
| 2 | Error 1| Success | Error 1 | Success |
| 3 | Error 1| Success | Error 1 | Success |

Device 2:
| Testing site | Browser 1 | Browser 2 | Browser 3 | Browser 4 | 
| --- | --- | --- | --- | --- |
| 1 | Success | Success | Success | Success |
| 2 | Success | Success | Success | Success |
| 3 | Success | Success | Success | Success |

Errors:
1. NotAllowedError: The request is not allowed by the user agent or the platform in the current context, possibly because the user denied permission.

TheMaxMur, 2025-03-23:
P.S. I can register the key in (for example) github in chromium and login with passkey in firefox, but i cannot use it for (for example) add email in gh. But if i setup `security.webauthn.ctap2 = false` in `about:config` i can add email and cannot login.

shizzgar, 2025-03-23:
Same here. 
I have some troubles with Firefox in two diffrent environments with my pico2. 

Raspberry pi pico2 (RP2350). Firmware: development-[37d7d7f](dev-https://github.com/polhenarejos/pico-fido/commit/37d7d7faeb0709a9ddd849cffac3867fc0203c2b)

Tested **env**ironments:

1. Firefox (ubuntu 24.04.2) 
2. Firefox (KDE neon 6.1)
3. Firefox (windows 10) 

On **env 1** I was able to use my pico only for authentication after registration my key using chrome. 
Key registration was not completed successfuly.

On **env 2** FF throws the same error, that @TheMaxMur has mentioned.

On **env 3** authentication completed successfully. Didn't have time test the Key registration.

ViZiD, 2025-03-24:
I have exactly the same problems, doesn't work on Librewolf and Firefox, but works fine on Chromium and Google Chrome Firmware version https://github.com/polhenarejos/pico-fido/releases/tag/v6.4-eddsa1

IsayIsee, 2025-03-24:
i flash devlopment fef46dc 3 site both Success, but vaultwarden wetauth can not select pico key device

TheMaxMur, 2025-03-24:
> i flash devlopment fef46dc 3 site on firefox both Success, but vaultwarden webauth can not select pico key device

U can temporary disable vaultwarden window for register passkey in settings

IsayIsee, 2025-03-25:
> > i flash devlopment [fef46dc](https://github.com/polhenarejos/pico-fido/commit/fef46dc1c53ae3d85f89cafe6b383ba142ff90b1) 3 site on firefox both Success, but vaultwarden webauth can not select pico key device
> 
> U can temporary disable vaultwarden window for register passkey in settings

i try in https://www.passkeys-debugger.io but also no picl key

[image]

polhenarejos, 2025-03-25:
I don't know what to say. I tested many times FF in macOS and Windows 11 and works all times with nothing special. I don't know what is different from flipper.

shizzgar, 2025-03-25:
> I don't know what to say. I tested many times FF in macOS and Windows 11 and works all times with nothing special. I don't know what is different from flipper.

Could you advise something on how can we debug the firefox-usb connection?

TheMaxMur, 2025-03-25:
find `RUST_LOG=authenticator=debug firefox`

```
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::transport::platform::device] new device "/dev/hidraw3"
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] Device "/dev/hidraw3" continues with the register process
[INFO  authenticator::ctap2] PIN Error that requires user interaction detected. Sending it back and waiting for a reply
[WARN  authenticator::ctap2] error happened: Error: Error issuing command: CommandError: Error while parsing: trailing data at offset 443
[INFO  authenticator::statemachine] Statemachine was cancelled. Cancelling transaction now.
[INFO  authenticator::transport::platform::transaction] Transaction was cancelled.
```

shizzgar, 2025-03-25:
I found this [site](https://webauthn.firstyear.id.au/compat_test), that makes some compatibility tests

Here is my results for chrome:
[image]

```json
{"direct_attest_1":{"Passed":{"rs":{"cred_id":"8dACAu-az0QhXaqscotIDjvdxYpSqY8RVE_tcbfJvqm5e2n7H3_QdbwOsnjKuxHKfXJ3C4yqkwyzd67Bw9HTWBEIOUa7Vdk7M3TXhaJUD9yCKwBA_FW-Z8u1K3wfLPKs6AvsoP6Kw8KiF_DJgVl-5K-kwUdqXgtTUAs3uGvl1JkciIMOJEOkdNK5J1UNSkoravRkWNjexqd35xSh7PzXSLe72-Ed91Abbyu6ujxT4Ok","uv":true,"alg":"ES256","extensions":{"cred_protect":"NotRequested","hmac_create_secret":"NotRequested","appid":"NotRequested","cred_props":"NotRequested"}},"ccr":{"publicKey":{"rp":{"name":"webauthn.firstyear.id.au","id":"webauthn.firstyear.id.au"},"user":{"id":"ELeUuFAoQhOcADMNKoqzog","name":"compatuser","displayName":"compatuser"},"challenge":"6Tnhv6Q-EUuUyNFqjCW21z2E2D6YoWfzlMTDJyv7dyY","pubKeyCredParams":[{"type":"public-key","alg":-7},{"type":"public-key","alg":-257}],"timeout":300000,"authenticatorSelection":{"residentKey":"discouraged","requireResidentKey":false,"userVerification":"discouraged"},"attestation":"direct","extensions":{}}},"rpkc":{"id":"8dACAu-az0QhXaqscotIDjvdxYpSqY8RVE_tcbfJvqm5e2n7H3_QdbwOsnjKuxHKfXJ3C4yqkwyzd67Bw9HTWBEIOUa7Vdk7M3TXhaJUD9yCKwBA_FW-Z8u1K3wfLPKs6AvsoP6Kw8KiF_DJgVl-5K-kwUdqXgtTUAs3uGvl1JkciIMOJEOkdNK5J1UNSkoravRkWNjexqd35xSh7PzXSLe72-Ed91Abbyu6ujxT4Ok","rawId":"8dACAu-az0QhXaqscotIDjvdxYpSqY8RVE_tcbfJvqm5e2n7H3_QdbwOsnjKuxHKfXJ3C4yqkwyzd67Bw9HTWBEIOUa7Vdk7M3TXhaJUD9yCKwBA_FW-Z8u1K3wfLPKs6AvsoP6Kw8KiF_DJgVl-5K-kwUdqXgtTUAs3uGvl1JkciIMOJEOkdNK5J1UNSkoravRkWNjexqd35xSh7PzXSLe72-Ed91Abbyu6ujxT4Ok","response":{"attestationObject":"o2NmbXRmcGFja2VkZ2F0dFN0bXSiY2FsZyZjc2lnWEcwRQIgaBOTuNqZnfZ5iGPT5KQIxNxO8ZLsQz4-tsNIWjc9dfMCIQDuAYQDopZHikeO7mot9-W0BDRy5IsbVb7xbJEb3SF3wmhhdXRoRGF0YVkBNGq5u_Dfmhb5Hbszu7Ey-vnRfHgsSCbG7HDs7ljZfvUqRQAAAMeJ-5S3Bsk2c5t-MFJtloFFALDx0AIC75rPRCFdqqxyi0gOO93FilKpjxFUT-1xt8m-qbl7afsff9B1vA6yeMq7Ecp9cncLjKqTDLN3rsHD0dNYEQg5RrtV2TszdNeFolQP3IIrAED8Vb5ny7UrfB8s8qzoC-yg_orDwqIX8MmBWX7kr6TBR2peC1NQCze4a-XUmRyIgw4kQ6R00rknVQ1KSitq9GRY2N7Gp3fnFKHs_NdIt7vb4R33UBtvK7q6PFPg6aUBAgMmIAEhWCCmXIvz0S4viYoRjjG39G4XSWpyIURPuIMIIr9YI32rMCJYIJmTayAVY6BksigKaT-OxuuZLcMoQQwmyHk34Fm-GTUT","clientDataJSON":"eyJ0eXBlIjoid2ViYXV0aG4uY3JlYXRlIiwiY2hhbGxlbmdlIjoiNlRuaHY2US1FVXVVeU5GcWpDVzIxejJFMkQ2WW9XZnpsTVRESnl2N2R5WSIsIm9yaWdpbiI6Imh0dHBzOi8vd2ViYXV0aG4uZmlyc3R5ZWFyLmlkLmF1IiwiY3Jvc3NPcmlnaW4iOmZhbHNlfQ"},"type":"public-key","extensions":{}}}},"indirect_attest_1":{"Passed":{"rs":{"cred_id":"Lpgv8sZr96_Yx7XLJFpt8JxZ6_eHeTyi9s7o3bsaCrU7KL-JseGfDOetihEb-x5qdhLBzIcg5vLlX7X3T9pldw","uv":false,"alg":"ES256","extensions":{"cred_protect":"NotRequested","hmac_create_secret":"NotRequested","appid":"NotRequested","cred_props":"NotRequested"}},"ccr":{"publicKey":{"rp":{"name":"webauthn.firstyear.id.au","id":"webauthn.firstyear.id.au"},"user":{"id":"ndbPmj9cTO2yQaLRvj8FwQ","name":"compatuser","displayName":"compatuser"},"challenge":"UyoK-Y6Pd2HUxwWlsJxX19T0c9SBjEkKktR4UA4ewn8","pubKeyCredParams":[{"type":"public-key","alg":-7},{"type":"public-key","alg":-257}],"timeout":300000,"authenticatorSelection":{"residentKey":"discouraged","requireResidentKey":false,"userVerification":"discouraged"},"attestation":"indirect","extensions":{}}},"rpkc":{"id":"Lpgv8sZr96_Yx7XLJFpt8JxZ6_eHeTyi9s7o3bsaCrU7KL-JseGfDOetihEb-x5qdhLBzIcg5vLlX7X3T9pldw","rawId":"Lpgv8sZr96_Yx7XLJFpt8JxZ6_eHeTyi9s7o3bsaCrU7KL-JseGfDOetihEb-x5qdhLBzIcg5vLlX7X3T9pldw","response":{"attestationObject":"o2NmbXRoZmlkby11MmZnYXR0U3RtdKJjc2lnWEYwRAIgIS6VikKZdRnRRyswpXuHscXQ-5B35rm9aug8az_3sjECIFlKAEp-B88We7FyZnnlB5u0FvnUHQEcfnwnsjQua9E3Y3g1Y4FZAcgwggHEMIIBa6ADAgECAhAcS3ppkePRLsqA_CwXeMD1MAoGCCqGSM49BAMCMDQxCzAJBgNVBAYTAkVTMREwDwYDVQQKDAhQaWNvIEhTTTESMBAGA1UEAwwJUGljbyBGSURPMCAXDTIyMDkwMTAwMDAwMFoYDzIwNzIwODMxMjM1OTU5WjA0MQswCQYDVQQGEwJFUzERMA8GA1UECgwIUGljbyBIU00xEjAQBgNVBAMMCVBpY28gRklETzBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABIJajTIBoAOSIqgqJCENOleEy0gP6q-RdkkX-HQ1avQHYB99dWcT-GT22NzPlnTefv1gKQyPaBDp3WVOzCXTELOjXTBbMAkGA1UdEwQCMAAwHQYDVR0OBBYEFPaRCZdh-_58vxHLoWrCps8wEZh_MB8GA1UdIwQYMBaAFPaRCZdh-_58vxHLoWrCps8wEZh_MA4GA1UdDwEB_wQEAwIChDAKBggqhkjOPQQDAgNHADBEAiBpXRIatikgBHKucx40cjUp6L5jc0p9dz1Wib1FhKudqQIgJ49qtTuLCdPDgU0mZifrwRaROq_fI7XaD0To9EUwLG9oYXV0aERhdGFYxGq5u_Dfmhb5Hbszu7Ey-vnRfHgsSCbG7HDs7ljZfvUqQQAAAAAAAAAAAAAAAAAAAAAAAAAAAEAumC_yxmv3r9jHtcskWm3wnFnr94d5PKL2zujduxoKtTsov4mx4Z8M562KERv7Hmp2EsHMhyDm8uVftfdP2mV3pQECAyYgASFYILdSr0jgq4N4EXxVo2Uc3Xygvcp-j_8ORKNBLeWBbEjKIlggpD1_MAEdEng_APaaoSukaxyV8ZFJ40AcPmeKn_R2bB8","clientDataJSON":"eyJ0eXBlIjoid2ViYXV0aG4uY3JlYXRlIiwiY2hhbGxlbmdlIjoiVXlvSy1ZNlBkMkhVeHdXbHNKeFgxOVQwYzlTQmpFa0trdFI0VUE0ZXduOCIsIm9yaWdpbiI6Imh0dHBzOi8vd2ViYXV0aG4uZmlyc3R5ZWFyLmlkLmF1IiwiY3Jvc3NPcmlnaW4iOmZhbHNlfQ"},"type":"public-key","extensions":{}}}},"none_attest_1":{"Passed":{"rs":{"cred_id":"maNJ_8mwaoHS2MXELYIvwx2L8ZTVGM7ceH2QtCpqB5UHlJlKddHASCuWeNuR-K1ji6nOWPItCyfhako1hww3bA","uv":false,"alg":"ES256","extensions":{"cred_protect":"NotRequested","hmac_create_secret":"NotRequested","appid":"NotRequested","cred_props":"NotRequested"}},"ccr":{"publicKey":{"rp":{"name":"webauthn.firstyear.id.au","id":"webauthn.firstyear.id.au"},"user":{"id":"RoD8BTQUSci5-OrqfvJIUg","name":"compatuser","displayName":"compatuser"},"challenge":"-2loqwhZYu7wIQxMN3t4jL9W-1BjZjr4Tpm_PyQZIAk","pubKeyCredParams":[{"type":"public-key","alg":-7},{"type":"public-key","alg":-35},{"type":"public-key","alg":-36},{"type":"public-key","alg":-257},{"type":"public-key","alg":-258},{"type":"public-key","alg":-259},{"type" [... cut ...] 5T4dS33Xsucy5Op10D6k","pubKeyCredParams":[{"type":"public-key","alg":-7},{"type":"public-key","alg":-257}],"timeout":300000,"authenticatorSelection":{"residentKey":"discouraged","requireResidentKey":false,"userVerification":"required"},"attestation":"none","extensions":{}}}}},"authrequired":"FailedPrerequisite","extn_uvm_supported":"FailedPrerequisite","extn_credprotect_supported":"FailedPrerequisite","extn_hmacsecret_supported":"FailedPrerequisite"}
```

This is linux environment, KDE Neon 6.1.

polhenarejos, 2025-03-26:
What is strange to me is this line:

```
[WARN  authenticator::ctap2] error happened: Error: Error issuing command: CommandError: Error while parsing: trailing data at offset 443
```
I debugged in Ubuntu and this offset corresponds to the end of the response, so I guess it seems it doesn't like trailing 0s. If I do not trail with 0 then nothing works (it breaks the specs).

shizzgar, 2025-03-26:
> What is strange to me is this line:
>
> ```
> [WARN  authenticator::ctap2] error happened: Error: Error issuing command: CommandError: Error while parsing: trailing data at offset 443
> ```
> I debugged in [...]

Is such behavior related to https://github.com/mozilla/authenticator-rs?
Could we help with something?

polhenarejos, 2025-03-26:
Not sure, I'm trying to find the string "trailing data at offset" but doesn't appear in authenticator-rs. Perhaps it's more related with FF and what is returned by authenticator-rs. I'd need to find where this error is raised.

TheMaxMur, 2025-03-26:
I found [this](https://github.com/mozilla/authenticator-rs/blob/9410d650378540fd60ec998cdcac17a9c5b4a888/src/ctap2/commands/mod.rs#L461) in authenticator-rs repo, and try found [call this](https://github.com/search?q=repo%3Amozilla%2Fauthenticator-rs+CommandError%3A%3ADeserializing&type=code) in repo. I think the promblem in somewhere near in the function call `from_slice`.

polhenarejos, 2025-03-26:
Yes, it's from the deserializer but it belongs to serde repo, and "Trailing data at offset" doesn't appear there.

TheMaxMur, 2025-03-26:
I found [this](https://github.com/pyfisch/cbor/blob/v0.11.2/src/error.rs#L310) in repo. And try found [call this](https://github.com/search?q=repo%3Apyfisch%2Fcbor%20TrailingData&type=code).

TheMaxMur, 2025-03-27:
@polhenarejos what do you think? How to fix this issue?

polhenarejos, 2025-03-28:
I don't know where this trailing data comes from, not from Pico Keys, for sure. Strangely, FF is not working in Linux only; in macOS and Windows works well, so it makes difficultier spotting the bug.

TheMaxMur, 2025-03-28:
> I don't know where this trailing data comes from, not from Pico Keys, for sure. Strangely, FF is not working in Linux only; in macOS and Windows works well, so it makes difficultier spotting the bug.

I think this is due to the fact that windows/macos uses the OS API to handle the security key

[image]

There is no such API on Linux, so authenticator-rs is used

I have started an issue in the authenticator-rs repository mozilla/authenticator-rs#346, can you help us with it? Maybe something will come to light.

TheMaxMur, 2025-03-28:
We tested Firefox (136.0.2) together with YubiKey 5C (Firmware version: 5.7.1). Firefox did not generate any errors.

```
RUST_LOG=authenticator=debug firefox
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::transport::platform::device] new device "/dev/hidraw9"
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] error happened with device: Error: requested operation is not available on device
[INFO  authenticator::statemachine] Device "/dev/hidraw9" continues with the register process
[INFO  authenticator::ctap2] PIN Error that requires user interaction detected. Sending it back and waiting for a reply
[INFO  authenticator::statemachine] Statemachine was cancelled. Cancelling transaction now.
[INFO  authenticator::transport::platform::transaction] Transaction was cancelled.
```

```
Contacting token... please perform your verification gesture (e.g., touch it, or plug it in)

Note: Raw response in console.

:: "None" Attestation Format ::
[PASS] Calculated RP ID hash must match what the browser derived.: pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs == pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs
[PASS] User presence and Attestation Object must both be set: 65 == 65
[PASS] Credential ID from CBOR and Raw ID match: b69d6131d31e4fb0f3b6ae0a16a71a91c7da9f16b156e94dc8ae7f84e344813402b4bf8cd6d5f984d50ab55ab355917f6ce6d17acbbedcec51c168e322412cc0 == b69d6131d31e4fb0f3b6ae0a16a71a91c7da9f16b156e94dc8ae7f84e344813402b4bf8cd6d5f984d50ab55ab355917f6ce6d17acbbedcec51c168e322412cc0
Keypair Identifier: b69d6131d31e4fb0f3b6ae0a16a71a91c7da9f16b156e94dc8ae7f84e344813402b4bf8cd6d5f984d50ab55ab355917f6ce6d17acbbedcec51c168e322412cc0
Public Key: 04dee67b6c095d2df38c5c20b88b62d4bdfd3f3eb7389834a9ad1378df296ef0523f6d6d6e3eac793b91e723984faa4377be054429f470867574f5b28795b3433a

:: CBOR Attestation Object Data ::
RP ID Hash: a642d21b7c6d55e1ce23c5399828d2c749bf6a6ef2fe03cc9e10cdf4ed53088b
Counter: 00000001 Flags: 69
AAGUID: 00000000000000000000000000000000

:: Client Data Information ::
Client Data object, in full:
{
  "type": "webauthn.create",
  "challenge": "CtS0d2pbkezQgqCVY7VpoQ",
  "origin": "https://webauthn.bin.coffee"
}

[PASS] Challenge matches: CtS0d2pbkezQgqCVY7VpoQ == CtS0d2pbkezQgqCVY7VpoQ
[PASS] ClientData.origin matches this origin (WD-06): https://webauthn.bin.coffee == https://webauthn.bin.coffee
[PASS] Type is valid (WD-08): webauthn.create == webauthn.create

Raw request:
{
  "challenge": {
    "0": 10,
    "1": 212,
    "2": 180,
    "3": 119,
    "4": 106,
    "5": 91,
    "6": 145,
    "7": 236,
    "8": 208,
    "9": 130,
    "10": 160,
    "11": 149,
    "12": 99,
    "13": 181,
    "14": 105,
    "15": 161
  },
  "rp": {
    "name": "Acme"
  },
  "user": {
    "id": {
      "0": 49,
      "1": 48,
      "2": 57,
      "3": 56,
      "4": 50,
      "5": 51,
      "6": 55,
      "7": 50,
      "8": 51,
      "9": 53,
      "10": 52,
      "11": 48,
      "12": 57,
      "13": 56,
      "14": 55,
      "15": 50
    },
    "name": "john.p.smith@example.com",
    "displayName": "John P. Smith",
    "icon": "https://pics.acme.com/00/p/aBjjjpqPb.png"
  },
  "pubKeyCredParams": [
    {
      "alg": -7,
      "type": "public-key"
    }
  ],
  "authenticatorSelection": {
    "authenticatorAttachment": "cross-platform",
    "requireResidentKey": false,
    "userVerification": "preferred"
  },
  "timeout": 60000,
  "excludeCredentials": [],
  "extensions": {
    "exts": true
  }
}

Failures: 0 TODOs: 0
```

shizzgar, 2025-03-28:
> Can you do the same and intercept the usb port with Wireshark and upload the pcap?

Yes, we could, I'll update this comment a bit later today. 
Thanks for your responsiveness!!!

shizzgar, 2025-03-28:
@polhenarejos , first of all, our man, who has a yubikey is not getting online, so pcap will be here a bit later.

After all, we didn't do nothing :))
We tried to play with authenticator-rs itself and with the Pico-Fido firmware. We found, that after commenting out this two [lines](https://github.com/polhenarejos/pico-fido/blob/fef46dc1c53ae3d85f89cafe6b383ba142ff90b1/src/fido/cbor_make_credential.c#L534-L535) and changing the values (5 : 4 -> 4 : 3) in this [line](https://github.com/polhenarejos/pico-fido/blob/fef46dc1c53ae3d85f89cafe6b383ba142ff90b1/src/fido/cbor_make_credential.c#L505) firefox get's in work. Authentication works everythere, on all sites. There is only one exception, in the https://webauthn.bin.coffee/. It still throws error, but this time it is diffrent. Credentials registration works showing this in it's output:

```
Contacting token... please perform your verification gesture (e.g., touch it, or plug it in)

Note: Raw response in console.

:: "None" Attestation Format ::
```

And here is output from assertion step:

```
Contacting token... please perform your verification gesture (e.g., touch it, or plug it in)

Raw response in console.
[PASS] Challenge is identical: DI1XmWkj_gzhTT0mugb6Ng == DI1XmWkj_gzhTT0mugb6Ng
[PASS] ClientData.origin matches this origin (WD-06): https://webauthn.bin.coffee == https://webauthn.bin.coffee
Extensions: {}
[PASS] User presence must be the only flag set: 1 == 1
[PASS] Counter must be 4 bytes: 4 == 4

:: CBOR Attestation Object Data ::
RP ID Hash: a642d21b7c6d55e1ce23c5399828d2c749bf6a6ef2fe03cc9e10cdf4ed53088b
Counter: 0000000f Flags: 5

[PASS] Calculated RP ID hash must match what the browser derived.: pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs == pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs
ClientData buffer: 7b2274797065223a22776562617574686e2e676574222c226368616c6c656e6765223a22444931586d576b6a5f677a685454306d756762364e67222c226f726967696e223a2268747470733a2f2f776562617574686e2e62696e2e636f66666565227d

ClientDataHash: a2a3a167b475471772c2dfa11f71311912182c56d61b9bb51e6b97e84188100e

Signed Data assembled: 166,66,210,27,124,109,85,225,206,35,197,57,152,40,210,199,73,191,106,110,242,254,3,204,158,16,205,244,237,83,8,139,5,0,0,0,15,162,163,161,103,180,117,71,23,114,194,223,161,31,113,49,25,18,24,44,86,214,27,155,181,30,107,151,232,65,136,16,14
Got error:
TypeError: buf is null

Raw request:
{
  "challenge": {
    "0": 12,
    "1": 141,
    "2": 87,
    "3": 153,
    "4": 105,
    "5": 35,
    "6": 254,
    "7": 12,
    "8": 225,
    "9": 77,
    "10": 61,
    "11": 38,
    "12": 186,
    "13": 6,
    "14": 250,
    "15": 54
  },
  "timeout": 60000,
  "allowCredentials": [
    {
      "type": "public-key",
      "id": {
        "0": 241,
        "1": 208,
        "2": 2,
        "3": 2,
        "4": 202,
        "5": 181,
        "6": 245,
[... 200 lines cut ...]
  }
}

Failures: 1 TODOs: 0
```

Also, we used debug mode in the authenticator-rs and tried to display all available information from authentication process. We found, that before applying changes mentioned earlier to the pico-fido, pico returns this value:

```
[2025-03-28T20:44:56Z DEBUG authenticator::ctap2::commands::make_credentials] Received OK status for MakeCredentials. Raw input buffer (len=407): [00, A4, 01, 66, 70, 61, 63, 6B, 65, 64, 02, 59, 01, 33, A3, 79, A6, F6, EE, AF, B9, A5, 5E, 37, 8C, 11, 80, 34, E2, 75, 1E, 68, 2F, AB, 9F, 2D, 30, AB, 13, D2, 12, 55, 86, CE, 19, 47, C5, 00, 00, 01, 81, 89, FB, 94, B7, 06, C9, 36, 73, 9B, 7E, 30, 52, 6D, 96, 81, 45, 00, A1, F1, D0, 02, 02, 61, 5A, 2B, 04, 5C, 8E, 9A, FB, 64, 2A, E1, 34, 09, 05, A8, BE, B2, F9, F1, 00, C8, 70, 66, 80, 06, 1F, 49, 94, A0, 8F, 3B, 1A, D0, 5B, 87, 96, 45, 95, A1, 2F, 9C, 6A, 00, 97, 14, 98, DF, 6C, E0, 85, C5, 8D, 18, C3, FF, 30, E8, D1, B2, 5E, A4, CC, A5, 11, E8, 63, 69, BE, 04, 5C, 63, 09, D6, 00, 15, A7, C4, D7, 7C, 8A, C9, B0, 1E, 08, 4E, 6E, 0F, 1B, DD, CE, E3, 00, 89, A6, 5C, 66, 9D, 35, AB, DF, D4, 5E, 3A, 94, 5D, 59, 19, 8B, A0, C7, 19, CA, 98, A5, 57, E4, 4C, 1A, 79, 0E, 1C, 36, 57, A7, 7C, A0, 56, DD, 0D, CB, D2, 6B, 5D, 5A, 53, 0D, 13, 80, AB, EB, AA, 16, 43, DF, 23, 97, B7, A7, 73, 4F, 75, 90, EF, 9C, C3, D5, 63, A5, 01, 02, 03, 26, 20, 01, 21, 58, 20, F5, 8B, 62, DF, A6, 33, 17, A8, 62, 88, AE, ED, 7E, E4, DF, 07, 75, 52, 33, 32, 2F, F3, FC, 32, F8, 17, 48, 58, 90, C4, C5, D6, 22, 58, 20, B3, EC, 5A, F0, 2F, 23, 7A, 3F, D9, 88, 3A, 42, 1C, 56, 65, 11, 62, B7, D9, 20, 23, FD, B2, AF, 70, 40, CF, F3, 81, 2C, 23, 2D, A1, 6B, 68, 6D, 61, 63, 2D, 73, 65, 63, 72, 65, 74, F5, 03, A2, 63, 61, 6C, 67, 26, 63, 73, 69, 67, 58, 47, 30, 45, 02, 21, 00, F0, 1F, 74, 3F, 93, 18, 5F, F8, 70, E9, 80, 10, 99, 3F, E3, 23, F4, D0, C7, 60, B1, DE, D3, AF, E7, 7D, CC, 71, 94, 5C, B3, 62, 02, 20, 6D, 52, AE, 36, 9B, 38, 72, 08, 0B, B4, 36, A9, E0, E1, 96, 3B, 1B, C2, 88, B8, 95, 63, 9A, 75, B7, 0F, 87, B7, 91, 9B, B8, 83, 04, F4]
```

After applying the changes, pico is returning this value:
```
[2025-03-28T22:52:13Z DEBUG authenticator::ctap2::commands::make_credentials] Received OK status for MakeCredentials. Raw input buffer (len=405): [00, A3, 01, 66, 70, 61, 63, 6B, 65, 64, 02, 59, 01, 33, A3, 79, A6, F6, EE, AF, B9, A5, 5E, 37, 8C, 11, 80, 34, E2, 75, 1E, 68, 2F, AB, 9F, 2D, 30, AB, 13, D2, 12, 55, 86, CE, 19, 47, C5, 00, 00, 01, 89, 89, FB, 94, B7, 06, C9, 36, 73, 9B, 7E, 30, 52, 6D, 96, 81, 45, 00, A1, F1, D0, 02, 02, F4, 25, AC, DA, E7, 54, 5B, 8D, 97, 56, EC, 45, CA, B9, DB, 6B, 2C, F9, 2D, 1E, B7, 1 [... cut ...]
On authenticator-rs side error seems to come from this [line](https://github.com/mozilla/authenticator-rs/blob/9410d650378540fd60ec998cdcac17a9c5b4a888/src/ctap2/commands/make_credentials.rs#L580).

Note, that in chrome on linux https://webauthn.bin.coffee/ works well with both pico-fido firmware versions: originals from development branch and with development branch with our modifications. 

If it is necessary, we could sniff usb from ours pico boards... Still waiting for response from our guy with yubikey...  
Again, thanks for your awesome work!

polhenarejos, 2025-03-29:
Awesome debug work. 

Then it seems it does not support epAtt return parameter. I’ll push a fix to return it only when EA is supported.

TheMaxMur, 2025-03-29:
Btw, we research mention "epAtt" in authenticator-rs and find [this](https://bugzilla.mozilla.org/show_bug.cgi?id=1905492) in mozilla/authenticator-rs#339.

[image]

So, mozilla knows about the bug, and has had a fix for it for like a year now, but it still hasn't merged.

commit 64f371e6 (2025-03-30): Despite it is described in the spec 2.1, do not return epAtt if is false, return only when it's true. It fixes a bug with Firefox and Linux that blocked the possibility to make credentials.

Fixes #129.
closed 2025-04-10 by polhenarejos