polhenarejos/pico-fido #130: OTP not work on ESP32S3

wbchn, 2025-03-24:
After flashed v6.4 on ESP32S3(SuperMini), plugin in MacOS(15.3), passed in webauthn.

But I fount the key not work good on OTP: setup `Yubico OTP` on slot 1, `Static password` on slot2, but i none work when i touch the boot button, neither the OTP nor the Static Password was typed into the active text input box.

Is there anything wrong with ESP32S3?

More information:
```
$ ykman info
Device type: YubiKey 5A
...
Enabled USB interfaces: OTP, FIDO, CCID

Applications
Yubico OTP  	Enabled
FIDO U2F    	Enabled
FIDO2       	Enabled
OATH        	Enabled

$ ykman otp chalresp --generate 1
Using a randomly generated key (hex): 92c632ca6af7f3b5f33eb6f986f01f01907aad90
Program a challenge-response credential in slot 1? [y/N]: y
challenge-response credential stored in slot 1.
$ ykman otp static --generate 2 --length 38
Slot 2 is already configured. Overwrite configuration? [y/N]: y
Static password stored in slot 2.

$ ykman otp info
Slot 1: programmed
Slot 2: programmed
```

Wrong message with `ykman otp calculate`:
```
$ ykman otp calculate -T 1
ERROR: The connection type required for this command is not supported/enabled on the YubiKey.
```

polhenarejos, 2025-03-24:
Paste the output of `lsusb -v -d 1050:`

wbchn, 2025-03-25:
Since the Pico Commissioner not working now, I used command `python3 pico-fido-tool.py -p $PIN phy vidpid 1050:0407` to update pid vid.

Here is the output on my MacOS:

```
    USB 3.1 Bus:

      Host Controller Driver: AppleEmbeddedUSBXHCIASMedia3142
      PCI Device ID: 0x2142
      PCI Revision ID: 0x0000
      PCI Vendor ID: 0x1b21

        Yubico Yubikey:

          Product ID: 0x0407
          Vendor ID: 0x1050
          Version: 7.00
          Serial Number: 341C1D97BACC0000
          Speed: Up to 12 Mb/s
          Manufacturer: Pol Henarejos
          Location ID: 0x02400000 / 1
          Current Available (mA): 500
          Current Required (mA): 2
          Extra Operating Current (mA): 0
```

that's different on Linux:

```
Bus 001 Device 079: ID 1050:0407 Yubico.com Yubikey 4/5 OTP+U2F+CCID
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1050 Yubico.com
  idProduct          0x0407 Yubikey 4/5 OTP+U2F+CCID
  bcdDevice            7.00
  iManufacturer           1 Pol Henarejos
  iProduct                2 Yubico Yubikey
  iSerial                 3 341C1D97BACC0000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00db
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 Yubico Yubikey Config
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              5 Yubico Yubikey HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
[... 120 lines cut ...]
      iInterface              8 (error)
      ** UNRECOGNIZED:  35 21 10 01 00 01 03 00 00 00 fc 0d 00 00 fc 0d 00 00 00 80 25 00 00 80 25 00 00 00 fe 00 00 00 00 00 00 00 00 00 00 00 40 08 04 00 00 08 00 00 ff ff 00 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

wbchn, 2025-03-25:
The ykman otp works now:
```
ykman otp calculate -T 1
166195
```

In the active text inputbox, still has nothing typed(expect 44 otp characters) when i touch the key(short or long press):

[image]

polhenarejos, 2025-03-30:
Can you use the development version? Some fixes regarding OTP have been pushed there.

wbchn, 2025-03-31:
> Can you use the development version? Some fixes regarding OTP have been pushed there.

Thanks for your reply. The night development version still not working on my ESP32S3.

I tried in this way:
1. There is no burn button on ESP32 Flasher online page
2. I flashed with [Nightly Development](https://github.com/polhenarejos/pico-fido/releases/tag/nightly-development) pico_fido_esp32-s3.bin (md5sum: 01e210580f19dad949d181f8ee496d55) with esptool
3. Update vendor and product name on Pico Commissioner page
4. The Passkeys works fine, but the OTP(slot 1) and Static password(slot2) still not work, nothing typed to secret key input box when i add account in Yubico Authenticator app.

polhenarejos, 2025-03-31:
I don't understand 4th step. You have to provide the secret key to the app, it will show nothing unless you introduce it.
Besides this, "Add Account" and OTP are different applets/protocols (YKOATH and Yubico OTP), they are not connected or linked at any level.

wbchn, 2025-03-31:
I just to find an activate input box, hope to get OTP when touched the key.

Also try to open a text editor instead of `Add account` dialog, no otp generated.

> A Yubico OTP (one-time password) is a unique 44-character string that is generated by the YubiKey when it is touched
> from: https://docs.yubico.com/yesdk/users-manual/application-otp/yubico-otp.html

wbchn, 2025-04-02:
I programed these on slot 1, using press boot once on ESP32S3, none output.
- Yubico OTP
- Static password
- OATH-HTOP

[image]

Also tried programed on slot 2, press twice, none output.

I suspect the issue might be related to HID. I tried to dive into the code and found the implementation of `otp_button_pressed`, but unfortunately, I wasn’t sure how to properly debug it on the ESP32-S3, so I had to give up for now.

As a workaround, I bought an RP2050-Zero yesterday, and it seems to be working fine.

Thanks!

wbchn, 2025-04-02:
By the way, I chose the ESP32-S3 because it supports capacitive touch. I was planning to modify the project to use a touch input instead of pressing the BOOT button.

When I tested with version 6.4, I found that if I initialized GPIO7 (other pins also worked) as a regular input, touching the pad could pull the pin low, which caused picok_board_button_read to return true. However, this behavior seems to have disappeared in the latest development branch.

Perhaps the RP2350 with TTP223(a touch pad detector IC) is a better fit after all.

wbchn, 2025-04-05:
@polhenarejos I used the development firmware on `waveshare_rp2350_zero` board, by short press and twice press, `OTP` and `Static password` output work fine.

closed 2025-04-10 by polhenarejos

wbchn, 2025-04-19:
Release v6.6 working fine on ESP32S3 now. Thanks @polhenarejos !

commit c8dbc213 (2025-04-06): Fix EPNUM counting for ESP32. It fixes the problem of not sending KB.

Fixes #130 #138.