polhenarejos/pico-fido #131: Pico-FIDO Not Working After Setup - Help!

Th4phat, 2025-03-24:
Hey! So, I flashed my Pico 2 with the Pico-FIDO firmware (the `pico_fido_pico2-6.4.uf2` one), and it's just doesn't work here's a vid that I'd record

https://github.com/user-attachments/assets/66e59a59-468d-40c1-ba86-6f5f5cbaa1dc
https://github.com/user-attachments/assets/e7be53d9-8db2-4c2c-971e-8322552785af

Is this something I'm messing up, or is there a problem with the firmware?  I've tried it on both Windows 10 and Linux, and neither works.  Any ideas?

I can confirm that I check the initialize button with yubikey 4/5 before pairing and commission.

Thanks!

polhenarejos, 2025-03-24:
For the fist video, the problem is that it's asking the PIN twice or what?

For the second video, I see that this appspot asks for an existing credential but it seems you did not create the credential, so it cannot find it. Did you create it first?

Th4phat, 2025-03-24:
1. Yes exactly I don't know either why it does that, even if I enter the correct PIN it would just ask me to touch my passkey for a second then just show some error message then make me enter PIN again and again.
2. I can confirm that I've already create an account.

After some diagnose from looking in the issue I wanted to ask you something, is it normal that when I'm running `ykman` it show `ERROR: No YubiKey detected!`

It also does the same thing if I do it on linux (fedora 41).

Th4phat, 2025-03-24:
First I thought it might be the windows problem because I'm using debloater tool when I'm installing my windows and stuff like that but when I also tried on linux by nuking the firmware then reinstall and commission it, it's still not working.

Th4phat, 2025-03-24:
First I thought it might be the windows problem because I'm using debloater tool when I'm installing my windows and stuff like that but when I also tried on linux by nuking the firmware then reinstall it, here's my tested result from linux.

[image]
[image]

polhenarejos, 2025-03-24:
I tried https://www.passkeys-debugger.io/ and it works to me. Do you select cross-platform authenticator attachment?

Th4phat, 2025-03-24:
The picture I showed was not selected as `cross-platform` but when I try again with cross-platform selected it just doing the thing like in the video 1. It just ask my PIN again and again

Th4phat, 2025-03-24:
It ask me to touch my security key. I click BOOTSEL and nothing happen.

Th4phat, 2025-03-24:
After I nuke the firmware then re nstall firmware and commission again with Presence Button Timeout to disable it still giving me the same result as my last comment.

Young-Lord, 2025-07-26:
You may open https://www.passkeys.io/ in Chromium, and set the PIN for your key. My key works after I set a PIN.