polhenarejos/pico-fido #134: How to clear / initialize?

erikschul, 2025-03-28:
I've used the Raspberry 2350 for testing. The passkey now contains multiple test accounts (which show up in Firefox/Linux). I've tried using the online Pico Commisioner, with "Initialize" selected, and clicked "Commission via WebAuthn", but it doesn't remove the keys.
I've also tried uploading a new ROM via U2F (which I assume shouldn't affect the storage?).

How do you clear it?

IsayIsee, 2025-03-28:
you can use nuke fireware, https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#resetting-flash-memory

polhenarejos, 2025-03-28:
Look for Pico Nuke in my repos and download the fw for your board.

erikschul, 2025-03-28:
@polhenarejos Do I understand correctly that configuring Secure Boot would only allow releases certified by your signing key, which includes the Pico Nuke?

On a side note, do you have any documentation regarding how to use my own signing key, if I plan to make modifications? Does the Pico Fido web commissioner configure the signing key? And can the key be commissioned locally?

polhenarejos, 2025-03-30:
I uploaded v1.4 which adds my signature in case you enabled secure boot to be able to nuke it too.

No, Pico Commissioner burns my public key in fuse 0. In case you want to use your own public key, then you should build the firmware locally and specify the SECURE_BOOT_KEY variable in the CMake to point to your .pem file.
Besides which key you use, it is only used during the boot sequence and Pico Commissioner is agnostic to which public key you use (except for Secure boot and lock, obviously). In case your burn your own key, you will be able to continue using the commissioner.