polhenarejos/pico-fido #135: yubico OTP Serial number is too long

Jonathan-Nathaniel, 2025-03-29:
[image]

I bought some development boards, and after flashing, the S/N was greater than 9 digits, and then after uploading the file, this prompt appeared.

Although I can delete one digit manually, I don't know if this will cause an exception

Jonathan-Nathaniel, 2025-03-29:
After testing, it seems to be able to be used normally

polhenarejos, 2025-03-29:
I don’t understand what are you trying to.

closed 2025-03-30 by Jonathan-Nathaniel

Jonathan-Nathaniel, 2025-03-30:
I apologize for the lack of clarity in my previous statement.

Nightly Development Firmware: pico_fido_waveshare_rp2350_one-6.4.uf2

Upon flashing, the serial number (S/N) will be displayed.

When selecting "Short" under "Slots" in the Yubico Authenticator and generating the Yubico OTP file, the serial number displays as "Serial number is too long" upon uploading to Yubico OTP key upload.

[image]

Jonathan-Nathaniel, 2025-03-30:
The file upload function supports 10-digit serial numbers, but some devices currently have 11-digit serial numbers.

Raito00, 2025-05-09:
I have the same problem on other softwares with one have integration Yubikey
Software waiting max 9 digits serial NR.
But some Pico2 when install Pico-Fido have 10 digit serial.

It is possible when flash set firmware to set serial 9 digits !?

polhenarejos, 2025-05-10:
Why is important to upload the S/N? You can put there any value you consider, it will work anyway.