polhenarejos/pico-fido #136: Webauthn bin coffee not working in Firefox and Linux

polhenarejos, 2025-03-30:
> @polhenarejos , first of all, our man, who has a yubikey is not getting online, so pcap will be here a bit later.
> 
> After all, we didn't do nothing :))
> We tried to play with authenticator-rs itself and with the Pico-Fido firmware. We found, that after commenting out this two [lines](https://github.com/polhenarejos/pico-fido/blob/fef46dc1c53ae3d85f89cafe6b383ba142ff90b1/src/fido/cbor_make_credential.c#L534-L535) and changing the values (5 : 4 -> 4 : 3) in this [line](https://github.com/polhenarejos/pico-fido/blob/fef46dc1c53ae3d85f89cafe6b383ba142ff90b1/src/fido/cbor_make_credential.c#L505) firefox get's in work. Authentication works everythere, on all sites. There is only one exception, in the https://webauthn.bin.coffee/. It still throws error, but this time it is diffrent. Credentials registration works showing this in it's output:
> 
> ```
> Contacting token... please perform your verification gesture (e.g., touch it, or plug it in)
> 
> Note: Raw response in console.
> 
> :: "None" Attestation Format ::
> ```
> 
> And here is output from assertion step:
> 
> ```
> Contacting token... please perform your verification gesture (e.g., touch it, or plug it in)
> 
> Raw response in console.
> [PASS] Challenge is identical: DI1XmWkj_gzhTT0mugb6Ng == DI1XmWkj_gzhTT0mugb6Ng
> [PASS] ClientData.origin matches this origin (WD-06): https://webauthn.bin.coffee == https://webauthn.bin.coffee
> Extensions: {}
> [PASS] User presence must be the only flag set: 1 == 1
> [PASS] Counter must be 4 bytes: 4 == 4
> 
> :: CBOR Attestation Object Data ::
> RP ID Hash: a642d21b7c6d55e1ce23c5399828d2c749bf6a6ef2fe03cc9e10cdf4ed53088b
> Counter: 0000000f Flags: 5
> 
> [PASS] Calculated RP ID hash must match what the browser derived.: pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs == pkLSG3xtVeHOI8U5mCjSx0m_am7y_gPMnhDN9O1TCIs
> ClientData buffer: 7b2274797065223a22776562617574686e2e676574222c226368616c6c656e6765223a22444931586d576b6a5f677a685454306d756762364e67222c226f726967696e223a2268747470733a2f2f776562617574686e2e62696e2e636f66666565227d
> 
> ClientDataHash: a2a3a167b475471772c2dfa11f71311912182c56d61b9bb51e6b97e84188100e
> 
> Signed Data assembled: 166,66,210,27,124,109,85,225,206,35,197,57,152,40,210,199,73,191,106,110,242,254,3,204,158,16,205,244,237,83,8,139,5,0,0,0,15,162,163,161,103,180,117,71,23,114,194,223,161,31,113,49,25,18,24,44,86,214,27,155,181,30,107,151,232,65,136,16,14
> Got error:
> TypeError: buf is null
> 
> 
> 
> Raw request:
> {
>   "challenge": {
>     "0": 12,
>     "1": 141,
>     "2": 87,
>     "3": 153,
>     "4": 105,
>     "5": 35,
>     "6": 254,
>     "7": 12,
>     "8": 225,
>     "9": 77,
>     "10": 61,
>     "11": 38,
>     "12": 186,
>     "13": 6,
>     "14": 250,
>     "15": 54
>   },
>   "timeout": 60000,
>   "allowCredentials": [
>     {
>       "type": "public-key",
>       "id": {
>         "0": 241,
>         "1": 208,
>         "2": 2,
>         "3": 2,
>         "4": 202,
[... 205 lines cut ...]
> Failures: 1 TODOs: 0
> ```
> 
> Also, we used debug mode in the authenticator-rs and tried to display all available information from authentication process. We found, that before applying changes mentioned earlier to the pico-fido, pico returns this value:
> 
> ```
> [2025-03-28T20:44:56Z DEBUG authenticator::ctap2::commands::make_credentials] Received OK status for MakeCredentials. Raw input buffer (len=407): [00, A4, 01, 66, 70, 61, 63, 6B, 65, 64, 02, 59, 01, 33, A3, 79, A6, F6, EE, AF, B9, A5, 5E, 37, 8C, 11, 80, 34, E2, 75, 1E, 68, 2F, AB, 9F, 2D, 30, AB, 13, D2, 12, 55, 86, CE, 19, 47, C5, 00, 00, 01, 81, 89, FB, 94, B7, 06, C9, 36, 73, 9B, 7E, 30, 52, 6D, 96, 81, 45, 00, A1, F1, D0, 02, 02, 61, 5A, 2B, 04, 5C, 8E, 9A, FB, 64, 2A, E1, 34, 09, 05, A8, BE, B2, F9, F1, 00, C8, 70, 66, 80, 06, 1F, 49, 94, A0, 8F, 3B, 1A, D0, 5B, 87, 96, 45, 95, A1, 2F, 9C, 6A, 00, 97, 14, 98, DF, 6C, E0, 85, C5, 8D, 18, C3, FF, 30, E8, D1, B2, 5E, A4, CC, A5, 11, E8, 63, 69, BE, 04, 5C, 63, 09, D6, 00, 15, A7, C4, D7, 7C, 8A, C9, B0, 1E, 08, 4E, 6E, 0F, 1B, DD, CE, E3, 00, 89, A6, 5C, 66, 9D, 35, AB, DF, D4, 5E, 3A, 94, 5D, 59, 19, 8B, A0, C7, 19, CA, 98, A5, 57, E4, 4C, 1A, 79, 0E, 1C, 36, 57, A7, 7C, A0, 56, DD, 0D, CB, D2, 6B, 5D, 5A, 53, 0D, 13, 80, AB, EB, AA, 16, 43, DF, 23, 97, B7, A7, 73, 4F, 75, 90, EF, 9C, C3, D5, 63, A5, 01, 02, 03, 26, 20, 01, 21, 58, 20, F5, 8B, 62, DF, A6, 33, 17, A8, 62, 88, AE, ED, 7E, E4, DF, 07, 75, 52, 33, 32, 2F, F3, FC, 32, F8, 17, 48, 58, 90, C4, C5, D6, 22, 58, 20, B3, EC, 5A, F0, 2F, 23, 7A, 3F, D9, 88, 3A, 42, 1C, 56, 65, 11, 62, B7, D9, 20, 23, FD, B2, AF, 70, 40, CF, F3, 81, 2C, 23, 2D, A1, 6B, 68, 6D, 61, 63, 2D, 73, 65, 63, 72, 65, 74, F5, 03, A2, 63, 61, 6C, 67, 26, 63, 73, 69, 67, 58, 47, 30, 45, 02, 21, 00, F0, 1F, 74, 3F, 93, 18, 5F, F8, 70, E9, 80, 10, 99, 3F, E3, 23, F4, D0, C7, 60, B1, DE, D3, AF, E7, 7D, CC, 71, 94, 5C, B3, 62, 02, 20, 6D, 52, AE, 36, 9B, 38, 72, 08, 0B, B4, 36, A9, E0, E1, 96, 3B, 1B, C2, 88, B8, 95, 63, 9A, 75, B7, 0F, 87, B7, 91, 9B, B8, 83, 04, F4]
> ```
> 
> After applying the changes, pico is returning this value:
> ```
> [2025-03-28T22:52:13Z DEBUG authenticator::ctap2::commands::make_credentials] Received OK status for MakeCredentials. Raw input buffer (len=405): [00, A3, 01, 66, 70, 61, 63, 6B, 65, 64, 02, 59, 01, 33, A3, 79, A6, F6, EE, AF, B9, A5, 5E, 37, 8C, 11, 80, 34, E2, 75, 1E, 68, 2F, AB, 9F, 2D, 30, AB, 13, D2, 12, 55, 86, CE, 19, 47, C5, 00, 00, 01, 89, 89, FB, 94, B7, 06, C9, 36, 73, 9B, 7E, 30, 52, 6D, 96, 81, 45, 00, A1, F1, D0, 02, 02, F4, 25, AC, DA, E7, 54, 5B, 8D, 97, 56, EC, 45, CA, B9, DB, 6B, 2C, F9, 2D, 1E, B7, 12, A0, 1A, DC, 94, 42, 35, CA, E8, 66, 48, 96, 1C, A4, CA, 92, 38, F7, A5, F5, 49, 64, B1, 63, EE, EE, 45, 97, B2, EF, 1B, 1B, 45, 93, 4C, 60, CF, 7F, 05, 75, 46, 8A, 55, 63, ED, 4F, D2, 61, 1A, 31, 26, 63, B1, 13, 72, 3B, 75, 43, BA, D5, B9, 7A, 17, B8, D9, 6E, 42, 24, 88, D5, 0E, 38, DB, 97, A3, E5, 42, 24, EB, 7D, 0D, 18, 61, 81, A4, B4, 19, 08, 5C, 62, F1, 96, 0B, 87, 02, A5, 9E, 40, 70, 6B, A0, 9B, C6, 5B, 74, ED, 7E, 5A, 26, 96, 5F, CE, 25, 8D, F6, 49, 68, 35, 25, EA, EA, 82, 58, D1, 1B, 14, DE, AD, 0E, D3, 8E, B2, FA, 3A, B8, 5B, A5, 01, 02, 03, 26, 20, 01, 21, 58, 20, B9, C7, 33, 7C, 82, D6, AD, 83, 33, B2, 22, 05, 9A, 89, DA, B0, B6, 9F, 49, 92, 19, 2C, 5E, D6, 27, DA, 74, 1B, 5B, BC, 93, 22, 22, 58, 20, D6, 28, 00, D7, 22, 5F, 50, 4F, D5, F1, FE, BC, 2C, E6, 12, 5C, BB, 6B, DC, 34, BE, 2C, 31, 4A, DD, 3F, F3, A9, 55, 88, 78, 32, A1, 6B, 68, 6D, 61, 63, 2D, 73, 65, 63, 72, 65, 74, F5, 03, A2, 63, 61, 6C, 67, 26, 63, 73, 69, 67, 58, 47, 30, 45, 02, 20, 54, 7A, 37, 42, 94, 25, 3D, E0, D6, 67, 29, 33, 54, 94, 60, 9F, DC, 84, 8D, 25, 50, 77, 09, BB, 70, 91, 76, 8D, 86, 5E, 99, 3F, 02, 21, 00, EE, 66, CC, 32, BE, 0A, BC, AE, 29, 31, F7, 94, 56, B2, CB, FF, 2C, 6E, 0D, 59, 27, 2E, 16, 77, E2, DE, B8, 3B, EE, 70, 27, A0]
> ```
> 
> (It seems, that there is only 2 last bytes gets removed)
> 
> On authenticator-rs side error seems to come from this [line](https://github.com/mozilla/authenticator-rs/blob/9410d650378540fd60ec998cdcac17a9c5b4a888/src/ctap2/commands/make_credentials.rs#L580).
> 
> Note, that in chrome on linux https://webauthn.bin.coffee/ works well with both pico-fido firmware versions: originals from development branch and with development branch with our modifications. 
> 
> If it is necessary, we could sniff usb from ours pico boards... Still waiting for response from our guy with yubikey...  
> Again, thanks for your awesome work!
>  

 _Originally posted by @shizzgar in [#129](https://github.com/polhenarejos/pico-fido/issues/129#issuecomment-2762791968)_

polhenarejos, 2025-03-30:
webauthn.bin.coffee throws an error of remaining bytes when processing cborPublicKey. 

It constructs the COSEpublicKey as the remaining bytes, but since there are extensions, it throws "Remaining bytes". Usually the remaining bytes shall not throw any error as they are the extensions cbor map, but it happens in linux and firefox.

TheMaxMur, 2025-03-30:
Checked c3ea413592620354619975037f118a79af7c2210 in librewolf on linux. Always works.

[image]

commit c3ea4135 (2025-03-30): Do not return extensions if they are not requested OR are false.

Fixes #136
closed 2025-04-10 by polhenarejos