polhenarejos/pico-fido #137: Yubico Authenticator not working

TheMaxMur, 2025-03-30:
In linux on development branch on commit (64f371e6e5a49a4b75c0f3b699f861c9ca103732) (In fact, it didn't work before that either) Yubico authernicator not working.

[image]

```
(yubioath-flutter:496109): Atk-CRITICAL **: 20:22:05.625: atk_socket_embed: assertion 'plug_id != NULL' failed
20:22:05.633 [desktop.init] INFO: Logging initialized, outputting to stderr
20:22:05.635 [desktop.init] INFO: Window hidden on startup: false
20:22:05.635 [desktop.init] INFO: Starting Helper subprocess: /nix/store/yrla9gmbfkiv0s25dkr9rzb4xp1ylf68-yubioath-flutter-7.1.1/app/yubioath-flutter/helper/authenticator-helper
20:22:05.635 [main] INFO: Running Yubico Authenticator...
{app_version: 7.1.1, dart: 3.5.4 (stable) (Wed Oct 16 16:18:51 2024 +0000) on "linux_x64", os: linux, os_version: Linux 6.13.7-cachyos #1-NixOS SMP PREEMPT_DYNAMIC Thu Mar 13 12:08:08 UTC 2025}
20:22:05.639 [desktop.init] INFO: Helper process started
20:22:05.769 [helper.ykman.logging] INFO: Logging at level: INFO
20:22:05.769 [helper.helper.device] INFO: Log level set to: INFO
20:22:05.769 [desktop.init] INFO: Helper log level set
20:22:06.792 [desktop.devices] INFO: USB state change
{"data":{"state":5121579336810014002,"pids":{"1031":1}},"actions":["get","scan"],"children":{"1116062926":{"pid":1031,"name":"YubiKey 5A","serial":1116062926}}}
20:22:06.805 [desktop.devices] INFO: USB state updated, unaccounted for: {}
20:22:09.704 [helper.helper] SEVERE: Unhandled exception
Traceback (most recent call last):
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/__init__.py", line 105, in process
    handler(
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/device.py", line 94, in __call__
    result = super().__call__(*args)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 126, in __call__
    response = self.get_child(target[0])(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/device.py", line 205, in __call__
    response = super().__call__(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 126, in __call__
    response = self.get_child(target[0])(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/device.py", line 277, in __call__
    response = super().__call__(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 126, in __call__
    response = self.get_child(target[0])(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/device.py", line 495, in __call__
    return super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 126, in __call__
    response = self.get_child(target[0])(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/fido.py", line 96, in __call__
    return super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 126, in __call__
    response = self.get_child(target[0])(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 133, in __call__
    response = self.get_child(action)(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/base.py", line 220, in get_child
    self._child = self.create_child(name)
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/fido.py", line 298, in create_child
    return CredentialsRpNode(self.credman, self._rps[name], self.refresh)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/fido.py", line 308, in __init__
    self.refresh()
  File "/nix/store/rz6hs5alnbaizzwccmhkbw6zcv3yhr4v-yubioath-flutter-helper-7.1.1/lib/python3.12/site-packages/helper/fido.py", line 320, in refresh
    for cred in self.credman.enumerate_creds(self.data["rp_id_hash"])
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/1s54ylfacvv5wlm3wlqwz7flsi8w4xpv-python3.12-fido2-1.2.0/lib/python3.12/site-packages/fido2/ctap2/credman.py", line 206, in enumerate_creds
    first = self.enumerate_creds_begin(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/1s54ylfacvv5wlm3wlqwz7flsi8w4xpv-python3.12-fido2-1.2.0/lib/python3.12/site-packages/fido2/ctap2/credman.py", line 185, in enumerate_creds_begin
    return self._call(
           ^^^^^^^^^^^
  File "/nix/store/1s54ylfacvv5wlm3wlqwz7flsi8w4xpv-python3.12-fido2-1.2.0/lib/python3.12/site-packages/fido2/ctap2/credman.py", line 124, in _call
    return self.ctap.credential_mgmt(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/1s54ylfacvv5wlm3wlqwz7flsi8w4xpv-python3.12-fido2-1.2.0/lib/python3.12/site-packages/fido2/ctap2/base.py", line 476, in credential_mgmt
    return self.send_cbor(
           ^^^^^^^^^^^^^^^
  File "/nix/store/1s54ylfacvv5wlm3wlqwz7flsi8w4xpv-python3.12-fido2-1.2.0/lib/python3.12/site-packages/fido2/ctap2/base.py", line 241, in send_cbor
    raise CtapError(status)
fido2.ctap.CtapError: CTAP error: 0x30 - NOT_ALLOWED
```

polhenarejos, 2025-03-30:
Can you try `ykman -l TRAFFIC fido info` ?

TheMaxMur, 2025-03-30:
> Can you try `ykman -l TRAFFIC fido info` ?

Yes, sure

```
INFO 21:44:29.305 [ykman.logging.set_log_level:60] Logging at level: TRAFFIC
WARNING 21:44:29.305 [ykman.logging.set_log_level:62] 
###################################################################################
#                                                                                 #
# WARNING: All data sent to/from the YubiKey will be logged!                      #
# This data may contain sensitive values, such as secret keys, PINs or passwords! #
#                                                                                 #
###################################################################################
INFO 21:44:29.305 [ykman._cli.__main__.cli:355] System info:
  ykman:            5.5.1
  Python:           3.12.9 (main, Feb  4 2025, 14:38:38) [GCC 14.2.1 20241116]
  Platform:         linux
  Arch:             x86_64
  System date:      2025-03-30
  Running as admin: False

DEBUG 21:44:29.888 [ykman.device.add:165] Add device for <class 'fido2.ctap.CtapDevice'>: CtapYubiKeyDevice(pid=0407, fingerprint='/dev/hidraw2')
TRAFFIC 21:44:29.939 [fido2.hid._do_call:191] SEND: ffffffff8600083b5389d93ed5248b
TRAFFIC 21:44:30.57 [fido2.hid._do_call:208] RECV: ffffffff8600113b5389d93ed5248b00000001020604000500000000000000000000000000000000000000000000000000000000000000000000000000000000
DEBUG 21:44:30.57 [yubikit.support.read_info:264] Attempting to read device info, using CtapHidDevice
DEBUG 21:44:30.57 [yubikit.management.__init__:558] Management session initialized for connection=CtapHidDevice, version=6.4.0
DEBUG 21:44:30.57 [yubikit.management.read_device_info:586] Reading DeviceInfo page: 0
TRAFFIC 21:44:30.57 [fido2.hid._do_call:191] SEND: 00000001c2000100
TRAFFIC 21:44:30.73 [fido2.hid._do_call:208] RECV: 00000001bb0001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DEBUG 21:44:30.73 [fido2.hid._do_call:223] Got keepalive status: 01
TRAFFIC 21:44:30.81 [fido2.hid._do_call:208] RECV: 00000001c2001d1c0102022302044285c4ce0401010503060400030202230801800a010000000000000000000000000000000000000000000000000000000000
DEBUG 21:44:30.81 [yubikit.support.read_info:292] Read info: DeviceInfo(config=DeviceConfig(enabled_capabilities={<TRANSPORT.USB: 'usb'>: <CAPABILITY.OTP|U2F|FIDO2|OATH: 547>}, auto_eject_timeout=0, challenge_response_timeout=0, device_flags=<DEVICE_FLAG.EJECT: 128>, nfc_restricted=False), serial=1116062926, version=Version(major=6, minor=4, patch=0), form_factor=<FORM_FACTOR.USB_A_KEYCHAIN: 1>, supported_capabilities={<TRANSPORT.USB: 'usb'>: <CAPABILITY.OTP|U2F|FIDO2|OATH: 547>}, is_locked=False, is_fips=False, is_sky=False, part_number=None, fips_capable=<CAPABILITY: 0>, fips_approved=<CAPABILITY: 0>, pin_complexity=False, reset_blocked=<CAPABILITY: 0>, fps_version=None, stm_version=None)
DEBUG 21:44:30.81 [yubikit.support.read_info:351] Device info, after tweaks: DeviceInfo(config=DeviceConfig(enabled_capabilities={<TRANSPORT.USB: 'usb'>: <CAPABILITY.OTP|U2F|FIDO2|OATH: 547>}, auto_eject_timeout=0, challenge_response_timeout=0, device_flags=<DEVICE_FLAG.EJECT: 128>, nfc_restricted=False), serial=1116062926, version=Version(major=6, minor=4, patch=0), form_factor=<FORM_FACTOR.USB_A_KEYCHAIN: 1>, supported_capabilities={<TRANSPORT.USB: 'usb'>: <CAPABILITY.OTP|U2F|FIDO2|OATH: 547>}, is_locked=False, is_fips=False, is_sky=False, part_number=None, fips_capable=<CAPABILITY: 0>, fips_approved=<CAPABILITY: 0>, pin_complexity=False, reset_blocked=<CAPABILITY: 0>, fps_version=None, stm_version=None)
DEBUG 21:44:30.82 [ykman.device.add:176] Resolved device 1116062926
TRAFFIC 21:44:30.133 [fido2.hid._do_call:191] SEND: ffffffff8600084436c0ed6295283b
TRAFFIC 21:44:30.193 [fido2.hid._do_call:208] RECV: ffffffff8600114436c0ed6295283b00000001020604000500000000000000000000000000000000000000000000000000000000000000000000000000000000
TRAFFIC 21:44:30.193 [fido2.hid._do_call:191] SEND: 0000000190000104
TRAFFIC 21:44:30.209 [fido2.hid._do_call:208] RECV: 00000001bb0001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DEBUG 21:44:30.209 [fido2.hid._do_call:223] Got keepalive status: 01
TRAFFIC 21:44:30.217 [fido2.hid._do_call:208] RECV: 0000000190015600af0183665532465f5632684649444f5f325f30684649444f5f325f3102866863726564426c6f626b6372656450726f746563746b686d6163
TRAFFIC 21:44:30.225 [fido2.hid._do_call:208] RECV: 00000001002d7365637265746c6c61726765426c6f624b65796c6d696e50696e4c656e67746871746869726450617274795061796d656e74035089fb94b706c9
TRAFFIC 21:44:30.233 [fido2.hid._do_call:208] RECV: 000000010136739b7e30526d96814504a9626570f462726bf568616c776179735576f568637265644d676d74f569617574686e72436667f569636c69656e7450
TRAFFIC 21:44:30.241 [fido2.hid._do_call:208] RECV: 0000000102696ef56a6c61726765426c6f6273f56e70696e557641757468546f6b656ef56f7365744d696e50494e4c656e677468f50519040006820102071008
TRAFFIC 21:44:30.249 [fido2.hid._do_call:208] RECV: 00000001031904000a83a263616c672664747970656a7075626c69632d6b6579a263616c67382264747970656a7075626c69632d6b6579a263616c6738236474
TRAFFIC 21:44:30.257 [fido2.hid._do_call:208] RECV: 00000001047970656a7075626c69632d6b65790b1908000cf40d040e1906040f188015821b03e43f56b34285e21b1831a40f04a25ed900000000000000000000
TRAFFIC 21:44:30.259 [fido2.hid._do_call:191] SEND: 0000000190000606a201020201
TRAFFIC 21:44:30.273 [fido2.hid._do_call:208] RECV: 00000001bb0001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DEBUG 21:44:30.273 [fido2.hid._do_call:223] Got keepalive status: 01
TRAFFIC 21:44:30.345 [fido2.hid._do_call:208] RECV: 0000000190000400a103080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
PIN:                    8 attempt(s) remaining
Minimum PIN length:     4
Always Require UV:      On
Enterprise Attestation: Disabled
```

polhenarejos, 2025-03-30:
So it seems working. I don’t know why it doesn’t in your Yubikey Authenticator (mine works too).

TheMaxMur, 2025-03-30:
After nuke, everything worked again

closed 2025-03-30 by TheMaxMur