polhenarejos/pico-fido #139: How to extract OAUTH (TOTP keys) from a pico-tool dump?

fucksophie, 2025-04-03:
I dumped the memory of my pico a while back, and the pico has now died and I need to get my TOTP keys out. How do I do that?

polhenarejos, 2025-04-06:
Is it recognized by `picotool`?

fucksophie, 2025-04-07:
Yes.

polhenarejos, 2025-04-07:
Then you can backup and restore the entire flash memory to another device. It will generate new FIDO credentials and older will not be recognized but in the case of OTP should work since it uses another filesystem.

fucksophie, 2025-04-08:
> Then you can backup and restore the entire flash memory to another device. It will generate new FIDO credentials and older will not be recognized but in the case of OTP should work since it uses another filesystem.

The death of the pico was because I applied a update to it, and that update has made it so I cannot access yubikey.

fucksophie, 2025-04-08:
Thanks, I tried a bunch of different UF2's, untill I saw that the pid:vid of the pico key was 0000:0000. I was running into #118 , I updated my pico key, and then comissioned it, and then all of my fido stuffs dissapeared. Is there a way to recover them? Passkeys are gone, but all of the normal OAUTH stuff is still there.

polhenarejos, 2025-04-08:
Try the development version and commission it through the web. It should work unless you installed another firmware prepared for less memory that overwrote the previous map.

fucksophie, 2025-04-08:
I am using the development version, commissioned through web. All passkey stuff is gone, I didnt use another firmware with less memory, only 
`pico_fido_seeed_xiao_rp2040-6.4.uf2`
`pico_fido_seeed_xiao_rp2040-6.4.uf2 (development ver)`
`pico_fido_seeed_xiao_rp2040-6.4.yubikey5.uf2 (pico patcher version)`

and almost flashed `pico_hsm_seeed_xiao_rp2040-6.4.uf2`, however I noticed it's the "pico_**hsm**" version.

polhenarejos, 2025-04-08:
Probably the HSM wiped all the previous data.

fucksophie, 2025-11-22:
@polhenarejos Cannot seem to get the passkeys out of it in any way shape or form. Even the previous dumps I can't get the passkey.

Is there a way that I can emulate the passkey or spoof it to get access to a bunch of my accounts?