polhenarejos/pico-fido #140: Support for 'other devices'?

jouellnyc, 2025-04-07:
Good day. Great software and idea. I have a few of these devices:

https://www.aliexpress.us/item/3256806523983628.html?spm=a2g0o.order_list.order_list_main.31.21ef1802kObhvV&gatewayAdapt=glo2usa

[image]

I suppose they are inexpensive 2040 usb boards. 
Apart from not being suggested due to the lack of built-in security features, does it seem like the pico fido firmware "should" work?

For fun, I tried the latest  'plain' pi pico firmware and it booted but not did not boot with the latest  pico fido firmware.
I did not expect it to, but curious. Thanks

polhenarejos, 2025-04-07:
I build for dozens of devices because all have different combinations of memory and LED pin. What you need is to find a board with the same flash memor, LED driver and MCU (either RP2040 or RP2350) as yours. LED pin number can be changed afterwards. If you find a clone of your board, it will work like the original.

jouellnyc, 2025-04-07:
Okay cool... that's helpful... I figured as much but also one never knows until one asks, so thank you very much

tinwhisker, 2025-04-19:
@polhenarejos Something that might be useful is an option to select LED type: Standard, Neo/WS2812, RGB. (RGB might need to show 3 IO fields in the configurator)

@jouellnyc I have the RP2350 version of the item you linked. I added the line "#define PICO_DEFAULT_WS2812_PIN 22" into pico_keys.h as a test (and built). Works great.

(Thankfully unlike the Waveshare versions, you don't need to enable another pin to power the LED.)