polhenarejos/pico-fido #141: OTP update doesn't work

polhenarejos, 2025-04-08:
To put it simply, as long as the otp is set to challenge the password attribute of the modified slot, the challenge data will be damaged and the challenge verification will fail.
For example, after setting the challenge, setting the password through the ykman otp settings -A 576bc54882f3 2 command will cause the challenge verification to fail.

By @IsayIsee

polhenarejos, 2025-04-08:
Fixed in 3212f95.

Tested with:

```
ykman otp chalresp 1 -T -f JBSWY3DPEHPK3PXP
ykman otp calculate 1 -T   
ykman otp settings -f -A 576bc54882f3 1   
ykman otp calculate 1 -T   
ykman otp --access-code 576bc54882f3 settings -f --delete-access-code 1
ykman otp calculate 1 -T
```

all three produce the same result.

IsayIsee, 2025-04-09:
@polhenarejos 

The test challenge successfully adds/deletes passwords, but the button's trigger setting still defaults from press to passive

[image]

polhenarejos, 2025-04-09:
And what does it mean?

polhenarejos, 2025-04-09:
Passive means the challenge does not requiere touch. If you configure the slot with "Require touch", a "Press" will appear instead of "Passive".

IsayIsee, 2025-04-10:
@polhenarejos but i set need touch, after add password the touch is not need

[image]

commit 3212f959 (2025-04-08): Fixes update OTP when LT_CHAL is enabled.

Fixes #141.
closed 2025-04-10 by polhenarejos