polhenarejos/pico-fido #144: The  "reset/Initialize"  not work?

gxcuit, 2025-04-18:
Hi, I'm using the `waveshare_rp2350_one` board.

Once I have set the PIN, and when I want to re-flash my board, the PIN seems not to be unset. I tried flash nuke firstly, but it still asked for the PIN code.

Is this behavior normal? Is there any way I can reset the PIN?

polhenarejos, 2025-04-18:
Is the PIN recognized?
Where is it prompted?
Please provide all information you can to reproduce it step by step.

gxcuit, 2025-04-22:
> Is the PIN recognized? Where is it prompted? Please provide all information you can to reproduce it step by step.

I apologize. It turns out that I flashed the wrong nuke. After flashing the right nuke, the PIN is reset.

closed 2025-04-22 by gxcuit