polhenarejos/pico-fido #145: Unable to use WebAuth authentication on websites that utilize the Casdoor solution.

IsayIsee, 2025-04-29:
Firmware Version: 6.6.0

I deployed a private instance of Casdoor and encountered an issue while adding the WebAuth authentication method. Everything seemed normal: the selection box popped up correctly, I entered the token password as usual, and the browser prompted that the registration was successful. However, when I checked the Passkeys, there was no data. I then tested on the official Casdoor website: https://demo.casdoor.com/, and the result was the same. I used both Firefox and Google Chrome for testing, and the outcome was identical.

polhenarejos, 2025-04-29:
I do not see how to register the key in https://demo.casdoor.com/signup

IsayIsee, 2025-04-30:
@hsluoyz I can't record video，so take a screenshot to illustrate

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

no passkeys in token key

[image]

polhenarejos, 2025-04-30:
What does it say last screenshot?
Having no keys is normal as they are non-resident.

polhenarejos, 2025-05-02:
The registration process seems broken. When you click "Add Webauthn" to register your key, the request does not contain resident key flag, so no resident key is created (and this is why you don't see in the yubico app). 
Then, when you want to log in, the web SHOULD provide a list of potential credentials to see which one matches with the registered, but NO list is sent, so no credentials are provided and therefore, it cannot login.

It's a design error. If you do not create resident keys, you must provide a credential list on login. On the contrary, if you create a resident key, then you must not provide a credential list on login. But you cannot create a resident key and do not provide a credential list because it will fail, as it happens.

closed 2025-05-02 by polhenarejos