polhenarejos/pico-fido #147: Pico FIDO with gpg smartcard

turbomaster95, 2025-05-05:
Hi there! I am sorry if this is a dumb question.

I want to use my Pico W (not the 2nd gen) with passkeys and otp like that and also want to use gpg with it as a smart card.

I was wondering if i could just drag and drop the fido uf2 file and then like upgrading it as usual i could just drop the openpgp version and it would magically combine and work.

`ykman info` output:
```
Device type: YubiKey 5A
Serial number: 3865136177
Firmware version: 6.6.0
Form factor: Keychain (USB-A)
Enabled USB interfaces: OTP, FIDO, CCID

Applications
Yubico OTP  	Enabled
FIDO U2F    	Enabled
FIDO2       	Enabled
OATH        	Enabled
PIV         	Not available
OpenPGP     	Not available
YubiHSM Auth	Not available
```

polhenarejos, 2025-05-05:
Both are separate projects. Firmwares are not merged when loaded but replaced.

turbomaster95, 2025-05-05:
but is there anyway i can add the smartcard functionality?

polhenarejos, 2025-05-05:
You have to choose either fido or openpgp.

turbomaster95, 2025-05-06:
oh ok! Thanks for answering my (dumb) question!

closed 2025-05-06 by turbomaster95