polhenarejos/pico-fido #149: Pico2 Serial Number Too Long (10 Digits Instead of max 9)

Raito00, 2025-05-09:
Hello!
When flashing firmware (e.g. Pico-FIDO) to some Raspberry Pi Pico2 devices, the generated serial number is 10 digits long. This causes compatibility issues with software that expects a maximum of 9-digit serial numbers. (Original Yubikey have 8 digit serial) For example, one such integration with YubiKey fails due to this constraint.

I’ve observed the same problem across multiple software systems that enforce a 9-digit serial number limit. This issue appears only with some Pico2 boards after flashing.

Request:
Is it possible to set or force a 9-digit serial number during firmware flashing or at compile time to maintain compatibility?

polhenarejos, 2025-05-10:
Related with #135

Raito00, 2025-05-12:
The core issue here stems from the fact that many software systems, particularly those integrating with identity or authentication frameworks like LDAP, enforce a 9-digit limit on serial numbers. This is primarily due to the use of a 32-bit signed integer for storing serial numbers in directories such as LDAP, which has an upper limit of 2,147,483,647. Once a serial number exceeds this limit—or even approaches 10 digits—it risks overflowing or causing validation failures within these systems.

Original YubiKeys and similar devices typically generate 8-digit serials to stay well within this boundary. However, some Raspberry Pi Pico2 boards appear to generate 10-digit serial numbers when flashing firmware like Pico-FIDO, which leads to incompatibility with software systems expecting values ≤ 9 digits.

Request Consideration:
To maintain compatibility, particularly with systems enforcing 32-bit integer constraints, it would be ideal if the firmware could:

Generate a serial number no longer than 9 digits.

Or allow for manual configuration of the serial number at compile time or during flashing.

Is there currently a mechanism or configuration flag to set or override the serial number during the firmware flashing process? If not, would you consider implementing one?

Thank you for your support and great work on Pico-FIDO!

Raito00, 2025-05-26:
The same problem also  on ESP32 S3 Mini

commit 91aaee5b (2025-05-19): Force 8-digit serial number

Fixes #149.