polhenarejos/pico-fido #154: Yubikey Authenticator fails to open smart card connection on Windows 11

joaogmedeiros, 2025-05-15:
It works perfectly on any linux distro, but on Windows 11 24H2 it simply doesn't work.
I've tried older firmware versions
I've enabled Smart Card service
I've installed the proper Yubikey Minicard driver

[image]

polhenarejos, 2025-05-15:
Try renaming the product name to yubico yubikey.

joaogmedeiros, 2025-05-15:
> Try renaming the product name to yubico yubikey.

Yep, that solved my issue, thank you <3

[image]

[image]

closed 2025-05-15 by joaogmedeiros