polhenarejos/pico-fido #159: HELP! My Waveshare RP2350-One Doesn't work as a FIDO2 key

turbomaster95, 2025-05-27:
I have flashed my Waveshare RP2350-One with this file `pico_fido_waveshare_rp2350_one-6.6.uf2` and commissioned it with the Yubikey 4/5 preset.

But when i try to create a passkey it fails with an error on *every* *single* *website* i used it on. 

Also, `fido2-token -C /dev/hidraw0` tries to change my pin but it gives a FIDO2 error

Please help

polhenarejos, 2025-05-30:
I need you provide more information. Specially, since you have been commissioned, at least it worked in the Pico Commissioner website and thus, it is not _in every single website_.

Let's start by pasting what `lsusb` returns.
Then, indicate which websites are you trying to.

shpinog, 2025-07-20:
I got the same until I compiled the firmware with an explicit reference to "Yubikey5", then downloaded the yubikey app and set the pin there. However, it does not work on android. The apps ask for the pin, after entering it they go to error, without asking for "press confirmation" and the Waveshare RP2350-One blinks a red LED.

Young-Lord, 2025-07-26:
You may open https://www.passkeys.io/ in Chromium, and set the PIN for your key. My key works after I set a PIN.

shpinog, 2025-07-29:
> You may open https://www.passkeys.io/ in Chromium, and set the PIN for your key. My key works after I set a PIN.

After this, does authorization work for you in Android? If I set a pin code, Android asks for a pin code, but authorization does not pass.

Young-Lord, 2025-07-30:
@shpinog yes it works. I am using pico-fido2 if that matters.