polhenarejos/pico-fido #167: Support for Large CBOR Payloads for PQC in CTAP2

devi1207, 2025-06-23:
Hi polhenarejos, I'm integrating a post-quantum signature algorithm (like ML-DSA) into the Pico-FIDO project. The CBOR response during makeCredential includes authData + signature which totals around 5587 bytes.

However, I see the current CBOR buffer limit is defined as:
#define CTAP_MAX_CBOR_PAYLOAD (USB_BUFFER_SIZE - 64 - 7 - 1) // 1976 bytes
This causes cbor_encode_byte_string() to fail with error -2147483648 when encoding large payloads.

Can you suggest the proper way to support larger CBOR responses (e.g., by increasing buffer sizes safely)?
Will expanding USB_BUFFER_SIZE and CTAP_MAX_CBOR_PAYLOAD to around 7 KB (like CTAP_MAX_PACKET_SIZE) work without side effects?

Thanks!

polhenarejos, 2025-06-23:
Which board?

devi1207, 2025-06-23:
RP2040

devi1207, 2025-06-23:
I increased the buffer size to support a larger CBOR payload 
#define USB_BUFFER_SIZE         5604
#define CTAP_MAX_CBOR_PAYLOAD  5603
#define CTAP_MAX_PACKET_SIZE   (64 - 7 + 95 * (64 - 5)) // = 5603
when I try to register on [webauthn.io](https://webauthn.io/), it fails with a “unknown device state”

devi1207, 2025-06-23:
I increased the buffer size to support a larger CBOR payload
#define USB_BUFFER_SIZE 5604
#define CTAP_MAX_CBOR_PAYLOAD 5603
#define CTAP_MAX_PACKET_SIZE (64 - 7 + 95 * (64 - 5)) // = 5603
when I try to register on [webauthn.io](https://webauthn.io/), it fails with a “unknown device state”

[image]