polhenarejos/pico-fido #172: pico fido 6.6 uf2 with yubikey 5 patch (waveshare rp2040 one) doesnot display passkeys on yubico authenticator app for android

staticle, 2025-07-11:
What works -
Accounts for TOTP work

What doesnt work
Passkeys section display Insert your yuikey

Any unexpected behaviour ?
Yes, Serial key on yubico authenticator windows 11 is different then on android 

Logs from yubico authenticator android
```
18:37:50:415 [main] DEBUG c.yubico.authenticator.MainActivity -  Setting FLAG_SECURE (disallow screenshots)
18:37:50:892 [main] DEBUG y2.e -  Current app preferences:
18:37:50:899 [main] DEBUG y2.e -  flutter.APP_STATE_LAST_SECTION: accounts
18:37:50:899 [main] DEBUG y2.e -  flutter.OATH_STATE_LAYOUT: list
18:37:50:899 [main] DEBUG y2.e -  flutter.prefUsbOpenApp: false
18:37:50:900 [main] DEBUG y2.e -  Current app preferences:
18:37:50:900 [main] DEBUG y2.e -  flutter.APP_STATE_LAST_SECTION: accounts
18:37:50:900 [main] DEBUG y2.e -  flutter.OATH_STATE_LAYOUT: list
18:37:50:900 [main] DEBUG y2.e -  flutter.prefUsbOpenApp: false
18:37:50:902 [main] DEBUG y2.t -  HomeContextManager initialized
18:37:50:903 [main] DEBUG G2.r -  OathManager initialized
18:37:50:904 [main] DEBUG A2.l -  FidoManager initialized
18:37:50:904 [main] DEBUG E2.i -  ManagementManager initialized
18:37:50:937 [main] DEBUG c.yubico.authenticator.MainActivity -  Context Oath is already supported by active manager.
18:37:50:937 [main] DEBUG c.yubico.authenticator.MainActivity -  A YubiKey is connected, using it with the context Oath
18:37:50:943 [main] TRACE y2.a -  Activity alias 'AliasMainActivity' is enabled: 1
18:37:50:943 [main] DEBUG y2.a -  Enabled USB discovery by setting state of AliasMainActivity to ENABLED
18:37:50:944 [main] DEBUG c.yubico.authenticator.MainActivity -  Starting nfc discovery
18:37:50:945 [main] DEBUG c.yubico.authenticator.MainActivity -  Starting usb discovery
18:37:50:949 [main] DEBUG N2.d -  UsbDevice attached: /dev/bus/usb/002/002
18:37:50:950 [main] DEBUG z2.i -  SCP params set to null
18:37:50:953 [main] DEBUG O2.j -  USB connection opened: O2.i@fb8e377
18:37:50:953 [main] TRACE O2.i -  10 bytes sent over ccid: 62 00 00 00 00 00 00 00 00 00 
18:37:50:953 [main] TRACE O2.i -  33 bytes received: 80 17 00 00 00 00 00 00 00 00 3b fd 13 00 00 81 31 fe 15 80 73 c0 21 c0 57 59 75 62 69 4b 65 79 40 
18:37:50:954 [main] TRACE O2.i -  23 bytes sent over ccid: 6f 0d 00 00 00 00 01 00 00 00 00 a4 04 00 08 a0 00 00 05 27 47 11 17 
18:37:50:955 [main] TRACE O2.i -  17 bytes received: 80 07 00 00 00 00 01 00 00 00 36 2e 36 2e 30 90 00 
18:37:50:955 [main] DEBUG c3.e -  Management session initialized for connection=i, version=6.6.0
18:37:50:955 [main] DEBUG c3.e -  Reading config page 0...
18:37:50:955 [main] TRACE O2.i -  15 bytes sent over ccid: 6f 05 00 00 00 00 02 00 00 00 00 1d 00 00 00 
18:37:50:956 [main] TRACE O2.i -  41 bytes received: 80 1f 00 00 00 00 02 00 00 00 1c 01 02 02 23 02 04 02 62 50 08 04 01 01 05 03 06 06 00 03 02 02 3b 08 01 80 0a 01 00 90 00 
18:37:50:957 [main] DEBUG e3.a -  Read info DeviceInfo{config=DeviceConfig{enabledCapabilities={USB=571}, autoEjectTimeout=0, challengeResponseTimeout=0, deviceFlags=128, nfcRestricted=false}, serialNumber=39997448, version=6.6.0, formFactor=USB_A_KEYCHAIN, supportedCapabilities={USB=547}, isLocked=false, isFips=false, isSky=false, partNumber=null, fipsCapable=0, fipsApproved=0, pinComplexity=false, resetBlocked=0, fpsVersion=null, stmVersion=null, versionQualifier=6.6.0.final.0}
18:37:50:957 [main] DEBUG O2.j -  USB connection closed: O2.i@fb8e377
18:37:50:958 [main] DEBUG z2.i -  SCP params set to null
18:37:50:958 [main] DEBUG c.yubico.authenticator.MainActivity -  There was no context with pending action
18:37:50:958 [main] DEBUG c.yubico.authenticator.MainActivity -  Device supports: [Home, Oath, FidoPasskeys, Management]
18:37:50:958 [main] DEBUG c.yubico.authenticator.MainActivity -  Current context: Oath
18:37:50:959 [main] DEBUG y2.e -  registering change listener
18:37:50:960 [pool-8-thread-1] DEBUG O2.j -  USB connection opened: O2.i@f6b9e8b
18:37:50:961 [pool-8-thread-1] TRACE O2.i -  10 bytes sent over ccid: 62 00 00 00 00 00 00 00 00 00 
18:37:50:961 [pool-8-thread-1] TRACE O2.i -  33 bytes received: 80 17 00 00 00 00 00 00 00 00 3b fd 13 00 00 81 31 fe 15 80 73 c0 21 c0 57 59 75 62 69 4b 65 79 40 
18:37:50:962 [pool-8-thread-1] TRACE O2.i -  23 bytes sent over ccid: 6f 0d 00 00 00 00 01 00 00 00 00 a4 04 00 08 a0 00 00 05 27 21 01 01 
18:37:50:962 [pool-8-thread-1] TRACE O2.i -  30 bytes received: 80 14 00 00 00 00 01 00 00 00 79 03 06 06 00 71 08 44 45 36 32 35 30 30 38 7b 01 01 90 00 
18:37:50:963 [pool-8-thread-1] DEBUG d3.i -  OATH session initialized (version=6.6.0, isAccessKeySet=false)
18:37:50:966 [pool-8-thread-1] INFO  d3.i -  Calculating all codes for time=1752239280966
18:37:50:967 [pool-8-thread-1] TRACE O2.i -  27 bytes sent over ccid: 6f 11 00 00 00 00 02 00 00 00 00 a4 00 01 00 00 0a 74 08 00 00 00 00 03 7b 3c 28 
18:37:50:967 [pool-8-thread-1] TRACE O2.i -  12 bytes received: 80 02 00 00 00 00 02 00 00 00 90 00 
18:37:50:968 [pool-8-thread-1] DEBUG O2.j -  USB connection closed: O2.i@f6b9e8b
18:37:51:046 [main] DEBUG G2.r -  Successfully read Oath session info (and credentials if unlocked) from connected key
18:37:51:157 [main] INFO  com.yubico.authenticator.Log -  Logging initialized, outputting to Android/logcat [android.logger]
18:38:06:651 [main] DEBUG com.yubico.authenticator.Log -  Log level set to TRAFFIC [about]
18:38:12:768 [main] DEBUG y2.e -  unregistered change listener
18:38:12:768 [main] DEBUG c.yubico.authenticator.MainActivity -  Stopped usb discovery
18:38:12:770 [main] TRACE y2.a -  Activity alias 'AliasMainActivity' is enabled: 0
18:38:12:770 [main] DEBUG y2.a -  Disabled USB discovery by setting state of AliasMainActivity to DEFAULT
18:38:12:771 [main] TRACE y2.a -  Activity alias 'AliasNdefActivity' is enabled: 0
18:38:12:771 [main] DEBUG y2.a -  Enabled NFC discovery by setting state of AliasNdefActivity to DEFAULT
18:38:12:776 [main] TRACE y2.a -  Activity alias 'AliasMainActivity' is enabled: 1
18:38:12:776 [main] DEBUG y2.a -  Enabled USB discovery by setting state of AliasMainActivity to ENABLED
18:38:12:777 [main] DEBUG c.yubico.authenticator.MainActivity -  Starting nfc discovery
[... 103 lines cut ...]
	at b3.e.<init>(SourceFile:13)
	at b3.e.<init>(SourceFile:28)
	at b3.e.<init>(SourceFile:6)
	at A2.l.T(SourceFile:9)
	at A2.l.C(SourceFile:1)
	at A2.l$h.a(SourceFile:22)
	at A2.l$h.b(SourceFile:3)
	at N2.h.o(SourceFile:14)
	at N2.g.run(SourceFile:1)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:487)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)

18:38:19:089 [main] DEBUG com.yubico.authenticator.Log -  Received nfc state: NfcState.failure [android.nfc_overlay]
18:38:19:090 [main] DEBUG com.yubico.authenticator.Log -  Event change: Instance of 'NfcEvent' -> Instance of 'NfcSetViewEvent' [android.nfc_event_notifier]
18:38:19:090 [main] DEBUG com.yubico.authenticator.Log -  Event change: Instance of 'NfcSetViewEvent' -> Instance of 'NfcHideViewEvent' [android.nfc_event_notifier]
18:38:19:096 [pool-10-thread-1] DEBUG O2.j -  USB connection closed: O2.g@437c6ca
18:38:23:039 [main] DEBUG com.yubico.authenticator.a -  App context is now Home
18:38:23:041 [main] DEBUG A2.l -  FidoManager deactivated
18:38:23:041 [main] DEBUG y2.t -  HomeContextManager activated
18:38:32:756 [main] INFO  com.yubico.authenticator.Log -  Copying log to clipboard (7.2.3)... [about]
18:38:32:757 [main] DEBUG com.yubico.authenticator.Log -  Getting logs... [android.logger]
```

polhenarejos, 2025-07-11:
Try changing Product Name to “Yubico Yubikey”.

staticle, 2025-07-11:
Already changed product name encountering error even after changing name to
Yubico Yubikey

staticle, 2025-07-28:
can you try it on android phone via otg ?

polhenarejos, 2025-08-31:
Really strange. I’ve been using yubico library without problems in PC/mac. Btw, there are other uint64 in other fields. You just commented VENDOR commands, which might be unsupported by this version if it’s really old.

sst311212, 2025-08-31:
ykman library support uint64 unpack but android's skip it, nice.
https://github.com/Yubico/python-fido2/blob/ed072c1b4fd4127a72758f00b55e0f3b7242096c/fido2/cbor.py#L102-L113

polhenarejos, 2025-08-31:
From the spec:

> This subCommand MUST include a subCommandParams map that MUST contain [vendorCommandId](https://fidoalliance.org/specs/fido-v2.2-ps-20250714/fido-client-to-authenticator-protocol-v2.2-ps-20250714.html#authconfig-vendorcommandid) as a member. The vendor randomly selects a 64-bit Unsigned Integer value to use for the value of vendorCommandId, e.g., by using a cryptographic random number generator. An example of such a vendorCommandId value is (in hex): 0x4e5a15aa89d2b8b6. This approach avoids collisions amongst different vendors' vendorCommandIds.

So, it seems the app doesn't follow the spec.

Besides this, this app has been discontinued in favour of https://github.com/Yubico/yubioath-flutter 
Could you check if it still happens with flutter?

sst311212, 2025-09-01:
Yes, all I talking about is flutter app, it use [yubikit-android](https://github.com/Yubico/yubikit-android) library.
https://github.com/Yubico/yubioath-flutter/blob/f2af8092fbc2dab014f7910b61598f6474d87e3c/android/build.gradle

polhenarejos, 2025-09-01:
Can you confirm you are using version 7.2.3?

sst311212, 2025-09-01:
https://youtu.be/XS5c2NEbWik

polhenarejos, 2025-09-01:
I tested with Android 9 and 44c5ad4 fixes this behavior.

sst311212, 2025-09-02:
The nightly dev's build fix `uint64` problem, but it has another issue.

When I first plug in picokey (or press reset btn) and enter into `Passkeys` menu, it will not showing enter PIN textbox.
It's likely `ctap_req` buffer doesn't contain the data which sent from android app. Maybe not yet received whole data?
https://github.com/polhenarejos/pico-keys-sdk/blob/113e720fcaaa6b9ca74d114bee1923bb2619ba3b/src/usb/hid/hid.c#L342-L345

```
09-02 14:59:12.494 15405 15405 V OplusZoomWindowDecorViewHelper: setLastReportedMergedConfiguration mZoomDisplayHeight: 2400 getDecorView.14238998
09-02 14:59:12.494 15405 15405 D com.yubico.authenticator.AppPreferences: unregistered change listener
09-02 14:59:12.494 15405 15405 D com.yubico.authenticator.MainActivity: Stopped usb discovery
09-02 14:59:12.495 15405 15405 V com.yubico.authenticator.ActivityUtil: Activity alias 'AliasNdefActivity' is enabled: 0
09-02 14:59:12.495 15405 15405 D com.yubico.authenticator.ActivityUtil: Enabled NFC discovery by setting state of AliasNdefActivity to DEFAULT
09-02 14:59:12.496 15405 15405 V com.yubico.authenticator.ActivityUtil: Activity alias 'AliasMainActivity' is enabled: 1
09-02 14:59:12.496 15405 15405 D com.yubico.authenticator.ActivityUtil: Enabled USB discovery by setting state of AliasMainActivity to ENABLED
09-02 14:59:12.496 15405 15405 D com.yubico.authenticator.MainActivity: Starting nfc discovery
09-02 14:59:12.497 15405 15405 D com.yubico.authenticator.MainActivity: Starting usb discovery
09-02 14:59:12.499 15405 15405 D com.yubico.yubikit.android.transport.usb.UsbDeviceManager: UsbDevice attached: /dev/bus/usb/001/011
09-02 14:59:12.499 15405 15405 D com.yubico.authenticator.device.DeviceManager: SCP params set to null
09-02 14:59:12.501 15405 15405 D com.yubico.yubikit.android.transport.usb.connection.UsbYubiKeyConnection: USB connection opened: com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection@8428659
09-02 14:59:12.501 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 10 bytes sent over ccid: 62 00 00 00 00 00 00 00 00 00
09-02 14:59:12.501 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 33 bytes received: 80 17 00 00 00 00 00 00 00 00 3b fd 13 00 00 81 31 fe 15 80 73 c0 21 c0 57 59 75 62 69 4b 65 79 40
09-02 14:59:12.502 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 23 bytes sent over ccid: 6f 0d 00 00 00 00 01 00 00 00 00 a4 04 00 08 a0 00 00 05 27 47 11 17
09-02 14:59:12.556 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 17 bytes received: 80 07 00 00 00 00 01 00 00 00 36 2e 36 2e 30 90 00
09-02 14:59:12.557 15405 15405 D com.yubico.yubikit.management.ManagementSession: Management session initialized for connection=UsbSmartCardConnection, version=6.6.0
09-02 14:59:12.557 15405 15405 D com.yubico.yubikit.management.ManagementSession: Reading config page 0...
09-02 14:59:12.557 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 15 bytes sent over ccid: 6f 05 00 00 00 00 02 00 00 00 00 1d 00 00 00
09-02 14:59:12.557 15405 15405 V com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection: 41 bytes received: 80 1f 00 00 00 00 02 00 00 00 1c 01 02 02 23 02 04 02 8e bf df 04 01 01 05 03 06 06 00 03 02 02 3b 08 01 80 0a 01 00 90 00
09-02 14:59:12.557 15405 15405 D com.yubico.yubikit.support.DeviceUtil: Read info DeviceInfo{config=DeviceConfig{enabledCapabilities={USB=571}, autoEjectTimeout=0, challengeResponseTimeout=0, deviceFlags=128, nfcRestricted=false}, serialNumber=42909663, version=6.6.0, formFactor=USB_A_KEYCHAIN, supportedCapabilities={USB=547}, isLocked=false, isFips=false, isSky=false, partNumber=null, fipsCapable=0, fipsApproved=0, pinComplexity=false, resetBlocked=0, fpsVersion=null, stmVersion=null, versionQualifier=6.6.0.final.0}
09-02 14:59:12.557 15405 15405 D UsbDeviceConnectionJNI: close
09-02 14:59:12.558 15405 15405 D com.yubico.yubikit.android.transport.usb.connection.UsbYubiKeyConnection: USB connection closed: com.yubico.yubikit.android.transport.usb.connection.UsbSmartCardConnection@8428659
09-02 14:59:12.558 15405 15405 D com.yubico.authenticator.device.DeviceManager: SCP params set to null
09-02 14:59:12.558 15405 15405 D com.yubico.authenticator.MainActivity: There was no context with pending action
09-02 14:59:12.558 15405 15405 D com.yubico.authenticator.MainActivity: Device supports: [Home, Oath, FidoPasskeys, Management]
09-02 14:59:12.558 15405 15405 D com.yubico.authenticator.MainActivity: Current context: FidoPasskeys
09-02 14:59:12.560 15405 15405 D com.yubico.authenticator.AppPreferences: registering change listener
09-02 14:59:12.563 15405 16944 D com.yubico.yubikit.android.transport.usb.connection.UsbYubiKeyConnection: USB connection opened: com.yubico.yubikit.android.transport.usb.connection.UsbFidoConnection@5f8081b
09-02 14:59:12.565 15405 15405 D com.yubico.authenticator.Log: appLifecycleStateChange triggered with: inactive [android.window_state_provider]
09-02 14:59:12.565 15405 15405 D com.yubico.authenticator.Log: Updated windowState to WindowState(focused: false, visible: false, active: false, hidden: false) [android.window_state_provider]
09-02 14:59:12.567 15405 15405 D com.yubico.authenticator.Log: appLifecycleStateChange triggered with: resumed [android.window_state_provider]
[... cut ...]
09-02 14:59:12.579 15405 15405 D com.yubico.authenticator.MainActivity:         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:651)
09-02 14:59:12.579 15405 15405 D com.yubico.authenticator.MainActivity:         at java.lang.Thread.run(Thread.java:1119)
09-02 14:59:12.609 15405 16944 D UsbDeviceConnectionJNI: close
09-02 14:59:12.609 15405 16944 D com.yubico.yubikit.android.transport.usb.connection.UsbYubiKeyConnection: USB connection closed: com.yubico.yubikit.android.transport.usb.connection.UsbFidoConnection@5f8081b
```

sst311212, 2025-09-03:
It's weird, I have this issue on my `OnePlus 7T` but my brother's `OnePlus 12R` doesn't.

And I noticed `ctap_req` has contains all zero at this situation, by adding this code after `driver_init_hid` can solve it.
```
driver_init_hid();
if (memcmp(ctap_req, ctap_resp, 64) == 0) {
    return last_packet_time = 0;
}
```

commit 44c5ad4a (2025-09-01): Some VIDs do not support VENDOR_CONFIG values.

Fixes #172.