polhenarejos/pico-fido #183: Google Again...

shpinog, 2025-08-31:
I can create and use keys on any sites, but I can't create a key in google getting an error that my device is not suitable. (I noticed this after updating to the latest nightly build) I tried the latest nightly build, I also built it myself, and yes I changed the product name to "Yubico Yubikey"

Am I mistaken somewhere or has Google changed something? Previously (in previous versions) I could create a key in Google.

polhenarejos, 2025-08-31:
Please provide more info, specially which website and browser/OS.

mannapijush1coretech, 2025-09-02:
yes!
i just flashed with pico with default vid pid which provided in release it shows on google "This security key can't be used"
OS- Windows 10 22H2 (Build 19045.6216)
Browser - Opera One(version: 120.0.5543.201)

mannapijush1coretech, 2025-09-05:
can you please kindly write the flow of registering the key