polhenarejos/pico-fido #184: Invalid Passkey error from Bitwarden when attempting to use Pico Key for passwordless login

lucor001, 2025-09-03:
I'm using the Waveshare Pico One RP2350.

I used the latest Nightly Development build and used Pico Commissioner to set it as a Yubikey 4/5.  It seems like it works on some sites, but not others.  Even for Bitwarden it works fine as a 2FA Passkey, just not passwordless.  It seems to work fine for logging into Google, but Amazon fails to create the Passkey.  I tried webauthn.io and it works successfully there.

I'm a bit surprised I'm getting trouble with Bitwarden, so I thought I'd check to see if perhaps I was just doing something boneheaded or not.  I've tried different versions of the fido hardware, I even tried pico-fido2, nothing seems to work.

polhenarejos, 2025-09-03:
The webpage reports "Error 400: invalid grant" with no more details. It will be hard to know why.

lucor001, 2025-09-03:
I guess it's reassuring that it's not only me that sees an issue :).  Perhaps this is what they mean by Beta.  I do have a Yubikey and it does work correctly, but maybe there's some difference in how Bitwarden is querying the device.

polhenarejos, 2025-09-03:
Try the following:
In your DevTools, in the Network view, you should find a POST to /token that returns error 400.
Copy-paste the payload you sent to the endpoint. It's in the Payload tab, and there's a value called `deviceResponse`, which is what the device returns.

lucor001, 2025-09-03:
Pico Fido (got a 400 Bad-Request):
deviceResponse
{"id":"8dACArx0bzQ90rVmMVPNvyjV4rl0dYzqWP4jU6onloUXqPdai-aZ-i57hiBYa2xiZeKlcT3EJIeKwMGDNs_fFd3sF1vsAJUfoHB24B2H6-pbIZt-RXYDW3AiofBF5OhNjcdxQfZpFDJkXzr02vNI1W5oB7btcnQZlIvNs1fkTR25y9W-mXUGV4GHIQZFoqOW7Jn8vFG4ql5QcrygKRJGiRm06z1ihlJSr15Jyr53j9CB1Fy0elHz1hNRDVAkoz0AAk8KRMCto17_4PzQtJe7JB0","rawId":"8dACArx0bzQ90rVmMVPNvyjV4rl0dYzqWP4jU6onloUXqPdai-aZ-i57hiBYa2xiZeKlcT3EJIeKwMGDNs_fFd3sF1vsAJUfoHB24B2H6-pbIZt-RXYDW3AiofBF5OhNjcdxQfZpFDJkXzr02vNI1W5oB7btcnQZlIvNs1fkTR25y9W-mXUGV4GHIQZFoqOW7Jn8vFG4ql5QcrygKRJGiRm06z1ihlJSr15Jyr53j9CB1Fy0elHz1hNRDVAkoz0AAk8KRMCto17_4PzQtJe7JB0","type":"public-key","extensions":{},"response":{"authenticatorData":"Bkl2q16S_uHjFFtbEhyNOaYwQzTR0MClU-ce_MUcSy2FAAAAkqFraG1hYy1zZWNyZXRYMMDee8B1fkVUvYHzZRo6UgldKtcSNkOyln2BHH2biehW5_2dgYiYZFdjCf9WkT6TxA","signature":"MEUCIQCd626srJXoiYdBcv-AbQLMYHt9Ad-IeKB1SkJWJCralQIgYti5KPmE6f_R5IpvYeFvHAyq5K9UrCpAIVtyNn0syOQ","clientDataJSON":"eyJ0eXBlIjoid2ViYXV0aG4uZ2V0IiwiY2hhbGxlbmdlIjoiQ0RUaGNROS04TDBjQkp0Vk9TUTdtZyIsIm9yaWdpbiI6Imh0dHBzOi8vdmF1bHQuYml0d2FyZGVuLmNvbSIsImNyb3NzT3JpZ2luIjpmYWxzZX0","userHandle":"QhgoqLyBgkm1hqzSAWi9OA"}}

Yubikey (got a 200 OK):
deviceResponse
{"id":"jnmvmcQdfNGNkwuSexrfHB2ZvElZLN06W6HKvIZ-eZo42538xb7lBwXfcToduNzn","rawId":"jnmvmcQdfNGNkwuSexrfHB2ZvElZLN06W6HKvIZ-eZo42538xb7lBwXfcToduNzn","type":"public-key","extensions":{},"response":{"authenticatorData":"Bkl2q16S_uHjFFtbEhyNOaYwQzTR0MClU-ce_MUcSy2FAAAAgqFraG1hYy1zZWNyZXRYMFUR59b4gDXJQT7xKVDVOG7cxg3LX8p5p-arUpa_LeYahNkVsgF3J2-Qw4g16C0sjA","signature":"MEUCIERfIsaDUhq7N56bMJ5e2k8T_3wHnfCTfmI_VUK6smhwAiEA4V9EGk3XnNy78bxTfZiOEcjfMHigH8GkzJFnEzDunZs","clientDataJSON":"eyJ0eXBlIjoid2ViYXV0aG4uZ2V0IiwiY2hhbGxlbmdlIjoicThVYlB5VUxrZXhUV084MFlrS0dodyIsIm9yaWdpbiI6Imh0dHBzOi8vdmF1bHQuYml0d2FyZGVuLmNvbSIsImNyb3NzT3JpZ2luIjpmYWxzZX0","userHandle":"QhgoqLyBgkm1hqzSAWi9OA"}}

sst311212, 2025-09-06:
You should do factory reset in Yubico Authenticator after flashing latest development firmware.

BTW, I can add Passkey to amazon without issues. (Windows 10 + Firefox 142)

polhenarejos, 2025-09-06:
It could be an incompatibility of old resident system. I pushed a fix for it but I'd try with @sst311212 advice of resetting Authenticator.
Tomorrow will be available the new nightly build.

lucor001, 2025-09-06:
You're right, good tip!  I reset FIDO2 in Yubico Authenticator and then commissioned it again which made me setup a pin.  After that I tried the Authenticator (works), Bitwarden (works), Amazon (works).  I also tried a few more accounts and everything works as expected.  I will download the nightly tomorrow and give it a shot.

Thanks folks!

lucor001, 2025-09-07:
I tried the latest nightly and everything still works perfectly!  I wasn't sure exactly how to confirm your latest changes, so this time I didn't reset the key in Yubico Authenticator after flashing the firmware.  I can confirm that the Authenticator still reads the key correctly and the keys stored previously also work. 

I think this issue is now resolved, thanks!  I will close.

commit 1ac628d2 (2025-09-04): Major refactor on resident keys.

Now, credential ids have shorter and fixed length (40) to avoid issues with some servers, which have maximum credential id length constraints.

Fixes #184
closed 2025-09-07 by lucor001