polhenarejos/pico-fido #187: PIN Complexity proposal

token2, 2025-09-17:
We hereby propose a modification (tested and confirmed on a Pico2 device) to implement PIN complexity, similar to Token2 [PIN+](https://www.token2.com/site/page/token2-fido2-pin-see-the-pin-complexity-in-action) devices. We did not submit a pull request for the modification to the original file (src/fido/cbor_client_pin.c), as this feature may not be required by all users. Instead, we are providing an alternative version for those who wish to add PIN complexity. Since the data is encrypted using the PIN, we recommend enforcing PIN complexity at the firmware level.

[cbor_client_pin.c_PIN-Complexity.txt](https://github.com/user-attachments/files/22378042/cbor_client_pin.c_PIN-Complexity.txt)

polhenarejos, 2025-09-24:
I saw your code and looks interesting. However I need to think on a way to enable/disable these particular features for the most of the public via pico-fido-tool/web commissioner.

token2, 2025-09-26:
> I saw your code and looks interesting. However I need to think on a way to enable/disable these particular features for the most of the public via pico-fido-tool/web commissioner.

Thank you for considering this. Can you confirm whether a complex PIN is the only effective defense against an offline brute-force attack? If an attacker is able to extract a copy of the key’s flash (e.g. using picotool), they could run unlimited PIN guesses offline; by contrast, a standard FIDO2 token enforces an on-device attempt limit (typically 8 attempts).

token2, 2025-09-26:
> It will not work. You cannot clone the OTP region where the primary key is located. If you do that, a new fresh key will be initialized on the second board and thus all data from flash will not be accessible (since the OTP key is different). Even if the PIN is correct, the data is corrupted.

Sorry, I did not try with two keys yet, but with one single key I have some concerning results:

---

## Step 1
- RP2350 dongle with the UF2 from your site.
- PIN set to `1234`.
- Three passkeys created.

## Step 2
Enter **BOOTSEL** mode.

Run:
```bash
picotool.exe save --all all_flash_dump.bin
Saving file: [==============================] 100%
Wrote 16777216 bytes to all_flash_dump.bin
```

Exit BOOTSEL and plug the device in normal mode. Create two more passkeys (now a total of five).

## Step 3
Re-enter **BOOTSEL** mode with the same key.

Load the previously dumped file:
```bash
picotool.exe load --ignore-partitions --family rp2350-arm-s all_flash_dump.bin -t bin -o 0x10000000 -v
Loading into Flash:   [==============================] 100%
Verifying Flash:      [==============================] 100%
OK
```

## Step 4
Plug the device in normal mode.

- The PIN and passkeys are the same as before.
- After entering wrong PIN attempts the key becomes blocked (`FIDO_ERR_PIN_BLOCKED` status).
- Repeat Step 3 (reflashing the saved dump) and the device is unblocked again — it is not in `FIDO_ERR_PIN_BLOCKED` status, it has **3 passkeys** (not 5), and shows **8 more wrong-PIN attempts**.

---

Am I missing something or this is a practical "offline" brute-force attack vector against the same physical device?

In short- dump its flash before attempting PIN guesses, perform up to the allowed wrong-PIN attempts, then restore the flash dump to reset the attempt counter and repeat.

My1, 2025-09-27:
can the firmware on the pico itself remove the disable_bootsel things or are these permanent? because a firmware could theoretically make it so it wipes down the data sections, verifies all is clean and then reopen these.

token2, 2025-09-27:
> In this case, yes you [@token2](https://github.com/token2) and [@My1](https://github.com/My1) are right.
> 
> There're `DISABLE_BOOTSEL_USB_PICOBOOT_IFC` and `DISABLE_BOOTSEL_USB_MSD_IFC` that would block the access to BOOTSEL mode. However, the device will become unupgradeable at all. It'll lock to this state forever. `DISABLE_BOOTSEL_USB_PICOBOOT_IFC` will lock `picotool` access but the data partition could still be overwrittable through MSD to a skill attacker.
> 
> One approach could be create a sideload interface to load new firmware versions, whilst `BOOTSEL` interface is blocked forever. But this is something it will not happen in a shor time. BTW, the RP2350 board will be locked to Pico Keys project forever.
> 
> What do you think.

Ok, if there is a way to disable the ability to flush the dump at the cost of making the device non-upgradeable, that should ultimately be the user’s decision — it’s a trade-off. However, since you confirmed that PIN brute force is a real attack vector, the code should not allow simple PINs like 1234 (similar to our PIN+ logic). We would actually recommend removing the option entirely and enforcing stronger requirements — for example, a minimum of 12 characters, including at least one lowercase letter, one uppercase letter, one digit, and one special character (similar to default Active Directory password policies).

With that policy, brute-forcing a PIN would take thousands of years, even using the most powerful computing resources available today or in the foreseeable future.

Regarding your last sentence, do you mean that you want to drop support for the RP2040?

My1, 2025-09-27:
AD only mentions to use 3 character types rather than all 4 (which one might need to check if that's still enough to be secure in this context, but if it is, would be better as especially special characters are a real pain to enter with different situations like language keyboards, mobile on screen keyboards etc. so 12 chars with 3 types might be less ugly and also not cause the problem where people just add a special character at the end just to satisfy the rules, something even I use from time to time as someone often using wordlist-based passwords.

My1, 2025-09-27:
Although assuming there is no code verification, i guess the speed goes up again as one could literally just patch out the function that handles the retry counter.

I hope that otp is at least handled in a way opaque to the fw because if you could flash firmware that reads that and gives it out, that'd be annoying at the least.

polhenarejos, 2025-09-27:
I shouldn’t say this but it’s easier than using a physical board. PIN is not encrypted, only hashed. So you could use an offline program to generate the correct hash and compare with the dumped one and check if both match. 
With a typical CPU of 5 Mhashes/sec could take like 250 days, less than a year for 8 figures of 3 classes. 
Actually the hash of PIN typically does 16 hashes, so this number is a bit higher but you see the point.

My1, 2025-09-27:
I think T2 said that not the PIN itself being encrypted but rather the actual data like key material and stuff is being encrypted by the PIN (+ the OTP data), although I obviously do not know if there is an extra PIN hash stored somewhere and/or if that additionally contains any protections aside from a simple salt attached to the hash (which in terms of a numeric pin would be completely too little, as rainbow tables really arent a problem in a space of 6-8-digit-only PINs.)

polhenarejos, 2025-10-01:
This project started with RP2040, without any measure for protecting data. So, PIN system inherited from this approach. With latest commits, it leverages OTP but now I am switching to a new system where the device key depends on the OTP rather than on PIN. However, it's a bit complex specially for the upgrade.

token2, 2025-10-01:
> The only way is to disable USB BOOTLOADER, which invalidates `picotool` but also upgrading through normal UF2 procedure. If you are able to write partitions, then you can always overwrite the PIN counter.

I would say this is an acceptable workaround (with a trade-off, of course)

sylvainpelissier, 2025-10-05:
With the new version using the OTP key, the key is hashed with the PIN concatenated several time using SHA-256 int the `hash_multi_ext`. The construction is not really standard and SHA-256 is vulnerable to [hash extension attack](https://en.wikipedia.org/wiki/Length_extension_attack). Even if I don't see a direct impact on the current usage, it would be more robust and more efficient to use HMAC-SHA256 since it uses internally only two calls to SHA-256. The updated function would simply calls:

```c
mbedtls_md_hmac(mbedtls_md_info_from_type(MBEDTLS_MD_SHA256), otp_key_1, 32, input, len, output);
```

I can create a PR with the modification if needed.

My1, 2025-10-05:
if we have a secret to hash the PIN with yeah HMAC would be the way to go if we cant use slow hashes.

token2, 2025-10-05:
> I think we may use PBKDF2-SHA256 algorithm with maybe 1000 iterations but since for boards with OTP keys it would already need a reflash of the board after each 8 trials I think it would be already very slow to bruteforce with HMAC.

Let me clarify: two distinct attack vectors exist. The first is a flash‑dump/physical attack in which the device can be reflashed and an attacker is afforded up to eight PIN attempts per reflash. The second is recovery of the PIN hash from the flash memory followed by offline cracking of that hash. The chosen hash function and parameterization must be sufficiently strong to mitigate both on‑device and offline brute‑force attacks.

token2, 2025-10-05:
> I was speaking of the on-device attack, since for offline attack the bruteforce would target the 256-bit OTP key used in the HMAC-SHA256.

On-device attack is literally just a slow brute-force, so hashing isn’t playing a role there. We’ve already found a _mitigation_: disabling BOOTSEL.