polhenarejos/pico-fido #188: Passkey Store in App issues.

syafiqshaary, 2025-09-18:
Just testing on Tenstar Rp2350 Usb flashing with firmware waveshare rp2350 one v6.6 nightly dev,its working but some account having a problem like i can create passkey for discord, after doned create login wont work using passkey instead, but it works inside an setting app like wanna remove passkey discord will ass confirmation passkey if want to remove it, on nightly dev all passkey stored in yubico key authenticator app iOS but some. if creating using mobile safari/chrome and PC/Laptop but it wont on stable firmware. Also for now stable firmware is working greate for now. The issues rn on stable is passkey i created doesn't stored in hardware not appears in iOS/App.
Pico Commissioner set on my tenstar rp2350 usb you can tried if you have one.

Known Vendor :  Yubikey 4/5
VID PID :  1050:0407
LED PIN :  22
Product Name :  Yubico Yubikey
product name is needed like i mention or you hardware probably wont recognized.
Then click on "Commission via WebAuthn" to testing.

polhenarejos, 2025-09-18:
So what’s the exact problem with nightly dev buil?

sst311212, 2025-09-18:
Because Discord will create a credential as security key instead of Passkey by default.

You can intercept `navigator.credentials.create` in browser to force it make a resident key.

polhenarejos, 2025-09-24:
Seems Discord is not following the standard. Nothing to do.
Even if I force resident key, it will not login. No errors, but no login.

syafiqshaary, 2025-09-24:
> Seems Discord is not following the standard. Nothing to do.
> Even if I force resident key, it will not login. No errors, but no login.

Yeahhh, it's will not work or i didnt completely tested but Nightly Development previously it wont worked when set using Windows,Web Browser, iOS & Android, didn't tested on Nightly Stable, for Stable worked for login all in all platform. U need to set the passkey through Android device will fixes everythings. Currently issues is i can only saw stored passkey on Yubico authenticator for windows, for mobile device it will not showed.

sst311212, 2025-09-24:
But I can use Picokey to login Discord after making a resident key.

[image]